negativo17/nvidia-driver #1: nvidia-340 driver for Fedora 24

KritonK, 2016-06-23:
Will you be updating the fedora-nvidia-340 repository for Fedora 24? The Fedora 23 packages seem to work fine after upgrading to Fedora 24, so it should simply be a matter of rebuilding/repackaging.

scaronni, 2016-06-23:
Well, if it works for you I can rebuild them, it does not cost me anything. I'm not maintaining them officially as I've really no interest and no hardware to test; so no advertisement.

Rebuild coming.

Regards,
--Simone

scaronni, 2016-06-23:
I did quite a few changes on the 340 nvidia-driver and nvidia-kmod packages:

https://github.com/negativo17/nvidia-driver-340
https://github.com/negativo17/nvidia-kmod-340

I've uploaded only the fedora-24 folder, can you please give me some feedback? As I've said I have no hardware like that. If it works, I will udpate the others as well.

I guess also that the 340.96 driver does not support recent 1.18 X.org servers and just works by chance. If the driver is working for you, can you also remove this file:

/etc/X11/xorg.conf.d/99-nvidia-ignoreabi.conf

And check if it still works?

KritonK, 2016-06-24:
Just to be sure, I assume that you are referring to the new set of 340.96-2.fc24 packages that  just appeared in the repository.

I just updated, rebooted, and the updated driver works fine. The same is true after removing /etc/X11/xorg.conf.d/99-nvidia-ignoreabi.conf and rebooting. (Should I keep it removed or do I need to put it back?)

By "works fine" I means that GDM runs, I can start KDE and can then go about my business, e.g., open a few windows and write this reply.

Tested using a GeForce 9600 GT.

scaronni, 2016-06-24:
Thanks for feedback, I will update the other branches as well.

Leve the ignore ABI file off, I will remove it and put it in Fedora 25+ packages. Apparently the 340.96 already support the X.org 1.18 driver ABI.

closed 2016-06-24 by scaronni