negativo17/nvidia-driver #4: GDM fails to boot after installing

hworld, 2016-11-05:
Hey there! I'm trying to get the nvidia drivers installed on my Dell XPS 15 (9550). There is an Intel integrated GPU and a GeForce GTX 960M. I'm running a fresh install of Fedora 25 beta that is fully updated.

I've been using Bumblebee in Fedora with the proprietary driver detailed here: https://fedoraproject.org/wiki/Bumblebee It's worked well for fedora 24 and lower. I was reading through this blog post: https://blogs.gnome.org/uraeus/2016/11/01/discrete-graphics-and-fedora-workstation-25/ and got really excited about possibly getting better support in Gnome/Fedora for GPU switching.

Running apps with DRI_PRIME=1 in Fedora 25 launches correctly on the NVIDIA GPU using Nouveau.

Anyway, when I try to reboot my computer after installing GDM shows the "Oops" message.

I installed using: `sudo dnf install nvidia-settings kernel-devel dkms-nvidia`. I also tried using `akmod-nvidia` instead with the same error.

I'm pretty clueless as to where to start digging in. The only thing I can think of right now is maybe related to this?

> That said there is a major catch here, due to the way things currently work once the NVidia binary driver is installed it expects to be rendering the screen all the time. So we need for the short term to figure out a way to allow it to do that, and in the long run we need to work with NVidia to figure out how the Intel open source driver can collaborate with the Nvidia driver to allow us to only use the Intel driver at times (to save on power for instance). We are still pushing hard on trying to have the short term fix in place, but as I write this we haven’t fully nailed this down yet.

Not sure if the Intel driver is conflicting or something. I am not too knowledgeable with debugging this stuff, but would love to help out in any way I can.

scaronni, 2016-11-07:
Nvidia drivers 375.10 do not yet suppport X.org 1.19, the driver installation is relying on the "IgnoreABI" directive in its configuration. Of course this does not give you any guarantee. Nvidia does not support unreleased X servers, so if you can not downgrade the X server to 1.18.x there's not much you can do.

On top of this, if your system is an Optimus one, there are a few caveats to take into account:

https://blogs.gnome.org/uraeus/2016/11/01/discrete-graphics-and-fedora-workstation-25/
http://negativo17.org/complex-setup-with-nvidia-optimus-nouveau-prime-on-fedora-20/

hworld, 2016-11-08:
Thanks for the info! Looks like I'll be waiting on nvidia, then.

closed 2016-11-08 by hworld