negativo17/nvidia-driver #5: Crash on Fedora 25 / GTX 960

ford-prefect, 2016-11-05:
I'm seeing the following crash after moving to Fedora 25:

```
[  1493.202] (II) NVIDIA(0): Setting mode "DFP-5:nvidia-auto-select"
[  1493.223] (EE) 
[  1493.223] (EE) Backtrace:
[  1493.224] (EE) 0: /usr/libexec/Xorg (OsLookupColor+0x139) [0x59e2d9]
[  1493.224] (EE) 1: /lib64/libpthread.so.0 (__restore_rt+0x0) [0x7f03434d15bf]
[  1493.224] (EE) 2: /lib64/libnvidia-glcore.so.375.10 (nvidiaAddDrawableHandler+0x54599e) [0x7f033e262a2c]
[  1493.224] (EE) 
[  1493.224] (EE) Segmentation fault at address 0x10
[  1493.224] (EE) 
Fatal server error:
[  1493.224] (EE) Caught signal 11 (Segmentation fault). Server aborting
[  1493.224] (EE) 
[  1493.224] (EE) 
```

scaronni, 2016-11-07:
Nvidia drivers 375.10 do not yet suppport X.org 1.19, the driver installation is relying on the "IgnoreABI" directive in its configuration. Of course this does not give you any guarantee. Nvidia does not support unreleased X servers, so if you can not downgrade the X server to 1.18.x there's not much you can do.

madig, 2016-11-12:
`sudo dnf --allowerasing --releasever=24 downgrade xorg-x11-server-Xorg` does the trick.

Conan-Kudo, 2016-11-18:
@scaronni nVidia driver 375.20 released today supports Xorg 1.19 ABI.

closed 2017-01-03 by scaronni