negativo17/nvidia-driver #7: Unnecessary path in ld.so.conf?

iavael, 2016-12-01:
Since libglvnd (with it's own proxy libGL and friends) is required anyway, why not remove these lines from specfile?

```sh
# Library search path
echo "%{_libdir}/nvidia" > %{buildroot}%{_sysconfdir}/ld.so.conf.d/nvidia-%{_lib}.conf
```

scaronni, 2016-12-02:
It's still required in `nvidia-driver-cuda-libs` for the OpenCL library:

https://github.com/negativo17/nvidia-driver/blob/master/nvidia-driver.spec#L422-L424

But in `nvidia-driver-libs` should not actually be there, it's a leftover:

https://github.com/negativo17/nvidia-driver/blob/master/nvidia-driver.spec#L406

I've removed it from the spec file, will trigger a new rebuild soon.

closed 2016-12-02 by scaronni

scaronni, 2016-12-02:
https://github.com/negativo17/nvidia-driver/commit/f2c5f01d4f2c05e512e79eb8a89f5ed77114d1f0