negativo17/nvidia-driver #8: Every login after the first fails on GTX 960

SriRamanujam, 2016-12-15:
Hi,

I'm using a GTX 960, and while I can login okay the first time after a reboot, every subsequent login fails to work and crashes back to the GDM login screen. The logs show nothing particularly interesting, but I've attached them anyway. 

[journalctl logs](https://github.com/negativo17/nvidia-driver/files/653587/out.txt)

[Xorg.0.log](https://github.com/negativo17/nvidia-driver/files/653584/Xorg.0.log.txt)

rhasti, 2017-05-14:
I have the same issue

egee-irl, 2017-11-14:
Yup same issue here

scaronni, 2019-02-06:
Please reopen if this is happening again. There has been too many changes since then.

Systems prior to Fedora 29 must select "X.org session" at login otherwise there are problems. Fedora 29 has added udev rules for always starting GDM with X, there is now separate packages for common kernel components, Mutter is no longer compiled with EGLStream, etc.

closed 2019-02-06 by scaronni