negativo17/nvidia-driver #9: Rsync access

jdieter, 2016-12-22:
I'm the sysadmin for a school running Fedora, and I'm trying to run a local mirror of your multimedia repository.  Is there any way you can add rsync access to your repository so it's easier to pull down?

scaronni, 2016-12-22:
Eh, my crappy hosting provider does not offer that. I'm planning to switch to Digital Ocean or a virtual private server on Amazon's AWS. It's on my todo list but I'm pretty busy at the moment and this has slipped further down the list of priorities.

But I kept the structure simple so people could use some spider mode of wget or similar tools.

jdieter, 2016-12-22:
I'm on Digital Ocean, and it's very convenient for stuff like this.

I had a wget script setup for mirroring, but it didn't do a great job of verifying the data and I ended up with half of a blender rpm.  

I'm currently using dnf repoquery, but it forces me to rebuild the metadata.  I'll just keep regenerating the metadata until you manage to get a DO or AWS server.

Thanks for the work you do on this.

scaronni, 2020-05-22:
Better late than never.... when moving away from GoDaddy I got a dedicated host with much better bandwidth / throughput: https://negativo17.org/back-online-thank-you/

Next thing to do is get rid of Wordpress and get something faster.

Read only `rsync` for every repository:

rsync -crlpvz --progress --delete-after rsync://negativo17.org:/repos/nvidia local/path

closed 2020-05-22 by scaronni

jdieter, 2020-05-23:
Hurrah!