negativo17/nvidia-driver #11: Apps and ldd not finding nvidia libraries when installed in /usr/lib/nvidia or /usr/lib64/nvidia

krutoileshii, 2017-01-07:
Ran into this with TVheadend and a couple others. LDD wouldn't locate them either so had to create the two filed under /etc/ld.so.conf.d/ with the path and run ldconfig to get it recognized.

scaronni, 2017-01-08:
There's a bit of confusion here, the drivers do not ship any library in `/usr/lib/nvidia` or `/usr/lib64/nvidia`, just the OpenCL override in the CUDA libraries subpackage. And for that, there's the necessary `ldconfig` configuration file.

```
$ rpm -ql nvidia-driver-libs.x86_64
/usr/lib64/libEGL_nvidia.so.0
/usr/lib64/libEGL_nvidia.so.375.26
/usr/lib64/libGLESv1_CM_nvidia.so.1
/usr/lib64/libGLESv1_CM_nvidia.so.375.26
/usr/lib64/libGLESv2_nvidia.so.2
/usr/lib64/libGLESv2_nvidia.so.375.26
/usr/lib64/libGLX_indirect.so.0
/usr/lib64/libGLX_nvidia.so.0
/usr/lib64/libGLX_nvidia.so.375.26
/usr/lib64/libnvidia-cfg.so.1
/usr/lib64/libnvidia-cfg.so.375.26
/usr/lib64/libnvidia-egl-wayland.so.375.26
/usr/lib64/libnvidia-eglcore.so.375.26
/usr/lib64/libnvidia-glcore.so.375.26
/usr/lib64/libnvidia-glsi.so.375.26
/usr/lib64/libnvidia-tls.so.375.26
/usr/lib64/vdpau/libvdpau_nvidia.so.1
/usr/lib64/vdpau/libvdpau_nvidia.so.375.26
/usr/share/glvnd/egl_vendor.d/10_nvidia.json
```

```
$ rpm -ql nvidia-driver-cuda-libs.x86_64
/etc/ld.so.conf.d/nvidia-lib64.conf
/usr/lib64/libcuda.so
/usr/lib64/libcuda.so.1
/usr/lib64/libcuda.so.375.26
/usr/lib64/libnvcuvid.so.1
/usr/lib64/libnvcuvid.so.375.26
/usr/lib64/libnvidia-compiler.so.375.26
/usr/lib64/libnvidia-encode.so.1
/usr/lib64/libnvidia-encode.so.375.26
/usr/lib64/libnvidia-fatbinaryloader.so.375.26
/usr/lib64/libnvidia-opencl.so.1
/usr/lib64/libnvidia-opencl.so.375.26
/usr/lib64/libnvidia-ptxjitcompiler.so.375.26
/usr/lib64/nvidia
/usr/lib64/nvidia/libOpenCL.so.1
/usr/lib64/nvidia/libOpenCL.so.1.0.0
```

What are you referring to?

closed 2017-01-08 by scaronni

krutoileshii, 2017-01-10:
nvidia-driver-cuda-libs-2:375.26-1.fc25.i686 : Libraries for nvidia-driver-cuda
Repo        : fedora-multimedia
 which provides libnvcuvid.so  could be install from either your handbrake or multimedia library. is the one i was having issues with.  Maybe i just installed it wrong.