negativo17/nvidia-driver #13: Enable nvidia-drm

samgurtman, 2017-01-20:
It appears like Fedora 25 is now working with Prime Synchronization. Anyway DRM and KMS support can be enabled so we can try it out?

I may also have misread the thread, so if that doesn't make sense than ignore.

https://devtalk.nvidia.com/default/topic/957814/linux/prime-and-prime-synchronization/7

scaronni, 2017-01-20:
Hello, everything should work out of the box, the KMS part (remember there is no framebuffer console on Nvidia's KMS) is only for running Wayland on Nvidia drivers; is not related to PRIME synchronization.

At the moment requires Mutter from Fedora 26.
Hans De Geode is working in bringing all components in Fedora:

http://hansdegoede.livejournal.com/

If you want to try Wayland yourself (Fedora 26), just copy the file `/usr/lib/modprobe.d/nvidia.conf` in `/etc/modprobe.d/nvidia.conf` with the appropriate editing and reboot.

closed 2017-01-20 by scaronni

samgurtman, 2017-01-20:
Ah, I must be confused. I was under the impression rpmfusion drivers
enabled prime synchronization on Fedora 25. When I tried to enable
nvidia-drm.modeset=1 on your version I had numerous errors that prevented
Gnome from booting. Is DRM not required for Prime Synchronization?

samgurtman, 2017-01-20:
According to Nvidia, KMS is infact required for Prime Synchronization.

"The NVIDIA driver’s PRIME Synchronization support relies on DRM-KMS, which is disabled by default due to its current incompatibility with SLI. To enable it, run ‘sudo rmmod nvidia-drm; sudo modprobe nvidia-drm modeset=1’. In other words, load the nvidia-drm module with the parameter modeset=1."

scaronni, 2017-01-20:
That's funny. It actually works (I also have a -useless in Linux- SLI system) also with SLI, but is actually not required for running X with PRIME synchronization, as depicted by Hans.

Have you tried the PRIME stuff without enabling it?

samgurtman, 2017-01-22:
I have tried enabling Prime Synchronization without it to no avail. I still have bad tearing. Checking the logs it says every time I try to enable it:
`/usr/libexec/gdm-x-session[1971]: randr: falling back to unsynchronized pixmap sharing`

The commit that added that error message states:
```
    Changes rrSetupPixmapSharing() to use double buffering and
    synchronization in the case that all required driver functions are
    available. Otherwise, falls back to unsynchronized single buffer.
```

Can we get an experimental version that has KMS-DRM enabled?

excieve, 2017-01-25:
I'm also experiencing kernel errors when launching Gnome3 using the `nvidia-drm.modeset=1` option in order to try PRIME Sync. Here's the traceback:
```
kernel: divide error: 0000 [#3] SMP
kernel: Modules linked in: cmac bnep binfmt_misc vfat fat hid_logitech_hidpp hid_logitech_dj uvcvideo btusb btrtl btbcm videobuf2_vmalloc btintel videobuf2_memops videobuf2_v4l2 bluetooth videobuf2_core videodev me
kernel:  industrialio lpc_ich tpm_tis tpm_tis_core mei_me mei shpchp soc_button_array tpm vboxpci(OE) vboxnetadp(OE) vboxnetflt(OE) nfsd vboxdrv(OE) auth_rpcgss nfs_acl lockd grace sunrpc rtsx_usb_sdmmc mmc_core hi
kernel: CPU: 4 PID: 2308 Comm: Xorg Tainted: P      D    OE   4.9.5-200.fc25.x86_64 #1
kernel: Hardware name: Acer Aspire VN7-591G/Aspire VN7-591G, BIOS V1.15 08/06/2015
kernel: task: ffff8c2118b7bd00 task.stack: ffffb3df44468000
kernel: RIP: 0010:[<ffffffffc081f718>]  [<ffffffffc081f718>] nvidia_drm_dumb_create+0x68/0x1b0 [nvidia_drm]
kernel: RSP: 0018:ffffb3df4446bd18  EFLAGS: 00010246
kernel: RAX: 0000000000000003 RBX: ffff8c213b3e1800 RCX: 0000000000001000
kernel: RDX: 0000000000000000 RSI: 0000000000000000 RDI: ffff8c2118bb1400
kernel: RBP: ffffb3df4446bd60 R08: 0000000000000001 R09: 0000000000000001
kernel: R10: ffff8c213b3e1800 R11: 00000000ffffffff R12: ffff8c2118bb1400
kernel: R13: ffff8c21491490c0 R14: ffffffffc0826000 R15: ffffb3df4446bdd0
kernel: FS:  00007f7c246eeec0(0000) GS:ffff8c215f300000(0000) knlGS:0000000000000000
kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
kernel: CR2: 00000000026e4028 CR3: 00000004492c2000 CR4: 00000000001406e0
kernel: Stack:
kernel:  0000000000000000 0000000000000000 0000000000000000 00000000ae50e949
kernel:  ffffb3df4446bdd0 0000000000000020 ffff8c2118bb1400 ffffffffc0353850
kernel:  ffff8c213b3e1800 ffffb3df4446bd70 ffffffffc033a286 ffffb3df4446be80
kernel: Call Trace:
kernel:  [<ffffffffc033a286>] drm_mode_create_dumb_ioctl+0xa6/0xc0 [drm]
kernel:  [<ffffffffc0330dcb>] drm_ioctl+0x21b/0x4c0 [drm]
kernel:  [<ffffffffc033a1e0>] ? drm_mode_crtc_set_obj_prop+0x60/0x60 [drm]
kernel:  [<ffffffffb226de03>] do_vfs_ioctl+0xa3/0x5f0
kernel:  [<ffffffffb226e3c9>] SyS_ioctl+0x79/0x90
kernel:  [<ffffffffb281cc77>] entry_SYSCALL_64_fastpath+0x1a/0xa9
kernel: Code: c7 45 c8 00 00 00 00 41 8b 75 50 48 c7 45 b8 00 00 00 00 48 c7 45 c0 00 00 00 00 83 c0 07 c1 e8 03 0f af 42 04 31 d2 8d 44 06 ff <f7> f6 0f af c6 41 89 47 14 41 0f af 07 48 8d b0 ff 0f 00 00 48 
kernel: RIP  [<ffffffffc081f718>] nvidia_drm_dumb_create+0x68/0x1b0 [nvidia_drm]
kernel:  RSP <ffffb3df4446bd18>
kernel: ---[ end trace 91d019155cb0fa00 ]---
```

This is with Fedora 25 and no SLI.

samgurtman, 2017-02-08:
Actually, it seems to work as is with latest updates! Thanks Negativo!

Gnork, 2017-05-20:
I just installed the nvidia-driver on my notebook with optimus (gtx 870m, fedora 25, kernel 4.10), but can't get it to work. The following happened:

1. If the nvidia-driver is not installed, the laptop just uses the Intel GPU as expected. `glxgears -info` confirms that. It also has a status LED, which is blue, indicating that the 870m is not turned on.
2. If I install nvidia-driver + nvidia-settings and reboot, the status LED turns red (also the fans are spinning loud), indicating that the 870m is turned on. But if I run `glxgears -info`, it shows that still the Intel GPU is used. Running `nvidia-settings` from the command line fails with `ERROR: Unable to find display on any available system`. (Never had that happen before on Ubuntu, where I was using optimus without any problems. It would automatically use the 870m if it is turned on and I could turn it off in nvidia-settings)
3. I tried copying the nvidia.conf to /etc/modprobe.d, uncommenting the line `options nvidia-drm modeset=1` and reboot. Nothing changed compared to 2.

Does anyone know what could be going on? Please let me know if you need additional information.

junrrein, 2017-06-30:
Hi, I'm hitting the same issue as @excieve and @samgurtman did.

On one hand, I can't activate PRIME sinchronization, hitting the same error outlined in [one of the comments](https://github.com/negativo17/nvidia-driver/issues/13#issuecomment-274360519).

On the other hand, I also can't boot with the `nvidia-drm.modeset=1` option in the kernel command line, with the same error described at [another comment](https://github.com/negativo17/nvidia-driver/issues/13#issuecomment-275109872).

Any pointers?

leigh123linux, 2017-06-30:
BORINGGGGGGGGGGG SELF INFLICTED ISSUE!

excieve, 2017-07-01:
@junrrein At some point enabling modeset and [running X with root privileges](https://devtalk.nvidia.com/default/topic/957814/linux/prime-and-prime-synchronization/post/5040106/#5040106) made PRIME Sync work for me. After some driver update it broke again though and I'm not using proprietary driver on my Optimus laptop anymore so not sure if it would work now.

junrrein, 2017-07-01:
@excieve Thanks for the tip, that worked!

For future reference:
kernel 4.11.7
nvidia-driver 381.22