negativo17/nvidia-driver #14: Add support for PRIME Synchronisation

excieve, 2017-01-28:
According to [this NVidia forum thread ](https://devtalk.nvidia.com/default/topic/957814/linux/prime-and-prime-synchronization/1) PRIME Sync requires KMS support enabled. I'm aware there's a way to do this with uncommenting an option in `modprobe.d` config. However, this ends up in a kernel error when launching Gnome3 (liekly others too, didn't test). [Apparently](https://devtalk.nvidia.com/default/topic/957814/linux/prime-and-prime-synchronization/post/5039885/#5039885) this also requires Xorg to be launched in root mode unlike the default rootless mode. This can be accomplished using an `/etc/X11/Xwrapper.config` with the following line in it: `needs_root_rights=yes`. With this and KMS enabled PRIME Sync works fine.

Please consider adding this to the package.

excieve, 2017-02-16:
@samgurtman Which updates do you mean? I have just upgraded to 378.13 on F25 and I can't make PRIME Sync work at all, moreover even without PRIME Sync I get GPU falling off the bus and system freezes after about 5 minutes of usage...

excieve, 2017-02-16:
Tried reinstalling the driver, disabling threaded optimizations (that seems to be the only major change in this release), various other things but the driver just freezes eventually.

@scaronni Could you please share the previous stable packages to try a downgrade? They don't seem to be available in the repository anymore.

scaronni, 2017-08-22:
Please see https://github.com/negativo17/nvidia-driver/issues/27

closed 2017-08-22 by scaronni