negativo17/nvidia-driver #15: [Question] Using negativo17 repo instead of nvidia proprietary driver

didli, 2017-02-09:
Hi ! As mentionned in the title, it's not a bug/issue report, but a question : 
I'm currently using nvidia proprietary driver 378.09 beta in Fedora 25 x64, with cuda 8.0.26 (.run) and blender 2.78b (standalone).
Will I be still able to use Cuda 8 if I switch to negativo17 repo instead of the proprietary driver ? Could you please point me toward the best way to do so, if it is even possible ?
Thank you in advance ! (and thank you for the fedora repo, by the way)

scaronni, 2017-02-12:
Hello, yes you will, but there are some caveats. You need to use this repository:

http://negativo17.org/HandBrake
http://negativo17.org/repos/multimedia

Which contains also all the Nvidia libraries and tools, so you don't need to add them:

http://negativo17.org/nvidia
http://negativo17.org/repos/nvidia

Then you can do the following, assuming you have NO Nvidia stuff installed:

`dnf install kernel-devel dkms-nvidia nvidia-driver-cuda nvidia-driver-libs.i686 blender-cuda && reboot`

This will install the Nvidia driver with the DKMS module variant, the 32 bit libs for playing Steam games and a Blender compiled with FFmpeg and CUDA support.

If you also want CUDA 8.0 (same packages that have built the above Blender):

`dnf install cuda-devel`

Headers are in `/usr/include/cuda` and libraries and tools are in the normal system path.

Just try in a VM if it satisfies you. Keep in mind that the multimedia repository is NOT compatible with RPMFusion.

didli, 2017-02-12:
hmm... i have plenty of nvidia stuff, and i ... think the RPMFusion is mandatory at the moment for me (i need to look carefully into this). Still, your answer was amazingly precise and very helpful (answer saved on my docs for further use !), so thank you a lot !
Keep up the (very) good work !

closed 2017-02-12 by didli

scaronni, 2017-02-13:
May I ask you what is "plenty of Nvidia stuff"? Just curious.
In that repository there are CUDA/NVENC enabled versions of:

- FFMpeg
- MPV
- Avidemux
- Blender
- "Granular" Nvidia drivers

My plan is NOT to support every package out there, so for example you need Game console emulators this is not the place to look for. You might want to browse the repository:

http://negativo17.org/repos/multimedia/fedora-25/x86_64/

And look at the various SPEC files here:

https://github.com/negativo17

didli, 2017-02-17:
First things first, sorry for the late reply. What I mean by 'plenty of nvidia stuff' is that I have a lot of related configurations files that will probably mess up a lot of things if I change things as of now. Or at least I will have to look carefully into this. For the RPMFusion, it's related to 3d printing stuff. I can't get some apps to work (even by compiling) unless I use the packages in the repo. I also prefer to use Blender as a standalone app than installing it from a repo, but that's not important. Anyway, thank you again for all your infos. 
I will make sure to use your repos on my next fresh install (some time ago I had already 'advertise it' on g+ and tested it with success).