negativo17/nvidia-driver #16: selinux-policy 3.13.1-225.10.fc25 prevents loading nvidia.ko

hit0ri, 2017-02-26:
After appying today's morning system updates I'm no longer able to boot into graphical target. Looks lke the new selinux policies are preventing loading nvidia driver module.
As a temporary workaround I booted into multi-user mode and disabled selinux.

Here's  an audit log:
```
Feb 26 13:10:52 rain audit[520]: AVC avc:  denied  { module_load } for  pid=520 comm="systemd-udevd" path="/usr/lib/modules/4.9.11-200.fc25.x86_64/extra/nvidia.ko" dev="sda2" ino=75793620 scontext=system_u:system_r:udev_t:s0-s0:c0.c1023 tcontext=unconfined_u:object_r:modules_object_t:s0 tclass=system permissive=0
```

List of applied updates:
```
Package                           Arch      Version                           Repository   Size
=================================================================================================
Installing:
kernel                            x86_64    4.9.11-200.fc25                   updates      95 k
kernel-core                       x86_64    4.9.11-200.fc25                   updates      20 M
kernel-devel                      x86_64    4.9.11-200.fc25                   updates      11 M
kernel-modules                    x86_64    4.9.11-200.fc25                   updates      22 M
kernel-modules-extra              x86_64    4.9.11-200.fc25                   updates     2.2 M
Upgrading:
crypto-policies                   noarch    20160921-4.gitf3018dd.fc25        updates      33 k
flatpak                           x86_64    0.8.3-3.fc25                      updates     692 k
flatpak-libs                      x86_64    0.8.3-3.fc25                      updates     214 k
grep                              x86_64    2.27-2.fc25                       updates     283 k
ibus-typing-booster               noarch    1.5.22-1.fc25                     updates     3.8 M
kernel-headers                    x86_64    4.9.11-200.fc25                   updates     1.1 M
libblkid                          i686      2.28.2-2.fc25                     updates     186 k
libblkid                          x86_64    2.28.2-2.fc25                     updates     180 k
libfdisk                          x86_64    2.28.2-2.fc25                     updates     221 k
libglvnd                          i686      1:0.2.999-10.gitdc16f8c.fc25      updates      83 k
libglvnd                          x86_64    1:0.2.999-10.gitdc16f8c.fc25      updates      89 k
libglvnd-egl                      i686      1:0.2.999-10.gitdc16f8c.fc25      updates      44 k
libglvnd-egl                      x86_64    1:0.2.999-10.gitdc16f8c.fc25      updates      43 k
libglvnd-gles                     i686      1:0.2.999-10.gitdc16f8c.fc25      updates      31 k
libglvnd-gles                     x86_64    1:0.2.999-10.gitdc16f8c.fc25      updates      31 k
libglvnd-glx                      i686      1:0.2.999-10.gitdc16f8c.fc25      updates     118 k
libglvnd-glx                      x86_64    1:0.2.999-10.gitdc16f8c.fc25      updates     124 k
libglvnd-opengl                   i686      1:0.2.999-10.gitdc16f8c.fc25      updates      41 k
libglvnd-opengl                   x86_64    1:0.2.999-10.gitdc16f8c.fc25      updates      43 k
libmount                          i686      2.28.2-2.fc25                     updates     200 k
libmount                          x86_64    2.28.2-2.fc25                     updates     198 k
libsmartcols                      x86_64    2.28.2-2.fc25                     updates     141 k
libuuid                           i686      2.28.2-2.fc25                     updates      80 k
libuuid                           x86_64    2.28.2-2.fc25                     updates      80 k
libva                             i686      1.7.3-3.fc25                      updates      90 k
libva                             x86_64    1.7.3-3.fc25                      updates      89 k
mesa-dri-drivers                  i686      13.0.4-1.fc25                     updates      12 M
mesa-dri-drivers                  x86_64    13.0.4-1.fc25                     updates      11 M
mesa-filesystem                   i686      13.0.4-1.fc25                     updates      24 k
mesa-filesystem                   x86_64    13.0.4-1.fc25                     updates      24 k
mesa-libEGL                       i686      13.0.4-1.fc25                     updates     105 k
mesa-libEGL                       x86_64    13.0.4-1.fc25                     updates     103 k
mesa-libGL                        i686      13.0.4-1.fc25                     updates     182 k
mesa-libGL                        x86_64    13.0.4-1.fc25                     updates     165 k
mesa-libGLES                      x86_64    13.0.4-1.fc25                     updates      22 k
mesa-libOSMesa                    i686      13.0.4-1.fc25                     updates     2.0 M
mesa-libOSMesa                    x86_64    13.0.4-1.fc25                     updates     1.9 M
mesa-libgbm                       i686      13.0.4-1.fc25                     updates      44 k
mesa-libgbm                       x86_64    13.0.4-1.fc25                     updates      42 k
mesa-libglapi                     i686      13.0.4-1.fc25                     updates      60 k
mesa-libglapi                     x86_64    13.0.4-1.fc25                     updates      50 k
mesa-libwayland-egl               i686      13.0.4-1.fc25                     updates      25 k
mesa-libwayland-egl               x86_64    13.0.4-1.fc25                     updates      25 k
mesa-libxatracker                 x86_64    13.0.4-1.fc25                     updates     1.4 M
mesa-vulkan-drivers               x86_64    13.0.4-1.fc25                     updates     697 k
pcre                              i686      8.40-4.fc25                       updates     196 k
pcre                              x86_64    8.40-4.fc25                       updates     199 k
pcre-utf16                        x86_64    8.40-4.fc25                       updates     189 k
perf                              x86_64    4.9.11-200.fc25                   updates     1.5 M
perl-Module-CoreList              noarch    1:5.20170220-1.fc25               updates      78 k
perl-Net-HTTP                     noarch    6.13-1.fc25                       updates      41 k
selinux-policy                    noarch    3.13.1-225.10.fc25                updates     487 k
selinux-policy-targeted           noarch    3.13.1-225.10.fc25                updates     8.7 M
sos                               noarch    3.3-1.fc25                        updates     397 k
util-linux                        x86_64    2.28.2-2.fc25                     updates     2.2 M
util-linux-user                   x86_64    2.28.2-2.fc25                     updates      89 k
vim-minimal                       x86_64    2:8.0.347-2.fc25                  updates     516 k
webkitgtk4                        x86_64    2.14.5-1.fc25                     updates      12 M
webkitgtk4-jsc                    x86_64    2.14.5-1.fc25                     updates     4.0 M
webkitgtk4-plugin-process-gtk2    x86_64    2.14.5-1.fc25                     updates     8.9 M
Removing:
kernel                            x86_64    4.9.9-200.fc25                    @updates      0  
kernel-core                       x86_64    4.9.9-200.fc25                    @updates     53 M
kernel-devel                      x86_64    4.9.9-200.fc25                    @updates     41 M
kernel-modules                    x86_64    4.9.9-200.fc25                    @updates     22 M
kernel-modules-extra              x86_64    4.9.9-200.fc25                    @updates    2.0 M

Transaction Summary
=================================================================================================
Install   5 Packages
Upgrade  60 Packages
Remove    5 Packages
```

scaronni, 2017-02-27:
Same here, I guess something changed in the policy or kernel side. Will look into it. Modprobe as root works.

scaronni, 2017-02-28:
This is fixed in the latest policy from updates-testing <code>selinux-policy-targeted-3.13.1-225.11.fc25.noarch</code>. Update and reboot.

closed 2017-02-28 by scaronni