negativo17/nvidia-driver #18: Doesn't build/work with latest 4.10.1 kernel

welbers, 2017-03-03:
dkms install nvidia/375.39 -k 4.10.1-1.el7.elrepo.x86_64

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j2 module SYSSRC=/lib/modules/4.10.1-1.el7.elrepo.x86_64/build IGNORE_XEN_PRESENCE=1 IGNORE_PREEMPT_RT_PRESENCE=1......(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.10.1-1.el7.elrepo.x86_64 (x86_64)
Consult /var/lib/dkms/nvidia/375.39/build/make.log for more information.

welbers, 2017-03-03:
Here the relevant output from make.log:

CONFTEST: drm_atomic_modeset_nonblocking_commit_available
  CC [M]  /var/lib/dkms/nvidia/375.39/build/nvidia/nv-frontend.o
  CC [M]  /var/lib/dkms/nvidia/375.39/build/nvidia/nv-instance.o
  CC [M]  /var/lib/dkms/nvidia/375.39/build/nvidia/nv.o
  CC [M]  /var/lib/dkms/nvidia/375.39/build/nvidia/nv-acpi.o
  CC [M]  /var/lib/dkms/nvidia/375.39/build/nvidia/nv-chrdev.o
  CC [M]  /var/lib/dkms/nvidia/375.39/build/nvidia/nv-cray.o
  CC [M]  /var/lib/dkms/nvidia/375.39/build/nvidia/nv-dma.o
  CC [M]  /var/lib/dkms/nvidia/375.39/build/nvidia/nv-gvi.o
  CC [M]  /var/lib/dkms/nvidia/375.39/build/nvidia/nv-i2c.o
  CC [M]  /var/lib/dkms/nvidia/375.39/build/nvidia/nv-mempool.o
  CC [M]  /var/lib/dkms/nvidia/375.39/build/nvidia/nv-mmap.o
  CC [M]  /var/lib/dkms/nvidia/375.39/build/nvidia/nv-p2p.o
  CC [M]  /var/lib/dkms/nvidia/375.39/build/nvidia/nv-pat.o
  CC [M]  /var/lib/dkms/nvidia/375.39/build/nvidia/nv-procfs.o
/var/lib/dkms/nvidia/375.39/build/nvidia/nv-pat.c: In Funktion »nvidia_cpu_callback«:
/var/lib/dkms/nvidia/375.39/build/nvidia/nv-pat.c:213:14: Fehler: »CPU_DOWN_FAILED« nicht deklariert (erste Benutzung in dieser Funktion)
         case CPU_DOWN_FAILED:
              ^
/var/lib/dkms/nvidia/375.39/build/nvidia/nv-pat.c:213:14: Anmerkung: jeder nicht deklarierte Bezeichner wird nur einmal für jede Funktion, in der er vorkommt, gemeldet
/var/lib/dkms/nvidia/375.39/build/nvidia/nv-pat.c:220:14: Fehler: »CPU_DOWN_PREPARE« nicht deklariert (erste Benutzung in dieser Funktion)
         case CPU_DOWN_PREPARE:
              ^
/var/lib/dkms/nvidia/375.39/build/nvidia/nv-pat.c: In Funktion »nv_init_pat_support«:
/var/lib/dkms/nvidia/375.39/build/nvidia/nv-pat.c:258:13: Fehler: Implizite Deklaration der Funktion »register_cpu_notifier« [-Werror=implicit-function-declaration]
             if (register_hotcpu_notifier(&nv_hotcpu_nfb) != 0)
             ^
/var/lib/dkms/nvidia/375.39/build/nvidia/nv-pat.c: In Funktion »nv_teardown_pat_support«:
/var/lib/dkms/nvidia/375.39/build/nvidia/nv-pat.c:283:9: Fehler: Implizite Deklaration der Funktion »unregister_cpu_notifier« [-Werror=implicit-function-declaration]
         unregister_hotcpu_notifier(&nv_hotcpu_nfb);
         ^
cc1: Einige Warnungen werden als Fehler behandelt
make[2]: *** [/var/lib/dkms/nvidia/375.39/build/nvidia/nv-pat.o] Fehler 1
make[2]: *** Warte auf noch nicht beendete Prozesse...
make[1]: *** [_module_/var/lib/dkms/nvidia/375.39/build] Fehler 2
make[1]: Leaving directory `/usr/src/kernels/4.10.1-1.el7.elrepo.x86_64'
make: *** [modules] Fehler 2

scaronni, 2017-03-03:
I've added the patch to the DKMS package for EPEL 7 and 6. It will be online soon.

But honestly, if you're swapping the kernel away what's the point of using CentOS/RHEL then?

closed 2017-03-03 by scaronni

welbers, 2017-03-03:
All our systems are running CentOS/RHEL normally with the distribution's kernel.
But we have a few systems which need for some special development cases the latest available kernel. That's the reason ;-)

scaronni, 2017-03-03:
Ok :)
New build uploaded.