negativo17/nvidia-driver #19: GeForce GTX1070 display issues

tla2k12, 2017-03-21:
Hi. First of all thanks for the great work.
I am running FC26. With the nouveau drivers I see upto 800% cpu utilisation from gnome-shell. I don't know but guess this is because GNOME is doing software rendering (also both mouse buttons seem to be a right button!)? Thought I would try your nvidia support which in terms of CPU usage is great - back to 0%. However, the max screen resolution is 1024x768 (monitor is capable of 2560*1600 16:10) and about 1/8 of the screen is repeated on the right hand side.
I guess this is probably a driver issue but I have had this card working under FC25 with the nvidia drivers from rpmfusion (nouveau never worked properly). An FC25 update broke the nvidia support hence I tried FC26 which at least partially works.
Are you interested in anything I can provide debug wise to help identify the issue? Happy to send any logs / try things out etc if it helps.

tla2k12, 2017-03-22:
This new package conflict appeared today:

Error: Transaction check error:
  file /usr/lib64/libGLX_indirect.so.0 from install of mesa-libGL-17.0.2-1.fc26.x86_64 conflicts with file from package nvidia-driver-libs-2:378.13-3.fc26.x86_64
  file /usr/lib/libGLX_indirect.so.0 from install of mesa-libGL-17.0.2-1.fc26.i686 conflicts with file from package nvidia-driver-libs-2:378.13-3.fc26.i686

scaronni, 2017-03-22:
Should be fixed in the latest update that I just pushed.

Regarding your problem with the desktop, you should see if everything is loaded. Check that the `nvidia` modules are loaded, if not check for errors in the build directories, look for X.org messages with `journalctl`, make sure `nouveau` is not loaded, etc.

tla2k12, 2017-03-23:
Hi,

> Should be fixed in the latest update that I just pushed.
> Regarding your problem with the desktop, you should see if everything is loaded. Check that the nvidia modules are loaded, if not check for [...]

Just tried the latest update:

$ sudo dnf clean packages
114 files removed
$

Notice below:
Error! Bad return status for module build on kernel: 4.11.0-0.rc3.git0.1.fc26.x86_64 (x86_64)
Consult /var/lib/dkms/nvidia/378.13/build/make.log for more information.

Log attached after the dnf upgrade output below.

Also the nvidia and nouveau modules are not loaded:
$ lsmod | grep -iE 'nvidia|nouveau'
$

$ sudo dnf upgrade --refresh
negativo17 - Nvidia                                                                                 1.2 MB/s |  83 kB     00:00
Last metadata expiration check: 0:00:00 ago on Thu Mar 23 10:28:55 2017 GMT.
Dependencies resolved.
====================================================================================================================================
 Package                                       Arch          Version                           Repository                      Size
====================================================================================================================================
Installing:
 kernel                                        x86_64        4.11.0-0.rc3.git0.1.fc26          updates-testing                 52 k
 kernel-core                                   x86_64        4.11.0-0.rc3.git0.1.fc26          updates-testing                 20 M
 kernel-modules                                x86_64        4.11.0-0.rc3.git0.1.fc26          updates-testing                 23 M
 kernel-modules-extra                          x86_64        4.11.0-0.rc3.git0.1.fc26          updates-testing                2.1 M
Upgrading:
 alsa-lib                                      i686          1.1.3-3.fc26                      updates-testing                436 k
 alsa-lib                                      x86_64        1.1.3-3.fc26                      updates-testing                417 k
 autocorr-en                                   noarch        1:5.3.2.1-1.fc26                  updates-testing                204 k
 binutils                                      x86_64        2.27-21.fc26                      updates-testing                5.6 M
 dnf                                           noarch        2.1.1-1.1.fc26                    updates-testing                289 k
 dnf-automatic                                 noarch        2.1.1-1.1.fc26                    updates-testing                 49 k
 dnf-conf                                      noarch        2.1.1-1.1.fc26                    updates-testing                 52 k
 dnf-plugins-core                              noarch        1.1.0-1.fc26                      updates-testing                 39 k
 dnf-yum                                       noarch        2.1.1-1.1.fc26                    updates-testing                 34 k
 empathy                                       x86_64        3.12.13-2.fc26                    updates-testing                4.1 M
 git                                           x86_64        2.12.1-1.fc26                     updates-testing                1.1 M
 git-core                                      x86_64        2.12.1-1.fc26                     updates-testing                4.0 M
 git-core-doc                                  x86_64        2.12.1-1.fc26                     updates-testing                2.3 M
 gnome-autoar                                  x86_64        0.2.2-1.fc26                      updates-testing                 49 k
 gnome-initial-setup                           x86_64        3.24.0-1.fc26                     updates-testing                1.3 M
 ibus-typing-booster                           noarch        1.5.29-1.fc26                     updates-testing                432 k
 kernel-headers                                x86_64        4.11.0-0.rc3.git0.1.fc26          updates-testing                1.1 M
 libdnf                                        x86_64        0.8.0-1.fc26                      updates-testing                120 k
 libreoffice-calc                              x86_64        1:5.3.2.1-1.fc26                  updates-testing                9.1 M
 libreoffice-core                              x86_64        1:5.3.2.1-1.fc26                  updates-testing                 73 M
 libreoffice-data                              noarch        1:5.3.2.1-1.fc26                  updates-testing                3.4 M
 libreoffice-draw                              x86_64        1:5.3.2.1-1.fc26                  updates-testing                693 k
 libreoffice-filters                           x86_64        1:5.3.2.1-1.fc26                  updates-testing                 74 k
 libreoffice-graphicfilter                     x86_64        1:5.3.2.1-1.fc26                  updates-testing                472 k
 libreoffice-gtk2                              x86_64        1:5.3.2.1-1.fc26                  updates-testing                301 k
 libreoffice-gtk3                              x86_64        1:5.3.2.1-1.fc26                  updates-testing                312 k
 libreoffice-impress                           x86_64        1:5.3.2.1-1.fc26                  updates-testing                1.3 M
[... cut ...]
  Cleanup     : system-python-libs-3.6.0-21.fc26.x86_64                                                                     213/228
  Cleanup     : mesa-libwayland-egl-17.0.1-2.fc26.x86_64                                                                    214/228
  Cleanup     : webkitgtk4-jsc-2.15.92-1.fc26.x86_64                                                                        215/228
  Cleanup     : alsa-lib-1.1.3-2.fc26.x86_64                                                                                216/228
  Cleanup     : pcre2-10

scaronni, 2017-03-23:
That's a separate thing, Nvidia drivers do not support yet kernel 4.11-rc*:

https://devtalk.nvidia.com/default/topic/995636/-patches-378-13-4-10-and-4-11-rc2/?offset=12

I will add the patch soon.

tla2k12, 2017-03-30:
Hi,

>Четверг, 23 марта 2017, 13:47 +01:00 от Simone Caronni <notifications@github.com>:
>
>That's a separate thing, Nvidia drivers do not support yet kernel 4.11-rc*:
>https://devtalk.nvidia.com/default/topic/995636/-patches-378-13-4-10-and-4-11-rc2/?offset=12
>I will add the patch soon. I noticed there was an nvidia driver and kernel update (to rc4) this morning.  No change in terms of errors:

$ cat /var/lib/dkms/nvidia/378.13/build/make.log
DKMS make.log for nvidia-378.13 for kernel 4.11.0-0.rc4.git0.1.fc26.x86_64 (x86_64)
Thu 30 Mar 08:46:16 BST 2017
make "CC=cc"  KBUILD_VERBOSE= -C /lib/modules/4.11.0-0.rc4.git0.1.fc26.x86_64/build M=/var/lib/dkms/nvidia/378.13/build ARCH=x86_64 NV_KERNEL_SOURCES=/lib/modules/4.11.0-0.rc4.git0.1.fc26.x86_64/build NV_KERNEL_OUTPUT=/lib/modules/4.11.0-0.rc4.git0.1.fc26.x86_64/build NV_KERNEL_MODULES="nvidia nvidia-uvm nvidia-modeset nvidia-drm" INSTALL_MOD_DIR=kernel/drivers/video modules
make[1]: Entering directory '/usr/src/kernels/4.11.0-0.rc4.git0.1.fc26.x86_64'
Makefile:923: "Cannot use CONFIG_STACK_VALIDATION, please install libelf-dev, libelf-devel or elfutils-libelf-devel"
  SYMLINK /var/lib/dkms/nvidia/378.13/build/nvidia/nv-kernel.o
  SYMLINK /var/lib/dkms/nvidia/378.13/build/nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: INIT_WORK
 CONFTEST: remap_pfn_range
 CONFTEST: follow_pfn
 CONFTEST: vmap
 CONFTEST: set_pages_uc
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: change_page_attr
 CONFTEST: pci_get_class
 CONFTEST: pci_choose_state
 CONFTEST: vm_insert_page
 CONFTEST: acpi_device_id
 CONFTEST: acquire_console_sem
 CONFTEST: console_lock
 CONFTEST: kmem_cache_create
 CONFTEST: on_each_cpu
 CONFTEST: smp_call_function
 CONFTEST: acpi_evaluate_integer
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: acpi_walk_namespace
 CONFTEST: pci_domain_nr
 CONFTEST: pci_dma_mapping_error
 CONFTEST: sg_alloc_table
 CONFTEST: sg_init_table
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: efi_enabled
 CONFTEST: proc_create_data
 CONFTEST: pde_data
 CONFTEST: proc_remove
 CONFTEST: pm_vt_switch_required
 CONFTEST: drm_driver_has_set_busid
 CONFTEST: drm_driver_has_gem_prime_res_obj
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: write_cr4
 CONFTEST: of_parse_phandle
 CONFTEST: for_each_online_node
 CONFTEST: node_end_pfn
 CONFTEST: pci_bus_address
 CONFTEST: pci_stop_and_remove_bus_device
 CONFTEST: pci_remove_bus_device
 CONFTEST: request_threaded_irq
 CONFTEST: remap_page_range
 CONFTEST: address_space_init_once
 CONFTEST: kbasename
 CONFTEST: fatal_signal_pending
 CONFTEST: list_cut_position
 CONFTEST: vzalloc
 CONFTEST: wait_on_bit_lock_argument_count
 CONFTEST: bitmap_clear
 CONFTEST: usleep_range
[... 118 lines cut ...]
                 from /var/lib/dkms/nvidia/378.13/build/nvidia-uvm/uvm_linux.h:39,
                 from /var/lib/dkms/nvidia/378.13/build/nvidia-uvm/uvm_common.h:62,
                 from /var/lib/dkms/nvidia/378.13/build/nvidia-uvm/uvm_common.c:29:
/var/lib/dkms/nvidia/378.13/build/nvidia-uvm/uvm_linux.h: In function ‘__fatal_signal_pending’:
/var/lib/dkms/nvidia/378.13/build/nvidia-uvm/uvm_linux.h:212:25: error: implicit declaration of function ‘sigismember’ [-Werror=implicit-function-declaration]
         return unlikely(sigismember(&p->pending.signal, SIGKILL));
                         ^
./include/linux/compiler.h:179:42: note: in definition of macro ‘unlikely’
 # define unlikely(x) __builtin_expect(!!(x), 0)
                                          ^
In file included from /var/lib/dkms/nvidia/378.13/build/nvidia-uvm/uvm_common.h:62:0,
                 from /var/lib/dkms/nvidia/378.13/build/nvidia-uvm/uvm_common.c:29:
/var/lib/dkms/nvidia/378.13/build/nvidia-uvm/uvm_linux.h: In function ‘fatal_signal_pending’:
/var/lib/dkms/nvidia/378.13/build/nvidia-uvm/uvm_linux.h:217:16: error: implicit declaration of function ‘signal_pending’; did you mean ‘timer_pending’? [-Werror=implicit-function-declaration]
         return signal_pending(p) && __fatal_signal_pending(p);
                ^~~~~~~~~~~~~~
                timer_pending
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:294: /var/lib/dkms/nvidia/378.13/build/nvidia-uvm/uvm_utils.o] Error 1
make[2]: *** Waiting for unfinished jobs....
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:294: /var/lib/dkms/nvidia/378.13/build/nvidia-uvm/uvm_common.o] Error 1
make[1]: *** [Makefile:1496: _module_/var/lib/dkms/nvidia/378.13/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.11.0-0.rc4.git0.1.fc26.x86_64'
make: *** [Makefile:81: modules] Error 2

leigh123linux, 2017-03-30:
@scaronni 

I been using this patch for 4.11 

https://leigh123linux.fedorapeople.org/pub/patches/nvidia/kernel_4.11.patch

leigh123linux, 2017-03-30:
@tla2k12 You haven't applied the patch needed for 4.11 kernel

tla2k12, 2017-03-30:
On Thu, 2017-03-30 at 01:48 -0700, leigh123linux wrote:
> @scaronni
> I been using this patch for 4.11
> https://leigh123linux.fedorapeople.org/pub/patches/nvidia/kernel_4.11
> .patch

I patched the build directory which created the kernel modules. I then
installed with "sudo dkms autoinstall". Never used dkms before so
probably not the optimal approach but the nvidia driver is loaded now
and seems fine. Thanks for your help.

scaronni, 2017-03-31:
@leigh123linux thanks for the patch, I was a bit busy and did not apply it yet.
I hope the review patches for RPMfusion will go well and will not take forever :/

kuboosoft, 2017-04-12:
dkms has a mature place with the kernel; I don't like the akmod...

tla2k12, 2017-04-12:
@leigh123linux Do you have a patch update for 4.11-rc5? Since the kernel and driver update (381.09) the patch doesnt apply and the build fails. Here is the build log without any patch applied:

  CC [M]  /var/lib/dkms/nvidia/381.09/build/nvidia-drm/nvidia-drm-drv.o
/var/lib/dkms/nvidia/381.09/build/nvidia-drm/nvidia-drm-drv.c: In function ‘nvidia_drm_get_client_capability’:
/var/lib/dkms/nvidia/381.09/build/nvidia-drm/nvidia-drm-drv.c:545:31: warning: unused variable ‘nv_dev’ [-Wunused-variable]
     struct nvidia_drm_device *nv_dev = dev->dev_private;
                               ^~~~~~
/var/lib/dkms/nvidia/381.09/build/nvidia-drm/nvidia-drm-drv.c: At top level:
/var/lib/dkms/nvidia/381.09/build/nvidia-drm/nvidia-drm-drv.c:682:31: error: initialization from incompatible pointer type [-Werror=incompatible-pointer-types]
     .unload                 = nvidia_drm_unload,
                               ^~~~~~~~~~~~~~~~~
/var/lib/dkms/nvidia/381.09/build/nvidia-drm/nvidia-drm-drv.c:682:31: note: (near initialization for ‘nv_drm_driver.unload’)
  CC [M]  /var/lib/dkms/nvidia/381.09/build/nvidia-drm/nvidia-drm-utils.o
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:294: /var/lib/dkms/nvidia/381.09/build/nvidia-drm/nvidia-drm-drv.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:1496: _module_/var/lib/dkms/nvidia/381.09/build] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.11.0-0.rc5.git0.1.fc26.x86_64'
make: *** [Makefile:81: modules] Error 2

Here's the patch application failure:
$ sudo patch -p2 < ~/Downloads/kernel_4.11.patch
patching file nvidia-uvm/uvm8_gpu_isr.c
Hunk #1 FAILED at 25.
Hunk #2 succeeded at 466 with fuzz 2 (offset 196 lines).
1 out of 2 hunks FAILED -- saving rejects to file nvidia-uvm/uvm8_gpu_isr.c.rej
patching file nvidia-uvm/uvm_linux.h
patching file nvidia-drm/nvidia-drm-drv.c
patching file nvidia-drm/nvidia-drm-fb.c
Hunk #2 FAILED at 153.
1 out of 2 hunks FAILED -- saving rejects to file nvidia-drm/nvidia-drm-fb.c.rej
patching file nvidia-drm/nvidia-drm-encoder.h
Hunk #1 FAILED at 30.
1 out of 1 hunk FAILED -- saving rejects to file nvidia-drm/nvidia-drm-encoder.h.rej
patching file nvidia-drm/nvidia-drm-gem.c
patching file nvidia-drm/nvidia-drm-linux.c
patching file nvidia-uvm/uvm_common.c
patching file nvidia-modeset/nvidia-modeset-linux.c
Hunk #1 succeeded at 21 (offset -1 lines).
Hunk #2 succeeded at 1290 (offset 3 lines).

leigh123linux, 2017-04-12:
@tla2k12 

https://pkgs.rpmfusion.org/cgit/nonfree/nvidia-kmod.git/commit/?id=d4885077e73f7d43133d11ecf64ddfba7aff6dd5
https://pkgs.rpmfusion.org/cgit/nonfree/nvidia-kmod.git/plain/kernel_4.11.patch

scaronni, 2017-04-13:
@leigh123linux I thought it was not possible to ship a patch that just overwrites the license in RPMFusion, even if it's temporary...

leigh123linux, 2017-04-13:
@scaronni  4.11 kernel refuse to load MIT, the dual licence idea was from

https://lkml.org/lkml/2017/3/8/168

scaronni, 2017-04-13:
That's cool, good to know.

Gwend4l, 2017-04-14:
@leigh123linux Thanks for the link to the patch (https://pkgs.rpmfusion.org/cgit/nonfree/nvidia-kmod.git/plain/kernel_4.11.patch), I managed to install nvidia drivers on f26 thanks to it :)

tla2k12, 2017-04-14:
@leigh123linux Many thanks for the patches. If it is of any help to anyone, the following worked for me on fedora 26 / linux 4.11.0-0.rc6 / nvidia-381.09:

sudo wget -O /usr/src/kernel_4.11.0-0.rc6.patch https://pkgs.rpmfusion.org/cgit/nonfree/nvidia-kmod.git/plain/kernel_4.11.patch
cd /usr/src/nvidia-381.09
sudo patch -p2 < /usr/src/kernel_4.11.0-0.rc6.patch
sudo dkms add nvidia/381.09
sudo dkms autoinstall

scaronni, 2017-04-14:
Btw, I just pushed an update with the patch applied. Thanks to both!

closed 2017-04-14 by scaronni