negativo17/nvidia-driver #23: Nouveau being loaded - Optimus

drahnr, 2017-06-16:
This will probably sound like the one in a million nvidia driver install issues, but I think it is sligtly different:

lspci
```
02:00.0 3D controller: NVIDIA Corporation GK107GLM [Quadro K1100M] (rev a1)
00:02.0 VGA compatible controller: Intel Corporation 4th Gen Core Processor Integrated Graphics Controller (rev 06)
```

/etc/default/grub
```
GRUB_CMDLINE_LINUX="rd.driver.blacklist=nouveau rd.luks.uuid=luks-xxxxxxxxx"
```

/etc/modprobe.d/blacklist.conf
```
blacklist nouveau
```

/etc/X11/xorg.conf.d/10-nvidia.conf
```
Section "OutputClass"
    Identifier "nvidia"
    MatchDriver "nvidia-drm"
    Driver "nvidia"
    Option "AllowEmptyInitialConfiguration"
    Option "PrimaryGPU" "yes"
    Option "SLI" "Auto"
    Option "BaseMosaic" "on"
    ModulePath "/usr/lib64/nvidia/xorg"
EndSection

Section "OutputClass"
    Identifier "intel"
    MatchDriver "i915"
    Driver "modesetting"
EndSection

```

Yet the driver is not loaded, I will attach the Xorg log shortly.

drahnr, 2017-06-16:
Note that this might also explain various other, old/long-standing bugs which I found on the way.

scaronni, 2017-06-21:
I did not spot any errors in the log you posted. Did you by chance had previously installed drivers in some other ways? Can you rebuild the initrd with `dracut --force` and reboot the system?

Also, can you paste the output of the `systemd-modules-load.service` unit (`systemctl status systemd-modules-load.service`)?

drahnr, 2017-06-21:
The issue was that appearantly akmod skips the rebuild of modules which failed to build in the past.

So once I was logged in, `akmod --force` was all that was necessary to fix it.

Yes I had your repository installed but did not update in a while (3-4 months) and then got all kind of conflicts (as you described in your blogpost, and I followed the wildcard removal advice).

drahnr, 2017-06-21:
I am back on a running system now.

closed 2017-08-22 by scaronni