negativo17/nvidia-driver #24: nvidia-settings error on fedora 26

scompo, 2017-07-17:
I own a hp 15-r018nl laptop with dual NVIDIA GeForce 820M and Intel 4400.
It was having suspend and restore problems with the nuveau drivers so I installed the closed drivers following the instructions [here](https://negativo17.org/nvidia-driver/).
The laptop it's now restoring from suspend as expected.
I think it's using the intel card but when launching nvidia-settings from the command line i get the following error:
```
$ nvidia-settings 
ERROR: Unable to find display on any available system
```
This is the output of lspci
```
$ lspci | grep -E "VGA|3D"
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
09:00.0 3D controller: NVIDIA Corporation GF117M [GeForce 610M/710M/810M/820M / GT 620M/625M/630M/720M] (rev a1)
```
I can provide more details if needed.

hex007, 2017-08-02:
Same here for GTX 950m
```
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 530 (rev 06)
01:00.0 3D controller: NVIDIA Corporation GM107M [GeForce GTX 950M] (rev a2)
```

scaronni, 2017-08-02:
With `lspci -k` you can see which driver is driving what.

hex007, 2017-08-02:
The output is as such

Gnome Wayland session / Gnome Xorg session :: 
```
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 530 (rev 06)
	Subsystem: Hewlett-Packard Company Device 80a9
	Kernel driver in use: i915
	Kernel modules: i915

01:00.0 3D controller: NVIDIA Corporation GM107M [GeForce GTX 950M] (rev a2)
	Subsystem: Hewlett-Packard Company Device 80a9
	Kernel driver in use: nvidia
	Kernel modules: nouveau, nvidia_drm, nvidia
```

Gnome Settings > Details > Graphics :
-   Wayland > Intel® HD Graphics 530 (Skylake GT2)
-   Xorg > Nvidia 950m XXXXXXXXXXXXXXXXX

scompo, 2017-08-02:
This is my output from `lspci -k`:
```
00:02.0 VGA compatible controller: Intel Corporation Haswell-ULT Integrated Graphics Controller (rev 0b)
	Subsystem: Hewlett-Packard Company Device 2212
	Kernel driver in use: i915
	Kernel modules: i915
09:00.0 3D controller: NVIDIA Corporation GF117M [GeForce 610M/710M/810M/820M / GT 620M/625M/630M/720M] (rev a1)
	Subsystem: Hewlett-Packard Company Device 2212
	Kernel driver in use: nvidia
	Kernel modules: nouveau, nvidia_drm, nvidia
```

scaronni, 2017-08-22:
There is no Wayland support yet in Fedora 26 for Nvidia drivers: https://github.com/negativo17/nvidia-driver/issues/27

Are your laptops using X.org?

scompo, 2017-08-22:
No, Wayland from as a new fedora install.

scaronni, 2017-08-30:
Closing as there is no support for Wayland on Nvidia drivers in Fedora 26. See https://github.com/negativo17/nvidia-driver/issues/27.

closed 2017-08-30 by scaronni

wenpengfen, 2017-11-20:
hi~I'am glad I meet the same question with you.  haaa~Could you please tell me how solution this?

habi3000, 2018-01-14:
i am having the same problem with 960m 
`lspci -k` ouput 
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 530 (rev 06)
	Subsystem: ASUSTeK Computer Inc. HD Graphics 530
	Kernel driver in use: i915
	Kernel modules: i915
01:00.0 3D controller: NVIDIA Corporation GM107M [GeForce GTX 960M] (rev ff)
	Kernel modules: nvidia

hex007, 2018-01-14:
@habi3000  Are you running on Xorg or Wayland?

habi3000, 2018-01-14:
hi @hex007  i am running on wayland and Debian 9 , for some reason wayland is the only working option the reset just freezes after login

hex007, 2018-01-14:
There is no wayland support with nvidia AFAIK

leigh123linux, 2018-01-14:
@hex007 Your wrong!