negativo17/nvidia-driver #25: Blank screen on install

enaut, 2017-07-27:
Hey guys,

thanks for your work! - unfortunately it is not working for me :(

If I install the nvidia-driver package I get a blank black screen on startup instead of the login screen.

On a VT I can verify that the nvidia module is loaded. So the modules are built.

I tried adding some kernel and gruboptions like: gfxpayload=keep nomodeset rd.driver.blacklist=nouveau But none seems to work.

I now installed the binary driver according to: https://www.if-not-true-then-false.com/2015/fedora-nvidia-guide/#nvidia-install which works. - I'd much appreciate the autoupdating repository though!

what could be wrong what should I check? I'm very willing to try stuff to resolve the issue I'm out of Ideas though!

enaut, 2017-07-27:
Mhm I checked the journal for errors... X seems fine, but gnome shell seems to have troubles:

```
Jul 27 11:23:56 HomePC org.gnome.Shell.desktop[1272]: libGL error: No matching fbConfigs or visuals found
Jul 27 11:23:56 HomePC org.gnome.Shell.desktop[1272]: libGL error: failed to load driver: swrast
[...]
Jul 27 11:23:56 HomePC org.gnome.Shell.desktop[1272]: X Error of failed request:  GLXBadContext
Jul 27 11:23:56 HomePC org.gnome.Shell.desktop[1272]:   Major opcode of failed request:  153 (GLX)
Jul 27 11:23:56 HomePC org.gnome.Shell.desktop[1272]:   Minor opcode of failed request:  6 (X_GLXIsDirect)
Jul 27 11:23:56 HomePC org.gnome.Shell.desktop[1272]:   Serial number of failed request:  73
Jul 27 11:23:56 HomePC org.gnome.Shell.desktop[1272]:   Current serial number in output stream:  72
```

scaronni, 2017-10-05:
If it is still happening please provide logs and configuration.

enaut, 2017-10-05:
I will test it. But I'm not available for the weekend so it'll take a week or so.

enaut, 2017-10-16:
So **Yes** the problem persists!
As you didn't specify which logs I made journalctl -b logfiles
the first of the logfiles is: systemctl disable gdm ; reboot; --- ; systemctl start gdm
Visual on screen during this boot: The loginscreen is correctly displayed but then when the desktop is supposed to come up it is completely black the nextcloud window opened but without any windowmanager stuff like a border and focus control.
[journal.log](https://github.com/negativo17/nvidia-driver/files/1387709/journal.log)

Second file is: If I have systemctl enable gdm and then reboot -- screen stays black after loading screen the log file is then pulled in a VT
[journal-gdm-enabled.log](https://github.com/negativo17/nvidia-driver/files/1387708/journal-gdm-enabled.log)

enaut, 2017-10-16:
Oh and while reinstalling I found that the nvidia driver is not content with the libGLvnd and replaces the installed one. The installer lets me opt out of this. But if I do not replace the libGLvnd with the NVIDIA one I get the same symptoms as described here.
Might that be some lead?

enaut, 2017-10-23:
I upgraded to Fedora 27 - hoping to fix the problem... but still the same issues...

enaut, 2018-05-01:
After a reinstall everything works as expected closing...

closed 2018-05-01 by enaut