negativo17/nvidia-driver #26: Fedora 26 KDE Desktop stuck in blackscreen

thunderbirdtr, 2017-08-02:
**Hello; 
 _I installed fresh Fedora 26 KDE and updated. Then I installed nvidia packages and stuck in blackscreen._** 

My journalctl logs about nvidia 

> https://da.gd/1Cvv

Lspci : 

```
00:02.0 VGA compatible controller: Intel Corporation 3rd Gen Core processor Graphics Controller (rev 09)
01:00.0 VGA compatible controller: NVIDIA Corporation GK104M [GeForce GTX 680M] (rev a1)
```

Xorg.0.log.old (from previous boot I had to revert intel so I can report it)

> https://paste.fedoraproject.org/paste/NL-TW~rmBD3VbHeo2~XTpA

I assume that cause because of wayland so I had to switch to Xorg or If you have different idea about that please let me know. 

**_Thank you_**.

PS: Please let me know If I need to send something else too.

scaronni, 2017-08-30:
Sorry for the delay. Please see https://github.com/negativo17/nvidia-driver/issues/27. Does it work for you with X.org?

thunderbirdtr, 2017-09-07:
@scaronni  I switched to "X.org" but still happening on KDE and can't open SSDM so I can't see login screen either too.

thunderbirdtr, 2018-07-07:
I fixed with changing ssdm conf to open nvidia and fixed the issues

closed 2018-07-07 by thunderbirdtr