negativo17/nvidia-driver #27: [Fedora 26] Unable to Start Any KMS/EGL Compositor That Uses EGLStream

Saroufim, 2017-08-08:
I'm currently unable to get any EGL compositor running via the NVIDIA driver installed through the negativo17 repo for Fedora 26.

I tried the following compositors:
* **Mutter**  from this repository https://copr.fedorainfracloud.org/coprs/mvicomoya/mutter-eglstream/packages/
* **Sway** from the official Fedora repositories
*  **Weston-eglstream** from AUR

Mutter yields the following output when gnome-shell --wayland is launched `[drm:nvidia_drm_gem_import_nvkms_memory [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to import NVKMS memory to GEM object `and then it terminates with a segmentation fault.

Weston launches as if it's working but displays no actual output and yields no errors when killed. 

Sway doesn't yield any errors but it doesn't start either.

I've been communicating with the maintainer of the mutter-eglstream copr repository who happens to be an employee at NVIDIA, he insists that everything _should_ work but it doesn't. Any clues on why it's not working would be greatly appreciated.

bugworm, 2017-08-12:
Pretty much the same on Arch Linux, but `gnome-shell --wayland` doesn't give me anything. I have 
`GRUB_CMDLINE_LINUX_DEFAULT=quiet loglevel=3 rd.udev.log-priority=3 rd.systemd.show_status=auto log-priority=3 nvidia-drm.modeset=1`
in `/etc/default/grub`
and
`MODULES="nvidia nvidia_modeset nvidia_uvm nvidia_drm"`
in `/etc/mkinitcpio.conf`
([this instruction](https://wiki.archlinux.org/index.php/NVIDIA#DRM_kernel_mode_setting)), so it should work, but it doesn't

leigh123linux, 2017-08-12:
Perhaps @mvicomoya can comment on the driver setup used as nvidia wayland has failed for several months with egl enabled mutter.

https://devtalk.nvidia.com/default/topic/1003294/linux/nvidia-modeset-error-when-trying-to-run-egl-streams-enabled-mutter-/

As for F26 sway, it isn't egl nvidia compatible till you update wlc to wlc-0.0.10-1.fc26

mvicomoya, 2017-08-14:
I'm not sure what's wrong/missing when installing the driver using the negativo17 repository. I spent some time last week trying to figure it out, and I was running into a variety of strange issues.

At some point, after re-installing the driver several times, and manually running akmods to build the kernel modules, I got it working.

It's worth mentioning that using our official .run installer, everything worked out of the box, and other users reporting similar issues have been able to get it working too using the official installer.

The main issue I was facing was that nvidia-modeset wouldn't be able to create the /dev/nvidiactl device node, needed for the EGLDevice extensions to enumerate all NVIDIA devices. Since EGL would fail to find any devices, the [EGL_EXT_device_base](https://www.khronos.org/registry/EGL/extensions/EXT/EGL_EXT_device_base.txt) client extension wouldn't be reported. Mutter would then fail to initialize the EGLDevice backend and try with GBM, which, in the best-case scenario would fall back to software rendering. nvidia-drm would then report those error messages.

leigh123linux, 2017-08-15:
@mvicomoya  adding a udev rule to create /dev/nvidiactl and nvidia0 nodes fixes the issue (using rpmfusion driver).

```
$ cat /usr/lib/udev/rules.d/60-nvidia-drm.rules 
KERNEL=="nvidia_drm", RUN+="/usr/bin/bash -c '/usr/bin/mknod -m 666 /dev/nvidiactl c 195 255"
KERNEL=="nvidia_drm", RUN+="/usr/bin/bash -c '/usr/bin/mknod -m 666 /dev/nvidia0 c 195 0"
```

Does the installer add a udev rule?, if not I don't see what difference using the run file would make.

> It's worth mentioning that using our official .run installer, everything worked out of the box, and other users reporting similar issues have been able to get it working too using the official installer.

Should the nvidia.com driver be fixed so when loading nvidia-drm module is creates the required nodes?

scaronni, 2017-08-15:
> The main issue I was facing was that nvidia-modeset wouldn't be able to create the `/dev/nvidiactl` device node, needed for the EGLDevice extensions to enumerate all NVIDIA devices.

@mvicomoya the nvidiactl device has alwatys been created when the DDX driver was loaded, if I remember correctly. What about Wayland and such? Does the driver assume that device should be already present? I agree that the device node should be created by the kernel module, as Leigh said.

> It's worth mentioning that using our official .run installer, everything worked out of the box, and other users reporting similar issues have been able to get it working too using the official installer.

I guess this is because the various binary components of the driver execute `nvidia-modprobe` to create all required device files. I see multiple references a bit everywhere in the libraries:

`$ for i in $(rpm -qal \*nvidia\* | grep lib); do
   strings $i | grep nvidia-modprobe > /dev/null && echo $i
done
/usr/lib/libnvidia-cfg.so.1
/usr/lib/libnvidia-cfg.so.384.59
/usr/lib/libnvidia-eglcore.so.384.59
/usr/lib/libnvidia-glcore.so.384.59
/usr/lib/libnvidia-glsi.so.384.59
/usr/lib/vdpau/libvdpau_nvidia.so.1
/usr/lib/vdpau/libvdpau_nvidia.so.384.59
/usr/lib64/libnvidia-cfg.so.1
/usr/lib64/libnvidia-cfg.so.384.59
/usr/lib64/libnvidia-eglcore.so.384.59
/usr/lib64/libnvidia-glcore.so.384.59
/usr/lib64/libnvidia-glsi.so.384.59
/usr/lib64/vdpau/libvdpau_nvidia.so.1
/usr/lib64/vdpau/libvdpau_nvidia.so.384.59
/usr/lib64/xorg/modules/drivers/nvidia_drv.so
/usr/lib64/libnvidia-ml.so.1
/usr/lib64/libnvidia-ml.so.384.59
/usr/lib64/libcuda.so
/usr/lib64/libcuda.so.1
/usr/lib64/libcuda.so.384.59
/usr/lib64/libnvcuvid.so.1
/usr/lib64/libnvcuvid.so.384.59
/usr/lib64/libnvidia-opencl.so.1
/usr/lib64/libnvidia-opencl.so.384.59
`
But as stated in the man page, `nvidia-modprobe` should be a last resort, as it is also needs to be installed setuid to load module.

scaronni, 2017-08-15:
If for some reason the driver expects the `/dev/nvidiactl` device and nothing creates it, I would go with the Udev rule instead of shipping `nvidia-modprobe` by default.

jwrdegoede, 2017-08-15:
I agree that the best solution to this seems to be to add the udev rules to the negativo17 pkgs so that the device-nodes get created.

scaronni, 2017-08-15:
@mvicomoya Can you clarify which devices are dynamically created? In nvidia-modprobe there are a ton of devices, including instantiated /dev/nvidiactl ones. Do we need udev rules for all of them?

https://github.com/NVIDIA/nvidia-modprobe/blob/master/modprobe-utils/nvidia-modprobe-utils.h#L33-L54

Otherwise I'm going to create all instantiated /dev/nvidia0 to /dev/nvidia8 devices as well.

scaronni, 2017-08-15:
Something like this, so all the device minor numbers used come actually from the number of installed cards:

```
KERNEL=="nvidia_drm", RUN+="/usr/bin/bash -c '/usr/bin/mknod -m 666 /dev/nvidiactl c 195 255"
KERNEL=="nvidia_drm", RUN+="/usr/bin/bash -c 'for i in $(cat /proc/driver/nvidia/gpus/*/information | grep Minor | cut -d \  -f 4); do /usr/bin/mknod -m 666 /dev/nvidia${i} c 195 {i}; done"
```

leigh123linux, 2017-08-15:
@scaronni Looks good to me.

mvicomoya, 2017-08-15:
Yes, sorry I missed it earlier, I probably screw something up while switching between drivers and didn't realize negativo17 drivers wouldn't install nvidia-modprobe.

In addition to /dev/nvidiactl and /dev/nvidia0-7, you'll also want /dev/nvidia-modeset

dadap, 2017-08-15:
You might also want to key off of KERNEL=="nvidia" rather than KERNEL=="nvidia_drm" for /dev/nvidiactl and /dev/nvidia0-7, as those device nodes interface with nvidia.ko rather than nvidia-drm.ko. And you'll want /dev/nvidia-modeset to be tied to KERNEL=="nvidia_modeset"

leigh123linux, 2017-08-15:
@dadap Thank you for the clarification, this is working here

```
KERNEL=="nvidia", RUN+="/usr/bin/bash -c '/usr/bin/mknod -m 666 /dev/nvidiactl c 195 255"
KERNEL=="nvidia", RUN+="/usr/bin/bash -c 'for i in $(cat /proc/driver/nvidia/gpus/*/information | grep Minor | cut -d \  -f 4); do /usr/bin/mknod -m 666 /dev/nvidia${i} c 195 {i}; done"
KERNEL=="nvidia_modeset", RUN+="/usr/bin/bash -c '/usr/bin/mknod -m 666 /dev/nvidia-modeset c 195 254"
```

leigh123linux, 2017-08-16:
@scaronni 

I've added a rules file to fix this in the rpmfusion driver

https://pkgs.rpmfusion.org/cgit/nonfree/xorg-x11-drv-nvidia.git/commit/?id=6e38d2ae8a78e04b1843273f36bf955c27f17040

There is also an added bonus that prime sync seems to be working ok with modeset enabled

https://pkgs.rpmfusion.org/cgit/nonfree/xorg-x11-drv-nvidia.git/commit/?id=2dec942e6cc48d3026be5204fc9a52458ef1ae10

scaronni, 2017-08-16:
Thanks to everyone! I've added the rules as is. I will make a test with Prime Sync soon.

leigh123linux, 2017-08-17:
@mvicomoya  I can start a gnome-wayland session but it's only SW

```
Aug 17 11:35:35 localhost.localdomain gnome-shell[1017]: Failed to apply DRM plane transform 0: Invalid argument
Aug 17 11:35:35 localhost.localdomain org.gnome.Shell.desktop[1017]: Disabling glamor and dri3, EGL setup failed
Aug 17 11:35:35 localhost.localdomain org.gnome.Shell.desktop[1017]: Failed to initialize glamor, falling back to sw
```

mvicomoya, 2017-08-18:
@leigh123linux Those messages are from glamor, I think I've always seen them when running with NVIDIA drivers.

Are you sure mutter is running in SW rendering mode? If so, have you tried https://copr.fedorainfracloud.org/coprs/mvicomoya/mutter-eglstream/ ?

mvicomoya, 2017-08-18:
BTW, I think GDM might run in SW rendering mode. I have to look into it. But once you passed the login screen, mutter should use HW acceleration.

leigh123linux, 2017-08-18:
@mvicomoya I am using your mutter packages, inxi comfirms gnome-shell is using SW

mvicomoya, 2017-08-18:
@leigh123linux, what version of the driver are you using?

Also, have you tried disabling the Mesa driver so that GLVND doesn't pick it?:

`export __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json`

Or removing 50_mesa.json under that same directory.

leigh123linux, 2017-08-18:
@mvicomoya  I have the same result using 375.82 or 384.59
I've already tried removing 50_mesa.json and it makes no difference as such, eglinfo returns uninitialized or something similar.

leigh123linux, 2017-08-19:
And it's no different using the nvidia .run file installer

```
[leigh@localhost ~]$ inxi -G
Graphics:  Card: NVIDIA GK106 [GeForce GTX 660]
           Display Server: wayland (X.org 119.3 )
           drivers: nvidia (unloaded: modesetting,fbdev,vesa,nouveau)
           Resolution: 3840x2160@59.98hz
           OpenGL: renderer: Gallium 0.4 on llvmpipe (LLVM 4.0, 128 bits)
           version: 2.1 Mesa 17.1.5
[leigh@localhost ~]$ eglinfo 
EGL API version: 1.4
EGL vendor string: Mesa Project
EGL version string: 1.4 (DRI2)
EGL client APIs: OpenGL OpenGL_ES 
EGL extensions string:
    EGL_KHR_config_attribs EGL_KHR_create_context
    EGL_KHR_get_all_proc_addresses EGL_KHR_gl_colorspace
    EGL_KHR_no_config_context EGL_KHR_reusable_sync
    EGL_KHR_surfaceless_context EGL_MESA_configless_context
EGL client extensions string:
    EGL_EXT_platform_base EGL_EXT_device_base
    EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions
    EGL_KHR_debug EGL_EXT_platform_x11 EGL_EXT_platform_device
    EGL_EXT_platform_wayland EGL_MESA_platform_gbm
    EGL_MESA_platform_surfaceless
Configurations:
     bf lv colorbuffer dp st  ms    vis   cav bi  renderable  supported
  id sz  l  r  g  b  a th cl ns b    id   eat nd gl es es2 vg surfaces 
---------------------------------------------------------------------
0x01 32  0  8  8  8  8  0  0  0 0 0x24TC      a  y  y  y     win,pb,pix
0x02 32  0  8  8  8  8 16  0  0 0 0x24TC      a  y  y  y     win,pb,pix
0x03 32  0  8  8  8  8 24  0  0 0 0x24TC      a  y  y  y     win,pb,pix
0x04 32  0  8  8  8  8 24  8  0 0 0x24TC      a  y  y  y     win,pb,pix
0x05 32  0  8  8  8  8 32  0  0 0 0x24TC      a  y  y  y     win,pb,pix
0x06 24  0  8  8  8  0  0  0  0 0 0x24TC      y  y  y  y     win,pb,pix
0x07 24  0  8  8  8  0 16  0  0 0 0x24TC      y  y  y  y     win,pb,pix
0x08 24  0  8  8  8  0 24  0  0 0 0x24TC      y  y  y  y     win,pb,pix
0x09 24  0  8  8  8  0 24  8  0 0 0x24TC      y  y  y  y     win,pb,pix
0x0a 24  0  8  8  8  0 32  0  0 0 0x24TC      y  y  y  y     win,pb,pix
0x0b 32  0  8  8  8  8  0  0  0 0 0x25DC      a  y  y  y     win,pb,pix
0x0c 32  0  8  8  8  8 16  0  0 0 0x25DC      a  y  y  y     win,pb,pix
0x0d 32  0  8  8  8  8 24  0  0 0 0x25DC      a  y  y  y     win,pb,pix
0x0e 32  0  8  8  8  8 24  8  0 0 0x25DC      a  y  y  y     win,pb,pix
0x0f 32  0  8  8  8  8 32  0  0 0 0x25DC      a  y  y  y     win,pb,pix
0x10 24  0  8  8  8  0  0  0  0 0 0x25DC      y  y  y  y     win,pb,pix
0x11 24  0  8  8  8  0 16  0  0 0 0x25DC      y  y  y  y     win,pb,pix
0x12 24  0  8  8  8  0 24  0  0 0 0x25DC      y  y  y  y     win,pb,pix
0x13 24  0  8  8  8  0 24  8  0 0 0x25DC      y  y  y  y     win,pb,pix
0x14 24  0  8  8  8  0 32  0  0 0 0x25DC      y  y  y  y     win,pb,pix
[leigh@localhost ~]$ ls /dev/ |grep nvidia
nvidia0
nvidiactl
nvidia-modeset
```

mvicomoya, 2017-08-21:
I don't know what might be preventing libEGL_nvidia.so from enumerating devices in your setup, and I'm actually out of ideas for you to try.

Could you file a formal bug attaching the nvidia-bug-report tarball and CC me? I'll try to get a setup as close as possible to yours and get a repro.

scaronni, 2017-08-22:
Got some time to fiddle with the udev rules today. They are not working properly as they are not getting the correct SELinux context and I'm pretty surprised that they don't get it correctly:

```
# semanage fcontext -l | grep nvidia
/dev/nvidia.*                                      character device   system_u:object_r:xserver_misc_device_t:s0 
/usr/(.*/)?lib(64)?(/.*)?/nvidia/.+\.so(\..*)?     regular file       system_u:object_r:textrel_shlib_t:s0 
/usr/(.*/)?nvidia/.+\.so(\..*)?                    regular file       system_u:object_r:textrel_shlib_t:s0 
/usr/bin/nvidia.*                                  regular file       system_u:object_r:xserver_exec_t:s0 
/usr/lib(/.*)?/nvidia/.+\.so(\..*)?                regular file       system_u:object_r:textrel_shlib_t:s0 
/usr/lib(/.*)?/nvidia_drv.*\.so(\.[^/]*)*          regular file       system_u:object_r:textrel_shlib_t:s0 
/usr/lib.*/libnvidia\.so(\.[^/]*)*                 regular file       system_u:object_r:textrel_shlib_t:s0 
/usr/lib/libnvidia\.so(\.[^/]*)*                   regular file       system_u:object_r:textrel_shlib_t:s0 
/usr/lib/nvidia-graphics(-[^/]*/)?libGL(core)?\.so(\.[^/]*)* regular file       system_u:object_r:textrel_shlib_t:s0 
/usr/lib/nvidia-graphics(-[^/]*/)?libXvMCNVIDIA\.so.* regular file       system_u:object_r:textrel_shlib_t:s0 
/usr/lib/nvidia-graphics(-[^/]*/)?libnvidia.*\.so(\.[^/]*)* regular file       system_u:object_r:textrel_shlib_t:s0 
/usr/lib/nvidia.*\.so(\.[^/]*)*                    regular file       system_u:object_r:textrel_shlib_t:s0 
/usr/lib/vdpau/libvdpau_nvidia\.so.*               regular file       system_u:object_r:textrel_shlib_t:s0 
/usr/lib/xorg/modules/drivers/nvidia_drv\.o        regular file       system_u:object_r:textrel_shlib_t:s0 
/usr/lib/xorg/modules/extensions/nvidia(-[^/]*)?/libglx\.so(\.[^/]*)* regular file       system_u:object_r:textrel_shlib_t:s0 
/var/log/nvidia-installer\.log.*                   regular file       system_u:object_r:xserver_log_t:s0 
```
So, according to the policy, any regexp matched device gets the `xserver_misc_device_t` type. But this is not happening. It is also not happening on all other udev created devices that were shipped for quite a long time, and I never spot it because it's not a problem. So, with autocreated devices from X starting:

```
# ls -alZ /dev/nvidia*
crw-rw-rw-. 1 root root system_u:object_r:xserver_misc_device_t:s0 195,   0 Aug 16 17:44 /dev/nvidia0
crw-rw-rw-. 1 root root system_u:object_r:xserver_misc_device_t:s0 195, 255 Aug 16 17:44 /dev/nvidiactl
crw-rw-rw-. 1 root root system_u:object_r:xserver_misc_device_t:s0 195, 254 Aug 16 17:44 /dev/nvidia-modeset
crw-rw-rw-. 1 root root system_u:object_r:device_t:s0              239,   0 Aug 16 17:44 /dev/nvidia-uvm
crw-rw-rw-. 1 root root system_u:object_r:device_t:s0              239,   1 Aug 16 17:44 /dev/nvidia-uvm-tools
```
With the same devices created from udev rules I get this:
```
# ls -alZ /dev/nvidia*
crw-rw-rw-. 1 root root system_u:object_r:xserver_misc_device_t:s0 195,   0 Aug 22 11:47 /dev/nvidia0
crw-rw-rw-. 1 root root system_u:object_r:xserver_misc_device_t:s0 195, 255 Aug 22 11:47 /dev/nvidiactl
crw-rw-rw-. 1 root root system_u:object_r:device_t:s0              195, 254 Aug 22 11:47 /dev/nvidia-modeset
crw-rw-rw-. 1 root root system_u:object_r:device_t:s0              239,   0 Aug 22 11:47 /dev/nvidia-uvm
crw-rw-rw-. 1 root root system_u:object_r:device_t:s0              239,   1 Aug 22 11:47 /dev/nvidia-uvm-tools
```
According to the policy, all devices should have `xserver_misc_device_t`, but this is not happening if the devices are created with udev. I also don't know why 2 out of 5 are actually getting the correct context. 

With the current X.org logic in Fedora 26, an X server is started if there is an Nvidia driver enabled on the system; so this fails with an SELinux error and GDM never starts. Modules are not included in the initrd:
```
# cat /usr/lib/dracut.conf.d/99-nvidia-dracut.conf 
# omit the nvidia driver from the ramdisk, to avoid needing to regenerate
# the ramdisk on nvidia driver updates
omit_drivers+=" nvidia-drm "
```
@jwrdegoede any idea why this might be happening? We can force the context in the udev rule, but this should not actually be needed. Maybe the rules are actually creating the devices before the SELinux policy is loaded in the kernel?

scaronni, 2017-08-22:
Assuming the udev/module/SElinux woes are fixed, I have as well Gnome running on Wayland with software rendering and the error pasted from the original poster:

```[drm:nvidia_drm_gem_import_nvkms_memory [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to import NVKMS memory to GEM object```

mvicomoya, 2017-08-22:
@leigh123linux, @scaronni, it seems there was an issue with the [Copr](https://copr.fedorainfracloud.org/coprs/mvicomoya/mutter-eglstream/) and the right RPM from the last build was removed. I just rebuilt it and it seems that mutter is picking NVIDIA by default again.

That still doesn't explain why it didn't work when @leigh123linux removed the 50_mesa.json file, as it should.

Here are the steps I followed to get it working with F26:

   1. Get a fresh install of F26
   2. Install NVIDIA drivers from negativo17
   3. Add udev rules as per comment https://github.com/negativo17/nvidia-driver/issues/27#issuecomment-323037208
   4. Reload nvidia-drm module / Reboot
   5. Reinstall mutter using [mvicomoya/mutter-eglstream](https://copr.fedorainfracloud.org/coprs/mvicomoya/mutter-eglstream/)

Then if I run
`mutter --wayland`
I can see the NVIDIA GPU being used.

Gnome-shell works with NVIDIA too. However, GDM seems to run in SW rendering (looking into it).

I haven't had any problems with SELinux.

scaronni, 2017-08-23:
Thanks! Will attempt to re-test. Still trying to figure out what is happening with SELinux .

leigh123linux, 2017-08-23:
@mvicomoya  I believe my issue maybe caused by Xwayland

weston without xwayland.so module

```
[leigh@localhost ~]$ eglinfo 
EGL API version: 1.4
EGL vendor string: NVIDIA
EGL version string: 1.4
EGL client APIs: OpenGL_ES OpenGL
EGL extensions string:
    EGL_EXT_create_context_robustness EGL_EXT_output_base
    EGL_EXT_output_drm EGL_EXT_stream_consumer_egloutput
    EGL_EXT_stream_acquire_mode EGL_IMG_context_priority
    EGL_KHR_config_attribs EGL_KHR_create_context_no_error
    EGL_KHR_create_context EGL_KHR_fence_sync
    EGL_KHR_get_all_proc_addresses EGL_KHR_swap_buffers_with_damage
    EGL_KHR_gl_renderbuffer_image EGL_KHR_gl_texture_2D_image
    EGL_KHR_gl_texture_3D_image EGL_KHR_gl_texture_cubemap_image
    EGL_KHR_image EGL_KHR_image_base EGL_KHR_reusable_sync EGL_KHR_stream
    EGL_KHR_stream_consumer_gltexture EGL_KHR_stream_cross_process_fd
    EGL_KHR_stream_fifo EGL_KHR_stream_producer_eglsurface
    EGL_KHR_surfaceless_context EGL_NV_stream_metadata EGL_NV_stream_sync
    EGL_NV_stream_consumer_gltexture_yuv EGL_NV_stream_attrib EGL_NV_sync
    EGL_NV_system_time EGL_NV_output_drm_flip_event
    EGL_WL_bind_wayland_display EGL_WL_wayland_eglstream
EGL client extensions string:
    EGL_EXT_platform_base EGL_EXT_device_base
    EGL_KHR_client_get_all_proc_addresses EGL_EXT_client_extensions
    EGL_KHR_debug EGL_EXT_platform_x11 EGL_EXT_platform_wayland
    EGL_EXT_platform_device EGL_MESA_platform_gbm
    EGL_MESA_platform_surfaceless
Configurations:
     bf lv colorbuffer dp st  ms    vis   cav bi  renderable  supported
  id sz  l  r  g  b  a th cl ns b    id   eat nd gl es es2 vg surfaces 
---------------------------------------------------------------------
0x01 32  0  8  8  8  8 24  8  0 0 0x00--         y  y  y     pb
0x02 32  0  8  8  8  8 24  0  0 0 0x00--         y  y  y     pb
0x03 32  0  8  8  8  8  0  0  0 0 0x00--         y  y  y     pb
0x04 32  0  8  8  8  8 24  8  2 1 0x00--         y  y  y     pb
0x05 32  0  8  8  8  8 24  0  2 1 0x00--         y  y  y     pb
0x06 32  0  8  8  8  8  0  0  2 1 0x00--         y  y  y     pb
0x07 32  0  8  8  8  8 24  8  4 1 0x00--         y  y  y     pb
0x08 32  0  8  8  8  8 24  8  4 1 0x00--         y  y  y     pb
0x09 32  0  8  8  8  8 24  0  4 1 0x00--         y  y  y     pb
0x0a 32  0  8  8  8  8 24  0  4 1 0x00--         y  y  y     pb
0x0b 32  0  8  8  8  8  0  0  4 1 0x00--         y  y  y     pb
0x0c 32  0  8  8  8  8  0  0  4 1 0x00--         y  y  y     pb
0x0d 32  0  8  8  8  8 24  8  8 1 0x00--         y  y  y     pb
0x0e 32  0  8  8  8  8 24  0  8 1 0x00--         y  y  y     pb
0x0f 32  0  8  8  8  8  0  0  8 1 0x00--         y  y  y     pb
0x10 24  0  8  8  8  0 24  8  0 0 0x00--         y  y  y     pb
0x11 24  0  8  8  8  0 24  0  0 0 0x00--         y  y  y     pb
0x12 24  0  8  8  8  0  0  0  0 0 0x00--         y  y  y     pb
0x13 24  0  8  8  8  0 24  8  2 1 0x00--         y  y  y     pb
0x14 24  0  8  8  8  0 24  0  2 1 0x00--         y  y  y     pb
0x15 24  0  8  8  8  0  0  0  2 1 0x00--         y  y  y     pb
0x16 24  0  8  8  8  0 24  8  4 1 0x00--         y  y  y     pb
0x17 24  0  8  8  8  0 24  8  4 1 0x00--         y  y  y     pb
0x18 24  0  8  8  8  0 24  0  4 1 0x00--         y  y  y     pb
0x19 24  0  8  8  8  0 24  0  4 1 0x00--         y  y  y     pb
0x1a 24  0  8  8  8  0  0  0  4 1 0x00--         y  y  y     pb
0x1b 24  0  8  8  8  0  0  0  4 1 0x00--         y  y  y     pb
0x1c 24  0  8  8  8  0 24  8  8 1 0x00--         y  y  y     pb
0x1d 24  0  8  8  8  0 24  0  8 1 0x00--         y  y  y     pb
0x1e 24  0  8  8  8  0  0  0  8 1 0x00--         y  y  y     pb
0x1f 16  0  5  6  5  0 24  8  0 0 0x00--         y  y  y     pb
0x20 16  0  5  6  5  0 24  0  0 0 0x00--         y  y  y     pb
0x21 16  0  5  6  5  0 16  0  0 0 0x00--         y  y  y     pb
0x22 16  0  5  6  5  0  0  0  0 0 0x00--         y  y  y     pb
[... 94 lines cut ...]
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_texture_from_pixmap, GLX_EXT_visual_info, 
    GLX_EXT_visual_rating, GLX_MESA_copy_sub_buffer, 
    GLX_MESA_multithread_makecurrent, GLX_MESA_query_renderer, 
    GLX_OML_swap_method, GLX_SGIS_multisample, GLX_SGIX_fbconfig, 
    GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, GLX_SGI_make_current_read
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: VMware, Inc. (0xffffffff)
    Device: llvmpipe (LLVM 4.0, 128 bits) (0xffffffff)
    Version: 17.1.5
    Accelerated: no
    Video memory: 16091MB
    Unified memory: no
    Preferred profile: compat (0x2)
    Max core profile version: 0.0
    Max compat profile version: 2.1
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 2.0
OpenGL vendor string: VMware, Inc.
OpenGL renderer string: Gallium 0.4 on llvmpipe (LLVM 4.0, 128 bits)
OpenGL version string: 2.1 Mesa 17.1.5
OpenGL shading language version string: 1.30
OpenGL extensions:
```

mvicomoya, 2017-08-23:
@leigh123linux, we don't have GLX support with XWayland. I don't think any OpenGL application that tries to use X on Wayland using XWayland will work.

Since eglinfo uses eglGetDisplay(EGL_DEFAULT_DISPLAY), GLVND might be trying to use X, or the first available vendor that can handle a "default" display.

Does the result change if you run

`EGL_PLATFORM=wayland eglinfo`

?

leigh123linux, 2017-08-27:
@scaronni This commit seems to address the context issue

https://pkgs.rpmfusion.org/cgit/nonfree/xorg-x11-drv-nvidia.git/commit/?id=f157a9e1b75320f0ac55208f89b1cf09b10be948

scaronni, 2017-08-27:
Thanks! I did the same test last time with `-Z/--context`, but I'm pretty surprised it's actually needed. Fedora does not ship any udev rule with `-Z`. Why are the Nvidia devices the only ones getting the wrong context at creation time? There are all sort of SELinux types in `/dev`, but they all get created correctly without `-Z`

scaronni, 2017-08-27:
I see the mutter patches in your copr repository will be in 3.26, so I guess modeset can be safely enabled for Fedora 27.

Saroufim, 2018-02-13:
An update must have fixed this. EGLsteams compositors now work under the negativo17 driver. Performance is horrible on gnome-shell since it uses software rendering but that can be remedied by enabling eglstreams for mutter. Everything works fine except Xwayland which has horrible openGL performance.

closed 2018-11-05 by scaronni