negativo17/nvidia-driver #28: [Question] how do install older version? (testing for opencl performance Regression)

arpu, 2017-08-19:
how can i install an older kernel module and driver ? 
since some last updates my opencl image blender has very poor performance with latest 
driver and opencl libnvidia-opencl.so.384.59

leigh123linux, 2017-08-19:
Your question is more suited for a forum!

arpu, 2017-08-19:
ok  i get this sort out
and installing the NVIDIA-Linux-x86_64-381.22-no-compat32.run
  Device Version                                  OpenCL 1.2 CUDA
  Driver Version                                  381.22
  Device OpenCL C Version                         OpenCL C 1.2 

with this version my image opencl multibandblender works fine
any idea where i should report this Regression?

scaronni, 2017-08-22:
Once you are sure it's an issue with specific drivers, i guess your best place would be on the nvidia linux driver forum:

https://devtalk.nvidia.com/default/board/98/linux/

closed 2017-08-22 by arpu