negativo17/nvidia-driver #29: [Fedora 26] CUDA not working

pstrejajo, 2017-08-28:
74c5de8677f27bc9da16353ccb1ddd786eab65e0 introduced hardcoded major numbers for UVM devices which has not been reverted in 1ac9f7d48a3abe276bf6865d8d22985eaee9e51b. This makes CUDA unusable unless it is used by root and in consequence UVM devices are rebuild with proper major number.

scaronni, 2017-08-29:
Thanks for opening the issue. Can you elaborate a bit more?
On my systems, the `/dev/nvidia*` devices are always created with the same major number, so I thought that the check on the major number was pretty useless:

```
$ ls -al nvidia*
crw-rw-rw-. 1 root root 195,   0 Aug 23 17:11 nvidia0
crw-rw-rw-. 1 root root 195, 255 Aug 23 17:11 nvidiactl
crw-rw-rw-. 1 root root 195, 254 Aug 23 17:11 nvidia-modeset
crw-rw-rw-. 1 root root 239,   0 Aug 23 17:11 nvidia-uvm
crw-rw-rw-. 1 root root 239,   1 Aug 23 17:11 nvidia-uvm-tools
```
Are you seeing different major numbers in your system? Can you paste the output of the above command?

Thanks.

scaronni, 2017-08-29:
https://github.com/negativo17/nvidia-driver/commit/7d64aec0866fb4f6366998715fd723ac9e26f9cc

Waiting for your reply before pushing the build.

scaronni, 2017-08-29:
Please paste the output of `cat /proc/devices`, I'm curious what made your major number shift.

According to `nvidia-modprobe` the major is hardcoded for the main nvidia devices but not for the UVM ones:

https://github.com/NVIDIA/nvidia-modprobe/blob/master/modprobe-utils/nvidia-modprobe-utils.c#L72

So probably I can change the previous commit to just make the UVM rules dynamic.

pstrejajo, 2017-08-29:
Hi, below are the details you were asking
```
➜  ~ ll /dev/nvidia*                              
crw-rw-rw-. 1 root root 195,   0 08-29 16:49 /dev/nvidia0
crw-rw-rw-. 1 root root 195, 255 08-29 16:49 /dev/nvidiactl
crw-rw-rw-. 1 root root 195, 254 08-29 16:49 /dev/nvidia-modeset
crw-rw-rw-. 1 root root 241,   0 08-29 16:48 /dev/nvidia-uvm
crw-rw-rw-. 1 root root 241,   1 08-29 16:48 /dev/nvidia-uvm-tools
```
```
➜  ~ cat /proc/devices                             
Character devices:
  1 mem
  4 /dev/vc/0
  4 tty
  4 ttyS
  5 /dev/tty
  5 /dev/console
  5 /dev/ptmx
  7 vcs
 10 misc
 13 input
 14 sound
 21 sg
 29 fb
 81 video4linux
 99 ppdev
116 alsa
128 ptm
136 pts
162 raw
180 usb
188 ttyUSB
189 usb_device
195 nvidia-frontend
202 cpu/msr
203 cpu/cpuid
226 drm
241 nvidia-uvm
242 nvlink_core
243 aux
244 tpm
245 media
246 hidraw
247 usbmon
248 bsg
249 watchdog
250 rtc
251 dax
252 dimmctl
253 ndctl
254 gpiochip

Block devices:
259 blkext
  8 sd
  9 md
 11 sr
 65 sd
 66 sd
 67 sd
 68 sd
 69 sd
 70 sd
 71 sd
128 sd
129 sd
130 sd
131 sd
132 sd
133 sd
134 sd
135 sd
253 device-mapper
254 mdp
```

scaronni, 2017-08-30:
Oh, nothing makes it shift, it's just different. The updates is pushed. Thanks!

closed 2017-08-30 by scaronni