negativo17/nvidia-driver #31: OpenCL not working

hubslave, 2017-09-05:
Hello, I installed this nvidia drivers on fresh EL7.4, on optimus laptop. Everythings works fine, except for OpenCL. I also installed packages opencl-filesystem and ocl-icd but it doesnt seem to help. I cant get it working in any application.

for example here is the output from darktable:

darktable -d opencl
…
[opencl_init] opencl related configuration options:
[opencl_init]
[opencl_init] opencl: 1
[opencl_init] opencl_library: ”
[opencl_init] opencl_memory_requirement: 768
[opencl_init] opencl_memory_headroom: 300
[opencl_init] opencl_device_priority: ‘/!0,//‘
[opencl_init] opencl_size_roundup: 16
[opencl_init] opencl_async_pixelpipe: 0
[opencl_init] opencl_synch_cache: 0
[opencl_init] opencl_number_event_handles: 25
[opencl_init] opencl_micro_nap: 1000
[opencl_init] opencl_use_pinned_memory: 0
[opencl_init] opencl_use_cpu_devices: 0
[opencl_init] opencl_avoid_atomics: 0
[opencl_init] opencl_omit_whitebalance: 0
[opencl_init]
[opencl_init] could not find opencl runtime library ‘libOpenCL’
[opencl_init] could not find opencl runtime library ‘libOpenCL.so’
[opencl_init] found opencl runtime library ‘libOpenCL.so.1’
[opencl_init] opencl library ‘libOpenCL.so.1’ found on your system and loaded
[opencl_init] could not get platforms: -1001
[opencl_init] FINALLY: opencl is NOT AVAILABLE on this system.
[opencl_init] initial status of opencl enabled flag is OFF.

If you need any more info or testing please let me know 🙂

Have a nice day

hubslave, 2017-09-05:
Resolved by installing nvidia-modprobe... Maybe it could be installed by default? :)

scaronni, 2017-09-06:
So, the `nvidia-driver-cuda` package adds the udev rules for the `nvidia-uvm` device. You need to reboot after installing it, otherwise the rules take no effect. `nvidia-modprobe` creates all devices directly because it is a suid program, but it should not be needed.

Did you try rebooting after installing `nvidia-cuda-driver`? If you don't know, just remove `nvidia-modprobe` and reboot, everything should be fine.

scaronni, 2017-09-06:
Also, the loader is already in the package, what did you have to create exactly?
```
$ rpm -ql nvidia-driver-cuda | grep CL
/etc/OpenCL/vendors/nvidia.icd
$ cat /etc/OpenCL/vendors/nvidia.icd 
libnvidia-opencl.so.1
```

hubslave, 2017-09-07:
Well I installed opencl-filesystem and ocl-icd after installing nvidia drivers and realizing that OCL is not working, so I probably didnt have OpenCL folders created before I installed the driver. Maybe the installation of the drivers expects having the OpenCL folders already in place? In other words icd loader is created only if I have these packages already installed?

scaronni, 2017-09-08:
Nope, everything is as should be, packages are installed as part of the dependencies:

```
$ rpm -q --requires nvidia-driver-cuda | grep -E "ocl|opencl"
ocl-icd
opencl-filesystem
```

hubslave, 2017-09-11:
@scaronni thank you very much for clarification... I tried installing it again and yes, icd loader is created fine. It was probably my mistake last time... But I tested OpenCL and I found out it does not work without having nvidia-modprobe installed. I even tried restarting the PC after installing nvidia-driver-cuda but it did not work without nvidia-modprobe

scaronni, 2017-09-11:
That's strange, it shouldn't be. Can you paste the output of these two commands?

```
$ cat /proc/devices | grep -i nvidia
$ ls -al /dev/nvidia*
```

Thanks.

scaronni, 2017-09-11:
So all those device files are created with udev rules:

```
$ cat /usr/lib/udev/rules.d/60-nvidia-*
```
`nvidia-modprobe` just creates them all when needed with a SUID binary and no udev rule present, so whether you have that installed or not it should not make a difference.

Do you have version 384.69 installed? Can you remove `nvidia-modprobe`, reboot, and if OpenCL still does not work, paste the output of the same 2 commands as per the previous command?

If `nvidia-modprobe` is required to recreate the devices, this means the above outputs should be different in the major number for the *uvm devices.

scaronni, 2017-09-11:
Maybe you did not reboot after installing `nvidia-driver-cuda`? That unfortunately is required, unless then you also have `nvidia-modprobe` installed, as the udev rules get created at boot time.

hubslave, 2017-09-11:
Yes I have latest version and I rebooted after installing nvidia-driver-cuda.

Ok so right after uninstalling nvidia-modprobe OpenCL was still working for me... After reboot OpenCL was no longer available in graphic softwares.

This is my output without nvidia-modprobe:

$ cat /proc/devices | grep -i nvidia
195 nvidia-frontend

$ ls -al /dev/nvidia*
crw-rw-rw-. 1 root root 195,   0 Sep 11 21:51 /dev/nvidia0
crw-rw-rw-. 1 root root 195, 255 Sep 11 21:51 /dev/nvidiactl
crw-rw-rw-. 1 root root 195, 254 Sep 11 21:51 /dev/nvidia-modeset

After installing nvidia-modprobe OpenCL works fine again.

scaronni, 2017-09-12:
Ok, so the issue is that somehow the `/usr/lib/udev/rules.d/60-nvidia-uvm.rules` file is ignored. I'll check into that.

scaronni, 2017-09-12:
Well, it is actually the loading of the module that fails:

`$ cat /usr/lib/modprobe.d/nvidia-uvm.conf`

hubslave, 2017-09-12:
If you needed any more testing or information just len me know :)

scaronni, 2017-09-12:
Thanks! Much appreciated. I'm going to make a test this evening on my CentOS systems and check.

hubslave, 2017-09-14:
Hello Simone, I have no problems with drivers right now, everything works fine with nvidia-modprobe, I was just curious if you found out what was the problem :)

obmun, 2017-12-16:
I'm seeing something similar in a Fedora 27 installation. I've just upgraded this morning from Fedora 26 using system-upgrade. In my case nvidia-uvm module is _not_being loaded automatically. I see that the modprobe softdep is there (/usr/lib/modprobe.d/nvidia-uvm.conf). These are the packages I have installed in the system:
```
[obmun@madrid modprobe.d]$ dnf list installed "*nvidia*"
Installed Packages
akmod-nvidia.x86_64                                                       2:387.34-1.fc27
kmod-nvidia-4.14.5-300.fc27.x86_64.x86_64                                 2:387.34-1.fc27
nvidia-driver.x86_64                                                      2:387.34-1.fc27
nvidia-driver-NVML.x86_64                                                 2:387.34-1.fc27
nvidia-driver-NvFBCOpenGL.x86_64                                          2:387.34-1.fc27
nvidia-driver-cuda.x86_64                                                 2:387.34-1.fc27
nvidia-driver-cuda-libs.x86_64                                            2:387.34-1.fc27
nvidia-driver-devel.x86_64                                                2:387.34-1.fc27
nvidia-driver-libs.x86_64                                                 2:387.34-1.fc27
nvidia-libXNVCtrl.x86_64                                                  2:387.34-1.fc27
nvidia-persistenced.x86_64                                                2:387.34-1.fc27
nvidia-settings.x86_64                                                    2:387.34-1.fc27
```

If I manually modprobe nvidia-uvm everything goes fine (udev rules are working), and I have a working Nvidia OpenCL. I'm not 100% familiar with the softdev mechanism, but I'll try to understand later what might be failing. The Nvidia GPU is an old GTX 860M.

scaronni, 2018-01-09:
I still can't reproduce this. All of my systems load the UVM module fine. Is this still happening?

hubslave, 2018-03-15:
Well to be honest I don't know. I just install nvidia-driver-cuda and nvidia-modprobe all the time and it keeps working without problems that way :)

scaronni, 2019-02-06:
Changed logic completely, now all kernel module variants (kabi, akmods, dkms) require `nvidia-kmod-common` which contains all the logic, including loading of everything.

The package `nvidia-driver` (DDX + OpenGL) and `nvidia-driver-cuda` (CUDA components) have a dependency on` nvidia-kmod-common` and can be installed together or standalone, depending on what you want to achieve.

Ergo, installing `nvidia-driver-cuda` only will give you a fully functional headless CUDA system. I've updated the main page on the web site, I will write a blog post soon.

Please reopen if it happens with the new logic. `nvidia-modprobe` is still not a required pacakge.

closed 2019-02-06 by scaronni