negativo17/nvidia-driver #32: Failed to connect to desktop Fedora 27 on 4.13.3-301.fc27.x86_64 Failed to import NVKMS memory to GEM object

yodatak, 2017-09-27:
With the driver and kernel i can't connect to my session the session crash and relance another 

i got  this about Failed to import NVKMS memory to GEM object

[   83.655556] nvidia: loading out-of-tree module taints kernel.
[   83.655559] nvidia: module license 'NVIDIA' taints kernel.
[   83.655559] Disabling lock debugging due to kernel taint
[   83.658064] nvidia: module verification failed: signature and/or required key missing - tainting kernel
[   83.661330] nvidia-nvlink: Nvlink Core is being initialized, major device number 242
[   83.661524] nvidia 0000:01:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
[   83.661578] NVRM: loading NVIDIA UNIX x86_64 Kernel Module  384.90  Tue Sep 19 19:17:35 PDT 2017 (using threaded interrupts)
[   83.681634] nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  384.90  Tue Sep 19 17:05:19 PDT 2017
[   83.683462] [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
[   84.286572] NVRM: Your system is not currently configured to drive a VGA console
               on the primary VGA device. The NVIDIA Linux graphics driver
               requires the use of a text-mode VGA console. Use of other console
               drivers including, but not limited to, vesafb, may result in
               corruption and stability problems, and is not supported.
[   84.286759] nvidia-modeset: Allocated GPU:0 (GPU-0417adc0-dbd3-ff8e-71ca-26c05cfb0586) @ PCI:0000:01:00.0
[   84.390373] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[   84.390374] [drm] No driver support for vblank timestamp query.
[   84.393175] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:01:00.0 on minor 0
[   85.576152] [drm:nvidia_drm_gem_import_nvkms_memory [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to import NVKMS memory to GEM object
[   85.714362] [drm:nvidia_drm_gem_import_nvkms_memory [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to import NVKMS memory to GEM object
[   85.716822] [drm:nvidia_drm_gem_import_nvkms_memory [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to import NVKMS memory to GEM object
[  113.879330] systemd-journald[832]: File /var/log/journal/df5af8c0f46c4a4fb5dda265e9ac334e/user-1000.journal corrupted or uncleanly shut down, renaming and replacing.
[  116.730790] [drm:nvidia_drm_gem_import_nvkms_memory [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to import NVKMS memory to GEM object
[  116.742151] [drm:nvidia_drm_gem_import_nvkms_memory [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to import NVKMS memory to GEM object
[  116.742856] [drm:nvidia_drm_gem_import_nvkms_memory [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to import NVKMS memory to GEM object
[  124.816188] ------------[ cut here ]------------
[  124.816194] WARNING: CPU: 0 PID: 5 at drivers/gpu/drm/drm_atomic_helper.c:1682 drm_atomic_helper_commit_hw_done+0x93/0xa0 [drm_kms_helper]
[  124.816195] Modules linked in: nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) drm_kms_helper drm xt_CHECKSUM ipt_MASQUERADE nf_nat_masquerade_ipv4 tun cmac arc4 md4 nls_utf8 cifs ccm dns_resolver fscache nf_conntrack_netbios_ns nf_conntrack_broadcast xt_CT ip6t_rpfilter ip6t_REJECT nf_reject_ipv6 xt_conntrack ip_set nfnetlink ebtable_nat ebtable_broute bridge stp llc ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack libcrc32c iptable_mangle iptable_raw iptable_security ebtable_filter ebtables ip6table_filter ip6_tables sunrpc vfat fat fuse squashfs loop snd_hda_codec_hdmi intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm snd_hda_codec_realtek irqbypass
[  124.816215]  snd_hda_codec_generic intel_cstate iTCO_wdt iTCO_vendor_support snd_hda_intel intel_uncore snd_hda_codec intel_rapl_perf snd_virtuoso snd_hda_core snd_oxygen_lib snd_mpu401_uart snd_hwdep snd_rawmidi snd_seq joydev snd_seq_device snd_pcm tpm_tis tpm_tis_core tpm snd_timer snd mei_me mei soundcore i2c_i801 shpchp lpc_ich dm_crypt mxm_wmi crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel r8169 hid_roccat_lua hid_roccat_common mii video wmi
[  124.816228] CPU: 0 PID: 5 Comm: kworker/u16:0 Tainted: P           OEL  4.13.3-301.fc27.x86_64 #1
[  124.816229] Hardware name: MSI MS-7866/Z87M GAMING (MS-7866), BIOS V1.1 12/03/2013
[  124.816232] Workqueue: events_unbound commit_work [drm_kms_helper]
[  124.816233] task: ffff914a95278000 task.stack: ffffa20a80c90000
[  124.816236] RIP: 0010:drm_atomic_helper_commit_hw_done+0x93/0xa0 [drm_kms_helper]
[  124.816237] RSP: 0018:ffffa20a80c93db8 EFLAGS: 00010282
[  124.816237] RAX: ffff914a810de000 RBX: 0000000000000002 RCX: ffff914a829f5800
[  124.816238] RDX: ffff9149ef153400 RSI: 0000000000000001 RDI: ffff9149f272e408
[  124.816238] RBP: ffffa20a80c93dd8 R08: ffff914a9ed50a20 R09: ffffa20a81b93ce0
[  124.816238] R10: ffffa20a80c93d78 R11: 0000000000000065 R12: ffff914a8041d008
[  124.816239] R13: ffff9149f2536240 R14: ffff9149f272e408 R15: ffff9149f272e408
[  124.816240] FS:  0000000000000000(0000) GS:ffff914a9ec00000(0000) knlGS:0000000000000000
[  124.816240] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  124.816241] CR2: 00000b36073e3008 CR3: 0000000213dd4000 CR4: 00000000001406f0
[  124.816241] Call Trace:
[  124.816244]  nvidia_drm_atomic_helper_commit_tail+0x125/0x190 [nvidia_drm]
[  124.816247]  commit_tail+0x3f/0x60 [drm_kms_helper]
[  124.816250]  commit_work+0x12/0x20 [drm_kms_helper]
[  124.816252]  process_one_work+0x193/0x3c0
[  124.816253]  worker_thread+0x4a/0x3a0
[  124.816255]  kthread+0x125/0x140
[  124.816255]  ? process_one_work+0x3c0/0x3c0
[  124.816256]  ? kthread_park+0x60/0x60
[  124.816258]  ret_from_fork+0x25/0x30
[  124.816259] Code: f5 49 8d 7d 30 e8 6e 4c 6f f5 41 c6 84 24 28 04 00 00 00 49 8b 4e 08 83 c3 01 39 99 38 03 00 00 7f 9d 5b 41 5c 41 5d 41 5e 5d c3 <0f> ff eb c5 f3 c3 0f 1f 80 00 00 00 00 0f 1f 44 00 00 55 48 89 
[  124.816273] ---[ end trace 8fd6b330eb4d7bae ]---
[  125.415721] ------------[ cut here ]------------
[  125.415730] WARNING: CPU: 6 PID: 5 at drivers/gpu/drm/drm_atomic_helper.c:1682 drm_atomic_helper_commit_hw_done+0x93/0xa0 [drm_kms_helper]
[  125.415730] Modules linked in: nvidia_drm(POE) nvidia_modeset(POE) nvidia(POE) drm_kms_helper drm xt_CHECKSUM ipt_MASQUERADE nf_nat_masquerade_ipv4 tun cmac arc4 md4 nls_utf8 cifs ccm dns_resolver fscache nf_conntrack_netbios_ns nf_conntrack_broadcast xt_CT ip6t_rpfilter ip6t_REJECT nf_reject_ipv6 xt_conntrack ip_set nfnetlink ebtable_nat ebtable_broute bridge stp llc ip6table_nat nf_conntrack_ipv6 nf_defrag_ipv6 nf_nat_ipv6 ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack libcrc32c iptable_mangle iptable_raw iptable_security ebtable_filter ebtables ip6table_filter ip6_tables sunrpc vfat fat fuse squashfs loop snd_hda_codec_hdmi intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm snd_hda_codec_realtek irqbypass
[  125.415753]  snd_hda_codec_generic intel_cstate iTCO_wdt iTCO_vendor_support snd_hda_intel intel_uncore snd_hda_codec intel_rapl_perf snd_virtuoso snd_hda_core snd_oxygen_lib snd_mpu401_uart snd_hwdep snd_rawmidi snd_seq joydev snd_seq_device snd_pcm tpm_tis tpm_tis_core tpm snd_timer snd mei_me mei soundcore i2c_i801 shpchp lpc_ich dm_crypt mxm_wmi crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel r8169 hid_roccat_lua hid_roccat_common mii video wmi
[  125.415768] CPU: 6 PID: 5 Comm: kworker/u16:0 Tainted: P        W  OEL  4.13.3-301.fc27.x86_64 #1
[  125.415769] Hardware name: MSI MS-7866/Z87M GAMING (MS-7866), BIOS V1.1 12/03/2013
[  125.415773] Workqueue: events_unbound commit_work [drm_kms_helper]
[  125.415774] task: ffff914a95278000 task.stack: ffffa20a80c90000
[  125.415777] RIP: 0010:drm_atomic_helper_commit_hw_done+0x93/0xa0 [drm_kms_helper]
[  125.415778] RSP: 0018:ffffa20a80c93db8 EFLAGS: 00010286
[  125.415779] RAX: ffff914a7fe8c800 RBX: 0000000000000000 RCX: ffff914a829f5800
[... 11 lines cut ...]
[  125.415798]  process_one_work+0x193/0x3c0
[  125.415799]  worker_thread+0x4a/0x3a0
[  125.415800]  kthread+0x125/0x140
[  125.415801]  ? process_one_work+0x3c0/0x3c0
[  125.415803]  ? kthread_park+0x60/0x60
[  125.415805]  ret_from_fork+0x25/0x30
[... cut ...]
[  157.183951] [drm:nvidia_drm_gem_import_nvkms_memory [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to import NVKMS memory to GEM object
[  157.330102] [drm:nvidia_drm_dumb_create [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to map NvKmsKapiMemory 0xffff914a7ead3ec8
[  157.335781] [drm:nvidia_drm_dumb_create [nvidia_drm]] *ERROR* [nvidia-drm] [GPU ID 0x00000100] Failed to map NvKmsKapiMemory 0xffff914a7ead3688
[  158.198976] gnome-shell[7761]: segfault at 1a ip 00007ff96f32534e sp 00007ffc0d380d40 error 4 in nouveau_dri.so[7ff96f100000+ac0000]

yodatak, 2017-09-27:
Thanks for the anwser its weird because in fedora 26 with nvidia driver i got no issue with wayland but it append on Fedora 27

scaronni, 2017-10-05:
Thanks @leigh123linux  for replying.

closed 2017-10-05 by scaronni