negativo17/nvidia-driver #34: [Fedora 27] Only with proprietary drivers, only 1 image per 5 seconds

Gwend4l, 2017-10-10:
Configuration:
- CPU: AMD Athlon(tm) II X3 445 Processor
- Graphic card: GeForce GTX 770
- OS: Fedora 27 beta
- Driver version: nvidia-driver-387.12-1.fc27, installed with `sudo dnf install nvidia-driver
`

With a freshly installed Fedora 27 beta, I observe the following behavior
* with nouveau, no problem
* with proprietary drivers, only 1 image per 5 seconds is displayed, and the computer is completely unusable (I can only connect to it using ssh)

It also appears that half of the time, gnome-shell uses all the available CPU. But half of the time, CPU usage is correct, and there is still only 1 image displayed every 5 seconds.

The driver seems to be loaded:
~~~
$ lsmod | grep nvidia
nvidia_drm             45056  1
nvidia_modeset        897024  1 nvidia_drm
nvidia              13815808  4 nvidia_modeset
drm_kms_helper        159744  1 nvidia_drm
drm                   352256  4 nvidia_drm,drm_kms_helper
~~~

And here is everything nvidia related in `dmesg`:
~~~
╰─➤  dmesg | grep nvidia
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.13.5-300.fc27.x86_64 root=UUID=37f5d90f-dd33-4f8a-8080-cb61e4fb7a14 ro rhgb quiet rd.driver.blacklist=nouveau nvidia-drm.modeset=1
[    0.000000] Kernel command line: BOOT_IMAGE=/boot/vmlinuz-4.13.5-300.fc27.x86_64 root=UUID=37f5d90f-dd33-4f8a-8080-cb61e4fb7a14 ro rhgb quiet rd.driver.blacklist=nouveau nvidia-drm.modeset=1
[    5.284838] nvidia: loading out-of-tree module taints kernel.
[    5.284845] nvidia: module license 'NVIDIA' taints kernel.
[    5.295692] nvidia: module verification failed: signature and/or required key missing - tainting kernel
[    5.302847] nvidia-nvlink: Nvlink Core is being initialized, major device number 244
[    5.303636] nvidia 0000:01:00.0: vgaarb: changed VGA decodes: olddecodes=io+mem,decodes=none:owns=io+mem
[    5.337345] nvidia-modeset: Loading NVIDIA Kernel Mode Setting Driver for UNIX platforms  387.12  Thu Sep 28 19:30:23 PDT 2017
[    5.343881] [drm] [nvidia-drm] [GPU ID 0x00000100] Loading driver
[    5.986230] nvidia-modeset: Allocated GPU:0 (GPU-2f195007-5c5d-7696-7974-7204ca09558f) @ PCI:0000:01:00.0
[    6.064553] [drm] Initialized nvidia-drm 0.0.0 20160202 for 0000:01:00.0 on minor 0
~~~

If anyone has any idea, I'm interested :)

Gwend4l, 2017-10-10:
I just noticed that Wayland is running, maybe that's the problem?

On my other computer, I upgraded from Fedora 26 to 27, and the nvidia driver is perfectly working - but it appears to be still running Xorg.

Gwend4l, 2017-10-10:
OK, as a workaround I have set `WaylandEnable=false` in `/etc/gdm/custom.conf`, which successfully disabled Wayland, and not everything is working perfectly.

leigh123linux, 2017-10-10:
It's gnome that is broken not nvidia.
The high CPU usage is gnome using mesa fallback for wayland (software rendering).

https://bugzilla.rpmfusion.org/show_bug.cgi?id=4632

gdm/mutter should test to see if the driver has the required support for wayland to function.
Fedora gnome doesn't compile mutter with eglstream support so can't use nvidia for wayland.

Gwend4l, 2017-10-10:
I see, thanks for the piece of info :). So they should add better code to detect whether wayland can be used or not, if I understand correctly

Sorry for the extraneous bug report!

closed 2017-10-10 by Gwend4l

mati865, 2017-10-17:
I have forked and updated `mutter-eglstream` repository for Fedora 27 and Rawhide: https://copr.fedorainfracloud.org/coprs/mati865/mutter-eglstream/

CPU spikes occur but it's not that much worse than Xorg session.

oghaki, 2017-12-08:
Hi mati865, I am no longer able to access this repository.  I get a timeout error each time.

markscsmith, 2017-12-08:
oghaki, I'm not positive, but I think it's related to the fedora infra move: https://pagure.io/fedora-infrastructure/issue/6537

oghaki, 2017-12-08:
@markscsmith, thank you very much.  That saved me a lot of wasted energy tonight.