negativo17/nvidia-driver #37: Nouveau driver is still loaded despite apparently successful install of nvidia binaries, Fedora 27

timrichardson, 2017-11-20:
HI. I am new to Fedora. I have a clean install on a Thinkpad W520. The Quadro card in this old machine is still supported by the current binary releases, and I have a working Ubuntu install on the same machine. 
The w520 has three graphics settings in BIOS: Intel, Nvidia ('Discrete') or Hybrid. The problems I'm having occur in both Hybrid and Discrete modes.
I have never succeeded with this. Currently I have this version 2:387.22-2.fc27 with an update date of Nov 19 2017. 

I followed the instructions to add the repository. In the GUI Software app, the Nvidia driver appears and there are no errors when it installs. 
Upon reboot, the font of the console is large and ugly, a sign of nvidia. However, during the boot process it reverts to small font, and I can see a boot message that a Nouveau fallback service has started. 

I do not know my way around the log files very well, I have no clues about why Nouveau is not being blacklisted or if there is a problem with the Nvidia driver.

goncalossilva, 2017-11-22:
I had the same issue. Reinstalling didn't work.

I ended up removing everything and installing again. Specifically:

`dnf remove nvidia-driver nvidia-driver-cuda nvidia-settings akmod-nvidia kernel-devel`

And then:

`dnf install nvidia-driver nvidia-driver-cuda`

Rebooted and everything worked again. 😄

ufa, 2017-11-25:
I tried to reinstall, but nouveau keeps loading. 
I'm on Fedora 27, fresh install

timrichardson, 2017-11-27:
After an update which I applied today, but which may be a few days old, Optimus with Nvidia drivers just started working. Not only that, but for the first time I have gdm3 and gnome working with the nvidia driver, under any OS.

closed 2017-11-27 by timrichardson