negativo17/nvidia-driver #38: Support for CUDA 8.0 and Fedora 27 onwards for Fermi GPUs (SM 2.0/2.1)?

simonlui, 2017-11-21:
Hi,

Let me start by thanking you for your hard work on this repository. It makes dealing with Nvidia proprietary drivers a lot less painful and I am grateful for the hard work that has been put in this.

Is there any way for CUDA 8 to be preserved going forward for Fermi generation of GPUs? I know they are pretty old by now but I am still working with one and just recently updated to Fedora 27, and I only see the option to use CUDA 9 and nothing else. What are your plans for this going forward?

I have no choice but to work around this by making a new repository of your Fedora 26 repository where CUDA 8 still exists and then blocking all updates going forward unless you plan to provide support somehow.

simonlui, 2017-11-27:
Just an update.

I have found some success setting up and using Nvidia's docker solution at https://github.com/NVIDIA/nvidia-docker for my needs so this isn't a big issue anymore for me personally. I do still want to know if you plan on deprecating Fermi CUDA support like Nvidia is with CUDA 9.0 only going forward.

scaronni, 2017-12-15:
Hello, sorry for the late reply. At the moment I have no plan to support more than one CUDA version on the system, and this is due to the fact that most of the libraries that use CUDA components are tied to the version they are built for (ffmpeg, blender, ccminer, mpv, etc.). What you can do, is just edit your repository file in `/etc/yum.repos.d` and remove the `$releasever` tag to use the Fedora 26 repository. The packages work fine on Fedora 27. This way you will stick with CUDA 8.

Or just mirror the repository somewhere and use the packages locally.

I can check if I can make compat-gcc53 compile on Fedora 27, if you need. Maybe I should rename it as `cuda-gcc` and rebase it on 5.4; so it would work on Fedora 27 as well.

simonlui, 2017-12-15:
Ah okay. Thank you for clarifying and explaining.

I am, as I stated above, currently using Nvidia's docker solution with CUDA 8 and using your latest CUDA and drivers which work out for now, and is a bit less hassle on my part since it's somewhat finnicky to mix and match your older packages together from different repositories.

I don't think either of those solutions with using your older packages or Nvidia's docker will work for long on Fermi cards but hopefully it should last long enough for me before I replace this machine.

Thanks once again. I will be closing this issue.

closed 2017-12-15 by simonlui