negativo17/nvidia-driver #39: Nvidia driver not working after kernel update Fedora 27 kernel 4.14.13-300.fc27.x86_64

gdubya, 2018-01-15:
Hi,
After updating the kernel, I can no longer get the nvidia driver to work.

DKMS status appears ok, but nvidia-settings says that the driver is not in use:

> $ dkms status
> nvidia, 387.34, 4.14.13-300.fc27.x86_64, x86_64: installed

Laptop is a Thinkpad P90 with a Quadro M3000M graphics card:

> lspci |grep -E "VGA|3D"
> 01:00.0 VGA compatible controller: NVIDIA Corporation GM204GLM [Quadro M3000M] (rev a1)

How should i debug this further?

Thanks for the help!

Saroufim, 2018-02-18:
I also am experiencing this issue. It's still valid with kernel 4.15. The driver isn't being loaded.

Saroufim, 2018-02-19:
Turns out that akmods was installed from RPMFusion and not the negativo17 repository. Switching repositories and forcing rebuilds fixed the problem.

closed 2018-02-21 by scaronni