negativo17/nvidia-driver #41: Can't login via gui after upgrading nvidia driver from 387.34 -> 390.12

xd009642, 2018-01-22:
Apologies if this is the wrong place to put this on Sunday I ran dnf update and accepted all the updates (I have the nvidia drivers, cuda and development libraries installed). After this, I found myself in a login loop where I attempt to log in, screen goes black and then it goes back to the login screen. 

So far I've tried reinstalling the drivers and development libraries. I've also tried uncommenting WaylandEnable=false in /etc/gdm/custom.conf but instead of a loop after logging in I was just met with a black screen.

I'm running Fedora 27 and the GPU is an Nvidia geforce GTX 960, also here are the outputs of dmesg and journalctl -r https://gist.github.com/xd009642/f4bc360cbb17e48a242f42b047706b44 

I'm tempted to try do dnf downgrade all packages but I've seen other people online who have had issues after updates of nvidia drivers and downgrade hasn't fixed it and it feels a bit bruteforce so I wondered if you had any guidance?

garyraposo, 2018-02-02:
Same problem on my system...

Fedora 27 (14.14-300.fc27.x86_64)
GeForce GT 640 (GK107) using HDMI connection
WaylandEnable=false

After installing 390.12 or 390.25 the system boots to a black screen. I can SSH in and even connect via VNC so I know the system is up... just a black screen instead of the desktop login.

Booting up and shutting down works fine (I can see the various messages scroll by) so I'm fairly confident it's not a hardware/connection problem.

Downgrading to 387.34 fixes the problem and the system operates as expected.

xd009642, 2018-02-02:
I've already tried downgrading and it didn't fix it for me. Going to try uninstalling and then reinstalling to see if that fixes it

edgan, 2018-02-06:
I had to blacklist nouveau after upgrading to 390.25, and reboot. I didn't need the nvidia driver in my initramfs.

    cat /etc/modprobe.d/blacklist-nouveau.conf 
    blacklist nouveau
    alias nouveau off

garyraposo, 2018-02-07:
Unfortunately, nouveau was already blacklisted on my system so that's not the fix for my black screen.

garyraposo, 2018-02-23:
Got the black screen to go away by removing the following from the screen section of my xorg configuration file:

`Option	"metamodes" "nvidia-auto-select +0+0 {ForceCompositionPipeline=On, ForceFullCompositionPipeline=On}"`

However... now I have tearing on video with fast moving action. The whole reason I set this metamode option was to deal with the tearing.

If we can't use this metamode option then what are some other strategies for eliminating tearing?

scaronni, 2018-11-05:
Sorry, too many changes since then. Reopen if still occurring.

closed 2018-11-05 by scaronni