negativo17/nvidia-driver #43: Kernel / user-space driver mismatch after upgrading to 390.25

danielkza, 2018-02-02:
After upgrading to version 390.25 in Fedora 27, I cannot boot to the GUI:

```
NVRM: API mismatch: the client has the version 390.25, but
NVRM: this kernel module has the version 390.12.  Please
NVRM: make sure that this kernel module and all NVIDIA driver
NVRM: components have the same version.
```

Thinking that it might have been some DKMS hiccup, I removed the nvidia module completely, reinstalled dkms-nvidia, and rebuilt the driver for all the kernels. The warning still persisted.

Downgrading all the packages to 390.12 fixes the issues.

scaronni, 2018-02-02:
Hello, something wrong with your system. The 390.25 packages are there, and contain 390.25 sources:

```
# modinfo nvidia nvidia_drm nvidia_modeset nvidia_uvm nvidia | grep ^version
version:        390.25
version:        390.25
version:        390.25
version:        390.25
# rpm -qa \*nvidia\* | sort
dkms-nvidia-390.25-1.fc27.x86_64
nvidia-driver-390.25-1.fc27.x86_64
nvidia-driver-cuda-390.25-1.fc27.x86_64
nvidia-driver-cuda-libs-390.25-1.fc27.i686
nvidia-driver-cuda-libs-390.25-1.fc27.x86_64
nvidia-driver-libs-390.25-1.fc27.i686
nvidia-driver-libs-390.25-1.fc27.x86_64
nvidia-driver-NVML-390.25-1.fc27.x86_64
nvidia-libXNVCtrl-390.25-1.fc27.x86_64
nvidia-persistenced-390.25-1.fc27.x86_64
nvidia-settings-390.25-1.fc27.x86_64
```

danielkza, 2018-02-02:
What could be wrong with my system that would persist after completely removing the module from the DKMS tree multiple times, and would not manifest again after the downgrade? I really would like to debug the issue so that I (or others) don't hit it again.

leigh123linux, 2018-02-02:
Maybe the old nvidia.ko got included in initramfs.

scaronni, 2018-02-02:
Good point! But if that is the case, this would mean that this:

```
$ cat /usr/lib/dracut.conf.d/99-nvidia-dracut.conf 
# omit the nvidia driver from the ramdisk, to avoid needing to regenerate
# the ramdisk on nvidia driver updates
omit_drivers+=" nvidia-drm "
```
is no longer sufficient now.

@danielkza can you try to execute this and paste here the results?
```
$ sudo lsinitrd | grep -i extra/nvidia
```
Thanks.

scaronni, 2018-02-02:
Uff. I just discovered that the nvidia modules are put inside the initrd even if the modules are explicitly omitted, they are loaded through this file: `/usr/lib/dracut/modules.d/50drm/module-setup.sh`

I'm checking if this has changed recently in dracut. As much as they have a special case for `radeon`, I guess we might need a specific entry for `nvidia`.

scaronni, 2018-02-02:
But the funny thing is that if you pass `--omit-drivers` on the `dracut` command line, those are not added. I think there is a bug somewhere.

scaronni, 2018-02-02:
Don't know when this has changed, but the dracut directory is *not* `%{_prefix}/lib/dracut.conf.d` but ``%{_prefix}/lib/dracut/dracut.conf.d`.

I'm checking if this is the case also on RHEL 6/7.

scaronni, 2018-02-02:
@leigh123linux I confirm, moving the file in `%{_prefix}/lib/dracut/dracut.conf.d` fixes it, so I guess you should do the same for the RPMFusion packages. The folder is the latter also on RHEL/CentOS.

I will push updated packages shortly. Somehow in the last years after adding the module exclusion to the original package, I never had the need to rebuild the initrd.

closed 2018-02-02 by scaronni

danielkza, 2018-02-03:
My previous initrd did contain the NVIDIA kernel modules. Upgrading the packages and rebuilding the initrd for all my kernels seems to have fixed everything. Thank you for digging further into the issue.