negativo17/nvidia-driver #44: Availability of older versions

CRCinAU, 2018-02-07:
Firstly, thanks for the awesome work on these drivers - this is the first time in years I've had to actually troubleshoot an issue.

I noticed with the last two updates after 390.12 that I get very bad response from the driver. Transitions such as the fade between desktop wallpapers in KDE brings the entire system to a crawl while this is animated. I also noticed very bad lag when it comes to video playback - even just YouTube.

I did note than 390.12 seemed to work fine.

Is there any archive of past packages that I can downgrade to 390.12 to confirm the issue, then try newer versions as they appear (and downgrading if the problem still persists)?

CRCinAU, 2018-02-10:
For what its worth, links to problems in case people find the same problem:

https://devtalk.nvidia.com/default/topic/1029484/linux/-linux-mint-18-3-numerous-performance-amp-rendering-issues-on-390-25/

https://www.reddit.com/r/archlinux/comments/7w2jnp/terrible_performance_regression_with_nvidia_39025/

https://bbs.archlinux.org/viewtopic.php?id=234241

pkovacs, 2018-02-18:
387.34 was smooth for me on multiple f27 machines.  I upgraded to 390.25 and now I get several bright flashes when the driver starts -- like a light bulb popping.  I feel it can damage my panels.  I really want to downgrade to 387.34, but now I can't.

scaronni, 2018-02-21:
387.34 is still in the repository. There are the last 3 builds of each package available in the Nvidia repository.

closed 2018-02-21 by scaronni

CRCinAU, 2018-02-21:
Ahhh, can these please be duplicated into the multimedia repo?

Seems there are a difference between the 390.25 packages there, as well as only 390.25-1 and 390.25-2.

scaronni, 2018-02-21:
Well the multimedia repo just had 2 versions because it's pretty big, but I will re-enable the 3 copies history in the next push.

pkovacs, 2018-02-21:
```
# dnf clean all
# dnf repoquery --show-duplicates fedora-nvidia
negativo17 - Nvidia                                                                                 378 kB/s | 296 kB     00:00
Fedora 27 - x86_64 - Updates                                                                        7.9 MB/s |  19 MB     00:02    
Fedora 27 - x86_64                                                                                  9.2 MB/s |  58 MB     00:06 
google-chrome                                                                                        52 kB/s | 3.7 kB     00:00
Last metadata expiration check: 0:00:14 ago on Wed 21 Feb 2018 11:14:21 AM EST.
nvidia-driver-2:390.12-1.fc27.x86_64
nvidia-driver-2:390.25-1.fc27.x86_64
nvidia-driver-2:390.25-2.fc27.x86_64
```

CRCinAU, 2018-02-22:
If space is an issue, could I suggest hard-linking files that are the same?

This would reduce the space required on the hosting end to just the size of the packages and not another copy of them per repo.

CRCinAU, 2018-02-22:
Also, I can't see what @pkovacs  gets:

```
# dnf repoquery --show-duplicates fedora-nvidia
Copr repo for keepassxc-nightly owned by hifi                                                                                                                                                                 23 kB/s |  26 kB     00:01
negativo17 - Multimedia                                                                                                                                                                                       93 kB/s | 682 kB     00:07
negativo17 - Nvidia                                                                                                                                                                                          147 kB/s | 296 kB     00:02
Fedora 27 - x86_64 - Test Updates                                                                                                                                                                            4.6 MB/s | 4.3 MB     00:00
Fedora 27 - x86_64 - Updates                                                                                                                                                                                 4.7 MB/s |  19 MB     00:04
Fedora 27 - x86_64                                                                                                                                                                                           5.0 MB/s |  58 MB     00:11
google-chrome                                                                                                                                                                                                 15 kB/s | 3.7 kB     00:00
RPM Fusion for Fedora 27 - Free - Updates                                                                                                                                                                     60 kB/s | 291 kB     00:04
RPM Fusion for Fedora 27 - Free                                                                                                                                                                              1.5 MB/s | 717 kB     00:00
RPM Fusion for Fedora 27 - Nonfree - Updates                                                                                                                                                                 4.4 kB/s |  27 kB     00:06
RPM Fusion for Fedora 27 - Nonfree                                                                                                                                                                           401 kB/s | 163 kB     00:00
Last metadata expiration check: 0:00:00 ago on Thu 22 Feb 2018 22:05:43 AEDT.
```

pkovacs, 2018-02-22:
@CRCinAU That's a cut-paste error on my part.  I'll edit my comment to correct it. Try this:
```
 dnf repoquery --show-duplicates nvidia-driver
```

CRCinAU, 2018-02-22:
Ahhh gotcha... My bad - I thought it was to show dupes in that specific repo... Makes sense when I think about it :)