negativo17/nvidia-driver #45: PRIME Sync doesn't work on 390-25 and kernel 4.15 (patch provided)

junrrein, 2018-02-17:
First of all, thanks for the work you put into packaging the Nvidia drivers! It really makes life simpler for a lot of people.

I'm running Fedora 27. Prime Sync used to work on kernel 4.14 (after enabling KMS for the Nvidia Driver) but, after upgrading to kernel 4.15, it doesn't work anymore.

A Google search shows this is a known problem with the Nvidia installer not being adequate for some kernel internal API changes.

You can see a short discussion about this, and a patch that seems to fix the issue, at https://bugs.archlinux.org/task/57305#comment166449

leigh123linux, 2018-02-21:
@scaronni 

It's really easy to test to see if the patch is needed, run while running 4.15 kernel

```
ls -1 /sys/class/drm
```
It should return

```
 leigh  ~  ls -1 /sys/class/drm
card0
card0-DP-1
card0-DVI-D-1
card0-HDMI-A-1
renderD128
version
 leigh  ~ 
```

https://pkgs.rpmfusion.org/cgit/nonfree/nvidia-kmod.git/commit/?id=e3618d0216f075a28a976a58105ada0b343c240b

https://devtalk.nvidia.com/default/topic/1029625/linux/no-sys-class-drm-card0-entries-on-kernel-4-15

scaronni, 2018-02-21:
Thanks! Packages being built now. They will be online soon.

closed 2018-02-21 by scaronni