negativo17/nvidia-driver #49: PRIME Synchronization doesn't work since update to 390.42

kazysmaster, 2018-03-19:
I own an optimus laptop and the prime synchronization was working fine with nvidia version 390.25. At least on eDP output. I didn't test on HDMI. Since the update to  390.42 it stopped working.  The modeset option is still turned on:

cat /etc/modprobe.d/nvidia.conf 
blacklist nouveau
# Uncomment the following line to enable kernel modesetting support.
# There is NO graphical framebuffer (like OSS drivers) at the moment; this is
# only for Wayland. Leave disabled if you plan to use SLI.
options nvidia-drm modeset=1

sudo cat /sys/module/nvidia_drm/parameters/modeset
Y 

however xrandr --verbose  for eDP-1-1 shows:
PRIME Synchronization: 0 
		supported: 0, 1

Running:
xrandr --output eDP-1-1 --set "PRIME Synchronization" 1
does not help

Could it be a packaging problem or rather a driver issue?

leigh123linux, 2018-03-19:
post

```
ls -1 /sys/class/drm
```

kazysmaster, 2018-03-20:
ls -1 /sys/class/drm
card0
card0-eDP-1
card0-HDMI-A-1
card1
renderD128
renderD129
version

chriscjs, 2018-03-21:
The same 4.15 patch for 390.25 also works with 390.42.  I rebuilt the akmod source rpm after applying the patch and tested it.

scaronni, 2018-03-21:
Does it work with the patch?

scaronni, 2018-03-21:
I will re-apply the patch.

closed 2018-03-21 by scaronni

kazysmaster, 2018-03-21:
Thank you, it works now.