negativo17/nvidia-driver #50: OpenCL not working.

jhilliar, 2018-03-22:
We have recently found a failure with OpenCL when running Maya (Autodesk).
We found that if we install the binary driver from nvidia, instead of the packaged drivers, we don't have the issue. We verified this by testing/comparing versions 384.90, 390.25, and 390.42 on our RHEL7.4 workstations.

The binary driver is installed with these options: " --accept-license --ui=none --no-questions --disable-nouveau --install-compat32-libs --no-precompiled-interface -j20 --install-libglvnd --dkms

Our packaged drivers get installed using this package list:
"dkms-nvidia nvidia-driver nvidia-driver-cuda nvidia-driver-libs nvidia-driver-cuda-libs nvidia-driver-NVML nvidia-driver-NvFBCOpenGL nvidia-driver-libs.i686 nvidia-driver-cuda-libs.i686 nvidia-driver-NVML.i686 nvidia-driver-NvFBCOpenGL.i686 nvidia-persistenced nvidia-libXNVCtrl nvidia-settings nvidia-modprobe libglvnd-devel libglvnd-devel.i686 ocl-icd-devel"

Maya launches in all cases, but we get this error (last line) and our opencl tests fail in Maya.

`OpenCL evaluator is attempting to initialize OpenCL.
Detected 1 OpenCL Platforms: 
 0: NVIDIA Corporation. NVIDIA CUDA. OpenCL 1.2 CUDA 9.1.84.
 Supported extensions: cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_khr_gl_event cl_nv_create_buffer
OpenCL evaluator choosing OpenCL platform NVIDIA Corporation.
Error: found cl_khr_gl_sharing but failed to get function pointers
`
I'd like to get to the bottom of this so I don't have to switch to the binary installer on all our systems (getting pressure to do that), so any help is greatly appreciated!

Meanwhile, I'll try to put together an opencl test to see if we can reproduce an issue using open source software (but luxmark benchmark runs fine on both the binary and rpm-based drivers) So this appears to be specific to cl_khr_gl_sharing which the benchmark may not test.

jhilliar, 2018-03-22:
Just to clarify a bit:
With the driver installed via rpms, Maya gives us this error:
   Error: found cl_khr_gl_sharing but failed to get function pointers
Using Nvidia's installer (The same version as the rpm based driver above) we get this instead:
   Choosing OpenCL Device Quadro K5000.  Device Type: GPU  Device is available.

jhilliar, 2018-03-22:
A sorted package list of each host is attached here so a diff will work. Ellingsen has the RPM based driver, amundsen has the driver from nvidia's installer.
[pkgs_amundsen.txt](https://github.com/negativo17/nvidia-driver/files/1838956/pkgs_amundsen.txt)
[pkgs_ellingsen.txt](https://github.com/negativo17/nvidia-driver/files/1838957/pkgs_ellingsen.txt)

jhilliar, 2018-03-23:
> This is not useful as the installer does not install any package. It just
> installs the files on the filesystem.
> I guess the difference is in the libOpenCL libraries, with the package
> you are using the ones provided by EPEL. I have access to a couple of
> systems with an old Maya version and this does not happen.
> Can you make a test? Extract the Nvidia binary installer somewhere (or
> just copy over the files from the workstation with the binary installer)
> and replace the OpenCL library:
> /usr/lib64/libOpenCL.so.1.0.0
> Then reload Maya and let me know.

Awesome! That was it. The ocl-icd package from EPEL fails.
I replaced it's /usr/lib64/libOpenCL.so.1.0.0 with the one from Nvidia's
binary installer and the problem disappeared.
Is this an EPEL/ocl-icd bug, perhaps?  Could you explain a bit for me the
split between the driver and EPEL for ocl?

Thanks again, I should have thought to try that.  Shame on me.
How best to proceed in your opinion?

Jay Hilliard

scaronni, 2018-03-23:
I think the best approach is to update the OpenCL loader to 2.2.12 (from 2.2.11) which has support for OpenCL 2.2. I think this should be the same. A quick build here, shows that it can be pushed as an update without breaking other stuff:

```
$ abipkgdiff ocl-icd-2.2.11-4.fc27.x86_64.rpm ocl-icd-2.2.12-1.fc27.x86_64.rpm
================ changes of 'libOpenCL.so.1.0.0'===============
  Functions changes summary: 0 Removed, 0 Changed, 0 Added function
  Variables changes summary: 0 Removed, 0 Changed, 0 Added variable
  Function symbols changes summary: 0 Removed, 2 Added function symbols not referenced by debug info
  Variable symbols changes summary: 0 Removed, 0 Added variable symbol not referenced by debug info

  2 Added function symbols not referenced by debug info:

    clSetProgramReleaseCallback@@OPENCL_2.2
    clSetProgramSpecializationConstant@@OPENCL_2.2

================ end of changes of 'libOpenCL.so.1.0.0'===============
```

Also because the library provides the various profiles:
```
$ rpm -qp --provides ocl-icd-2.2.11-4.el7.x86_64.rpm | grep OPENCL
libOpenCL.so.1(OPENCL_1.0)(64bit)
libOpenCL.so.1(OPENCL_1.1)(64bit)
libOpenCL.so.1(OPENCL_1.2)(64bit)
libOpenCL.so.1(OPENCL_2.0)(64bit)
libOpenCL.so.1(OPENCL_2.1)(64bit)
$ rpm -qp --provides ocl-icd-2.2.12-1.el7.x86_64.rpm | grep OPENCL
libOpenCL.so.1(OPENCL_1.0)(64bit)
libOpenCL.so.1(OPENCL_1.1)(64bit)
libOpenCL.so.1(OPENCL_1.2)(64bit)
libOpenCL.so.1(OPENCL_2.0)(64bit)
libOpenCL.so.1(OPENCL_2.1)(64bit)
libOpenCL.so.1(OPENCL_2.2)(64bit)
```
I will ask you to make another test with proper packages, ok? So we can see if it actually helps.

scaronni, 2018-03-23:
Can you please try this build?

https://negativo17.org/temp/

Just install over the packages you already have on the RHEL/CentOS system. If you don't have the headers the `ocl-icd` package is enough:

```
rpm -Uvh https://negativo17.org/temp/ocl-icd-2.2.12-1.el7.centos.x86_64.rpm
```

jhilliar, 2018-03-23:
Thanks Simone,
Unfortunately I get the same error with the updated ocl-icd package as I
did with the previous version.

[jhilliar@vanisher lib64]$ rpm -qa ocl-icd\*
ocl-icd-2.2.12-1.el7.centos.x86_64

I did the binary replacement afterwards to verify that fixes things again.

Jay

scaronni, 2018-03-25:
That's weird. The OpenCL loader should be vendor agnostic (it's even written in Nvidia's README file inside the driver tarball).

By googling around, it seems this error is only on Maya. Which Maya version are you using?

scaronni, 2018-03-25:
> Meanwhile, I'll try to put together an opencl test to see if we can reproduce an issue using open source software (but luxmark benchmark runs fine on both the binary and rpm-based drivers) So this appears to be specific to cl_khr_gl_sharing which the benchmark may not test.

My guess is that Maya is doing something not really compliant when performing the OpenCL tests. It did the same when Nvidia introduced GLVND support in the libraries, old versions (2106) were failing in the OpenGL tests.

FDreesen, 2019-07-19:
Just wanted to let you know that we were also still running into the exact same issue with current versions on our CentOS 7 systems (nvidia driver 410.78/430.26/430.34, ocl-icd 2.2.12, Maya 2018.5/2018.6/2019.0/2019.1). 
Yet the workaround with the _libOpenCL.so.1.0.0_ from the binary installer also still worked fine for us. 
Since it seemed that this might be a very Maya-specific issue, we did not replace the global lib but used the _lib_ subfolder inside the Maya installation(s) which allows placing libraries or links as overrides to use for given Maya installation. 
In our case we placed a link inside named _libOpenCL.so.1_ linking to the _libOpenCL.so.1.0.0_ extracted from the binary installer and afterwards it loaded up fine and OpenCL features (eg GPU Override) worked fine inside Maya.

LightColoredLines, 2019-09-23:
Hello. I have follow your solution. And i just installed new Nvidia driver (435.21) from Ubuntu Software Center. And i got this error in Maya 2019:

```
}
OpenCL evaluator is attempting to initialize OpenCL.
OpenCL evaluator failed to find OpenCL device with vendor NVIDIA Corporation.
```

Is it possible to get my OpenCL worked with this nvidia driver (from Software Center version)?
or is it only worked with Nvidia's Installer?

FDreesen, 2019-09-23:
Sorry, I can't really answer that question. We don't have Ubuntu running here and if I'm informed correctly the driver in this repo is only done for rpm based distros. I think it is essentially the one from nVidia repackaged as rpms. And I have no idea which driver is installed via Ubuntu's Software Center.

LightColoredLines, 2019-09-23:
i have tried to install with Nvidia's Installer. And it worked fine. So Nvidia's Installer is better to support Maya than the one from software center.

LightColoredLines, 2019-09-24:
everything worked fine. but i got this thing (error or else). Is it a problem caused by OpenCL or not?

```
~$ maya
Initialized VP2.0 renderer {
  Version : 2016.11.53.12. Feature Level 5.
  Adapter : GeForce GTX 1050 Ti/PCIe/SSE2
  Vendor ID: 4318. Device ID : 
  Driver : 4.6.0 NVIDIA 435.21.
  API : OpenGL V.4.6.
  Max texture size : 16384 * 16384.
  Max tex coords : 32
  Shader versions supported (Vertex: 5, Geometry: 5, Pixel 5).
  Shader compiler profile : (Best card profile)
  Active stereo support available : 0
  GPU Memory Limit : 4096 MB.
  CPU Memory Limit: 7523.94 MB.
  MultiDraw consolidation: enabled
}
OpenCL evaluator is attempting to initialize OpenCL.
Detected 1 OpenCL Platforms: 
 0: NVIDIA Corporation. NVIDIA CUDA. OpenCL 1.2 CUDA 10.1.0.
 Supported extensions: cl_khr_global_int32_base_atomics cl_khr_global_int32_extended_atomics cl_khr_local_int32_base_atomics cl_khr_local_int32_extended_atomics cl_khr_fp64 cl_khr_byte_addressable_store cl_khr_icd cl_khr_gl_sharing cl_nv_compiler_options cl_nv_device_attribute_query cl_nv_pragma_unroll cl_nv_copy_opts cl_nv_create_buffer 
OpenCL evaluator choosing OpenCL platform NVIDIA Corporation.
Choosing OpenCL Device GeForce GTX 1050 Ti.  Device Type: GPU  Device is available.

**file:///usr/autodesk/maya2019/resources/builtin.toolclips/ui/ResourceList.qml:175:11: Unable to assign double to QQuickAnchorLine**
```

pberto, 2021-07-05:
I am putting this comment here in case I will forget: using nvidia driver from debian buster-backports, v460.73.01, all I needed to do was to delete the maya prefs in the user folder.

closed 2024-12-09 by scaronni