negativo17/nvidia-driver #56: Module not being rebuilt after kernel upgrade?

madig, 2018-08-18:
For a while now, I have to manually do `sudo dkms install nvidia/xxx.xx` after a kernel upgrade (Fedora 28). Why is this not happening automatically anymore?

jp7677, 2018-08-18:
I had an issue like this too. Turns out that dkms was somewhat confused on my machine. Pleas check with `dkms status` if the nvidia modules are listed. It didn't on my machine.  I had to cleanup some older folders in  `/var/lib/dkms/nvidia/`. I just deleted all folders there that correspond to nvidia driver version that I had no longer installed.

After doing that `dkms status` should look like:
```
#dkms status
nvidia, 396.45, 4.17.11-200.fc28.x86_64, x86_64: installed
nvidia, 396.45, 4.17.12-200.fc28.x86_64, x86_64: installed
nvidia, 396.45, 4.17.14-202.fc28.x86_64, x86_64: installed
```

madig, 2018-08-18:
Ah!
```
$ dkms status
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/nvidia/390.48/source/dkms.conf does not exist.
```
I `rm -rf`ed `/var/lib/dkms/nvidia/` and re-did `sudo dkms install nvidia/396.45` and now I get
```
$ dkms status
nvidia, 396.45, 4.17.14-202.fc28.x86_64, x86_64: installed (original_module exists)
```
Might be nice to do a sanity check there somehow...

scaronni, 2018-08-19:
There is an upstream issue with DKMS itself, if the kernel is upgraded along with a DKMS module package, it gets confused.

closed 2018-08-19 by scaronni

sbrl, 2020-06-16:
I can confirm that this is still a problem on Fedora 31.