negativo17/nvidia-driver #58: 390.87 Driver Released

juxi500, 2018-09-06:
Hello, I just wanted to let you know that NVIDIA 390.87 Linux Driver was released, it may fix some issues with Vulkan and OGL.

Thank you very much for providing this great repo and have a nice day :)

closed 2018-09-07 by scaronni

leigh123linux, 2018-09-07:
396.54 has the same fix

https://github.com/negativo17/nvidia-driver/commit/3ca3bdc43d19be8fb1a0538d38ab5284f299ea6d

juxi500, 2018-09-07:
But 390.87 is the long lived branch driver

Saroufim, 2018-09-14:
Not only that but it is also a legacy driver. Those of us who have devices with legacy cards might need this.

leigh123linux, 2018-09-14:
@Saroufim  We develop for the cards we use, TBH there hasn't been much interest in 390xx

https://bugzilla.rpmfusion.org/show_bug.cgi?id=4928

3 months and still no interest/volunteer to package it.