negativo17/nvidia-driver #59: [Question] Nvidia and Gnome on Wayland

jeffshee, 2018-09-09:
Hi! 
I'm experiencing screen tearing problem on Xorg. It will be nice if I can switch to Wayland, while still be able to use proprietary Nvidia driver. I thought following the steps listed here [https://copr.fedorainfracloud.org/coprs/mati865/mutter-eglstream/](https://copr.fedorainfracloud.org/coprs/mati865/mutter-eglstream/) would be enough for everything to work on Wayland, but it seems I still lack something.

nvidia-settings
```ERROR: Unable to find display on any available system```

mutter --wayland
```Can't initialize KMS backend: Could not get session ID: No such file or directory```

DRI_PRIME=1 inxi -G
```
Graphics:
  Device-1: Intel Skylake GT2 [HD Graphics 520] driver: i915 v: kernel 
  Device-2: NVIDIA GM107M [GeForce GTX 950M] driver: nvidia v: 396.45 
  Display: wayland server: Fedora Project X.org 11.0 
  driver: modesetting,nvidia unloaded: nouveau resolution: 1920x1080~60Hz 
  OpenGL: renderer: llvmpipe (LLVM 6.0 256 bits) v: 3.3 Mesa 18.0.5 
```

DRI_PRIME=1 glxinfo | grep vendor
```
libGL error: failed to create dri screen
libGL error: failed to load driver: nouveau
server glx vendor string: SGI
client glx vendor string: Mesa Project and SGI
OpenGL vendor string: VMware, Inc.
```

Doesn't this mean that Nvidia isn't being used on Wayland? Sorry I'm quite new about these... Please kindly advise.

leigh123linux, 2018-09-09:
What is DRI_PRIME used for?, because nvidia doesn't use it!
You will need to read up on xrandr and how to use it for switching optimus source.

https://devtalk.nvidia.com/default/topic/957814/linux/prime-and-prime-synchronization/

Also don't expect glxinfo or nvidia-settings to work or provide useful info as they are X tools, nvidia currently has no Xwayland support so both these tools will fallback to mesa under nvidia wayland.
The best way to test for wayland is to run the xrandr command.

jeffshee, 2018-09-09:
`DRI_PRIME`, isn't that variable to use discrete GPU? According to [Arch Wiki here](https://wiki.archlinux.org/index.php/PRIME#PRIME_GPU_offloading) , or did I misunderstood anything?
> nvidia currently has no Xwayland support

Does this mean applications or games which use X simply don't work with Nvidia on Wayland?

> The best way to test for wayland is to run the xrandr command

Could you explain how to test it, so that I can confirm my setup is working or not? 

Also, I'm quite confused with the current state of Nvidia + Wayland, if the things still aren't ready yet for now, then guess I should stay with X and wait for it...

leigh123linux, 2018-09-09:
If you read the arch wiki properly you will see "Note: For the proprietary NVIDIA driver see [1] instead."

http://us.download.nvidia.com/XFree86/Linux-x86/375.26/README/randr14.html

Under nvidia any apps that require xwayland fallback to mesa cpu rendering.
To test for wayland, run 'xrandr' and it should mention wayland in the first or second line.

TBH I would stick with Xorg till gnome ups it's game.