negativo17/nvidia-driver #60: Add support for "special legacy NVIDIA GPU driver releases... The 390.xx driver supports the following set of GPUs:"

I-Iawk, 2018-09-20:
It appears that video cards in the following list: https://download.nvidia.com/XFree86/Linux-x86_64/396.54/README/supportedchips.html are no longer supported in via the unified driver set you provide.

> Below are the legacy GPUs that are no longer supported in the unified driver. These GPUs will continue to be maintained through the special legacy NVIDIA GPU driver releases.

> The 390.xx driver supports the following set of GPUs

Are you able to provide the alternative driver set so we don't need to manually install the 390.xx driver set currently?

leigh123linux, 2018-10-15:
rpmfusion has 390xx in the repo

http://download1.rpmfusion.org/nonfree/fedora/updates/testing/28/x86_64/repoview/index.html

I-Iawk, 2018-10-31:
Odd, I don't see what you see...

http://download1.rpmfusion.org/nonfree/fedora/updates/testing/29/x86_64/repoview/ shows nothing...

I-Iawk, 2018-11-02:
Interesting... that said, it appears fedora-workstation-repositories package does not have the 390xx drivers though it's referencing the rpmfusion's repos... 

Since I had a moment to go screw around with graphic drivers, I set-up rpmfusion's non-free and installed the following:

`dnf -y install xorg-x11-drv-nvidia-390xx xorg-x11-drv-nvidia-390xx-cuda akmod-nvidia-390xx`

It appears there's a good reason as to why the fedora-workstation-repositories does not have this driver set... it just does not work and I don't see any development from rpmfusions side to get it working at the moment.

Thus in my situation, the best option was to install the `NVIDIA-Linux-x86_64-390.87` driver directly from nVidia itself... 3rd party repos such as negativo17 should try to build out support for legacy drivers  until a better solution comes around.

leigh123linux, 2018-11-02:
> It appears there's a good reason as to why the fedora-workstation-repositories does not have this driver set... it just does not work and I don't see any development from rpmfusions side to get it working at the moment.

As far as I know the driver works, there are no open reports so there is nothing to fix.
If it doesn't work you should file a bugreport and provide full info.

scaronni, 2018-11-05:
@I-Iawk sorry but I'm hosting only long lived, short lived and beta branches, depending on the distribution as depicted there: https://negativo17.org/nvidia-driver/

Scroll down to “Distribution and Nvidia driver version support”.

closed 2018-11-05 by scaronni