negativo17/nvidia-driver #61: gnome session doesn't load, tty switching broken on fedora 29

decathorpe, 2018-10-11:
I've noticed several issues when using the updated drivers for fedora 29, and they've been hard to debug because I can't even get a tty up.

Booting f29 works fine, gdm starts correctly, but switching ttys doesn't work (it does when using nouveau or nomodeset). Attempting to log in to a gnome/xorg session results in a black screen, and leaves me completely unable to interact with the system in any way other than a hardware reset.

The only useful messages that I've been able to collect are these messages from Xorg when trying to log into a gnome session. There are no errors preceding this, and nothing other than unrelated messages following it, as far as I can tell.

```
(EE) NVIDIA(GPU-0): Failed to acquire modesetting permission.
(EE) NVIDIA(0): Failing initialization of X screen 0
(II) UnloadModule: "nvidia"
(II) UnloadSubModule: "glxserver_nvidia"
(II) Unloading glxserver_nvidia
(II) UnloadSubModule: "wfb"
(II) UnloadSubModule: "fb"
(EE) Screen(s) found, but none have a usable configuration.
(EE)
Fatal server error:
(EE) no screens found(EE)
```

If it helps, I can try to get full logs off the broken installation.

Side note: I already switched SELinux to permissive mode, it didn't help.

The installed versions of gdm/gnome-shell/mutter/xorg are:
gdm-3.30.1-2.fc29
gnome-shell-3.30.1-1.fc29
mutter-3.30.1-1.fc29
xorg-x11-xserver-Xorg-1.20.1-4.fc29

And the latest version of nvidia-driver from your multimedia repo for f29 is installed. I didn't change anything about the setup, I only upgraded from fedora 28 (where it worked fine).

I'm not sure whether this is an issue with GNOME or the NVIDIA driver.

For the record, I have a GTX 1070 installed in my system, and the display is connected via DisplayPort.

leigh123linux, 2018-10-11:
> but switching ttys doesn't work

The tty switching issue is probably a nvidia issue, it's the only issue I've noticed with the beta driver.

decathorpe, 2018-10-12:
That's a good idea, but adding `nvidia-drm.modeset=1` to the kernel command line makes it slightly worse - now I don't get a black screen when logging in, now it looks like gdm gets stuck.

I'll try disabling wayland explicitly ...

decathorpe, 2018-10-12:
> I'll try disabling wayland explicitly ...

.. which didn't change things either.

leigh123linux, 2018-10-12:
GDM already disables wayland for nvidia in /usr/lib/udev/rules.d/61-gdm.rules , try commenting out the nvidia rule.

decathorpe, 2018-10-12:
I tried something different and recompiled mutter without support for eglstreams, and I am still getting the same error (black screen when logging in) -with and without the modesetting option set.

I don't think the issue is with the udev rule, since gdm is never started in wayland mode anyway AFAICT.

leigh123linux, 2018-10-12:
I use the rpmfusion driver, gdm is working fine with wayland or xorg session

```
$ inxi -G
Graphics:  Device-1: NVIDIA GP107 [GeForce GTX 1050 Ti] driver: nvidia v: 410.57 
           Display: x11 server: Fedora Project X.org 1.20.1 driver: nvidia 
           unloaded: fbdev,modesetting,nouveau,vesa resolution: 3840x2160~60Hz 
           OpenGL: renderer: GeForce GTX 1050 Ti/PCIe/SSE2 v: 4.6.0 NVIDIA 410.57
```

decathorpe, 2018-10-12:
What's the contents of your `/etc/X11/xorg.conf.d/*nvidia*.conf`? Maybe that's the source of the problem.

leigh123linux, 2018-10-12:
```
$ cat /etc/X11/xorg.conf.d/*nvidia*.conf
cat: '/etc/X11/xorg.conf.d/*nvidia*.conf': No such file or directory
```
and
```
$ cat /usr/share/X11/xorg.conf.d/*nvidia*.conf
#This file is provided by xorg-x11-drv-nvidia
#Do not edit

Section "OutputClass"
	Identifier "nvidia"
	MatchDriver "nvidia-drm"
	Driver "nvidia"
	Option "AllowEmptyInitialConfiguration"
	Option "PrimaryGPU" "yes"
	Option "SLI" "Auto"
	Option "BaseMosaic" "on"
EndSection
```

decathorpe, 2018-10-12:
I think I'm going crazy. I've just started having the same issues (no tty switching, broken gdm login) on fedora 28. Undoing the latest updates didn't help either.

decathorpe, 2018-10-13:
Today I gave up trying to solve this problem.

I performed a clean reinstall of fedora 28 and everything works as expected again. So, I'll close this issue for now. If it happens again when I'll try to upgrade to fedora 29 at some point in the future, I'll reopen the issue.

closed 2018-10-13 by decathorpe

decathorpe, 2018-10-25:
For the record, I tried again with the latest version of the nvidia driver (410.66-2), and the issue is still present on fedora 29, but fedora 28 is fine with the same version of the driver. So I guess this is a GNOME issue after all.

RustyRaptor, 2019-05-05:
Has this issue had any follow up from GNOME? What the status? If you continued this elsewhere can you link it please.

decathorpe, 2019-05-05:
If I remember correctly, this turned out to be an issue with the "rhgb" boot parameter / the plymouth boot screen not handing off control of the graphics correctly in this case. It's been fixed for fedora 29.

yuzaipiaofei, 2019-09-03:
rhel8 also can reproduce the issue ,plymouth is 0.9.3 version.