negativo17/nvidia-driver #62: Add Fedora repository for NVidia Vulkan development drivers

jp7677, 2018-10-14:
With all the hype for Steam Play and DXVK it would be very cool if you could add a repository for Fedora 28/29 that includes the NVidia Vulkan development driver from https://developer.nvidia.com/vulkan-driver

I could imagine that this repository would be usable side by side with your multimedia repository whereas the Vulkan development driver supersedes the normal short term driver.

leigh123linux, 2018-10-15:
I don't see any need as I intend to track vulkan to nvidia api from now on.

jp7677, 2018-10-15:
@leigh123linux I don't understand you response, what do you mean with "I intend to track vulkan to nvidia api from now on"?

jp7677, 2018-10-15:
Making the nvidia Vulkan development driver available for F29 only and not for f28 would of course be fine.

leigh123linux, 2018-10-15:
@jp7677  I am the fedora vulkan maintainer and I'm saying f27 and f28 are not getting any more vulkan updates so are stuck on 1.1.77.0 till they go EOL.
F29 will only get vulkan updates equal to the nvidia stable or beta releases.
So this will make your request pointless as it wont have newer vulkan.

jp7677, 2018-10-15:
@leigh123linux Thanks for your answer.
I'm not sure if I get what you are saying, may be I'm missing something. Let me rephrase my question. Nvidia has several driver lines (that I'm aware of): long term (390.87), short term (396.54), short term beta (410.57), vulkan development (396.54.09). Currently the negativo repositories hold version 410.57 for F29. My question is, would it be possible to have another negativo repository for F29 for the 396.54.09 vulkan development driver.

leigh123linux, 2018-10-15:
@jp7677  To get any of the new 396.54.09 (vulkan api 1.1.88) vulkan extensions would require you to compile vulkan git yourself.

f27 and f28 have vulkan api 1.1.77
f29 and rawhide have vulkan api 1.1.82

FTR vulkan latest release is 1.1.82

https://github.com/KhronosGroup/Vulkan-Loader/releases

Does that clarify it?

jp7677, 2018-10-15:
Ok, thanks. Then I guess it is wise to first wait until there is a release of the vulkan loader that contains the 1.1.88 (or higher) specifications (https://github.com/KhronosGroup/Vulkan-Docs/blob/master/ChangeLog.txt) before thinking about adding the Nvidia Vulkan development driver to fedora.

Solitary, 2018-10-19:
@leigh123linux I don't think that is neccessary. 

From the mouth of DXVK creator
> The Vulkan loader also doesn't have to be updated for newer extensions because DXVK and winevulkan both dynamically query the function pointers they need, rather than linking to the loader library.

source: https://www.reddit.com/r/wine_gaming/comments/9o6jpp/wine_318_instability/e7rus88/?context=3

jp7677's comment about 396.54 driver is valid, it really should be updated and maintained besides the new 410 driver, because it contains all the Vulkan specific features and fixes and currently is THE driver to use for gaming. Also upgrading is not just about the new features, supposedly the 396.54.05 version saw some noticeable performance improvements, which would be also nice to have.

scaronni, 2018-11-05:
Sorry, I'm hosting only long lived, short lived and beta branches, depending on the distribution as depicted there: https://negativo17.org/nvidia-driver/

Scroll down to “Distribution and Nvidia driver version support”.

closed 2018-11-05 by scaronni

Solitary, 2018-11-05:
I don't understand your differentiation. Long lived, short lived and beta branches... yet you only mention one driver version supported. 
You "upgraded" to 410, but essentially it's feature downgrade for many people. 410 is interesting for Turing owners, which I don't think there is many at the moment.

Solitary, 2018-11-05:
Sorry, but then you are wrong, 410 is the new long lived, but only newest by date... actually just by 3 days. 396.54.09 beta has Vulkan specific features that did not hit 410 yet. 396.54 is still the beta branch that culminates Vulkan development which is IMHO current focus for many people (for gaming reasons).

scaronni, 2018-11-05:
You're wrong. The 29 branch had 410.57 as a beta, it came out middle of September:

https://www.nvidia.com/drivers/results/138279

Which contained a few specific changes which in case of a previous release I would have to revert.

396.54 was the previous short lived branch, and 396.54.09 is a specific Vulkan release that happens every once in a while. As far as I know it happened a couple of times only. It's not even listed in the beta branch:

https://www.nvidia.com/drivers/beta

dreamer, 2018-11-08:
@leigh123linux do I understand correctly? Fedora 29 will not get newer patch releases of Vulkan, it is stuck on 1.1.82 until EOL (while Khronos just released 1.1.91) and for this reason there will be no further updates of nvidia driver from short-lived branch (so no 396.54.09 drivers for Fedora 29 ever).

So if I want to use extensions introduced after 1.1.82, I need to wait until Fedora 30 or maybe even longer? Is this correct?

leigh123linux, 2018-11-08:
@dreamer Vulkan ownership has changed.

https://lists.fedoraproject.org/archives/list/devel@lists.fedoraproject.org/thread/Z5HMD5UINAPRX7JQ5HX4O7FA7KMP4ONX/

396.54.09 is older than 410.73 and has limited hardware support.

https://devtalk.nvidia.com/default/topic/533434/linux/current-graphics-driver-releases/

@scaronni  I have no intention to package additional nvidia vulkan branch for rpmfusion as it lags too far behind main and doesn't get updated enough for fedora kernel changes.

https://developer.nvidia.com/vulkan-driver

I'm currently looking at 415.15 beta for f30  :-)
https://devtalk.nvidia.com/default/topic/1043869

jp7677, 2018-12-07:
Short lived release 415.22 is out (https://www.nvidia.com/Download/driverResults.aspx/141072/en-us) which obsoletes my initial request.