negativo17/nvidia-driver #64: Latest driver causes fallback to Nouveau? (Fedora 28)

7oxicshadow, 2018-10-27:
I have been using the negativo17 repo for a long time and have never had any issues but I recently did "sudo dnf update" and noticed that there were some updated NVIDIA drivers available.

I allowed it to complete and restarted my machine. Now every time I login it starts a nouveau Wayland session.

I have the following in my logs:
```
"started fallback to nouveau if nvidia did not load"
"File: /var/lib/dkms/nvidia/396.54/source/dkms.conf does not exist."
```

396.54 is old now isn't it?

If I login and try to run nvidia-settings i get the following:

`"Error: Unable to find display on any available system"`

I have tried removing all nvidia drivers and then re-installing them but seems to have made things worse. After re-installing the driver the machine slows to a crawl with the mouse cursor moving by a small amount once a second and if I look at my processor stats all cores are at 100% constantly.

If I uninstall the drivers and use nouveau the processor loads return to normal.

Any ideas what might have caused this. Is there an easy way of going back a version on the nvidia drivers via dnf?

jp7677, 2018-10-27:
May be this?
https://github.com/negativo17/nvidia-driver/issues/56#issuecomment-414080093

7oxicshadow, 2018-10-27:
I have fixed the DKMS issue but unfortunately the problem still exists.

If no other people are reporting any problems then it must be something unique with my machine. Sounds like F29 will be released next week anyway so I guess its time for a fresh install....

Out of curiosity, are there any NVIDIA specific logs anywhere that would give a more descriptive error?

pjperry, 2018-10-27:
What graphics card do you have? See output from lspci

NVIDIA dropped support for older Fermi-based cards after the 390.xx driver release, so if you have an older card and you updated drivers to 396.xx or newer, they will not support your hardware and you would need to revert to the legacy 390.xx driver branch.

7oxicshadow, 2018-10-27:
@pjperry I have a GTX 1070 so it should be fine.

I have the rpmfusion repo'e enabled. Maybe something was updated from them that conflicts with the latest driver. I will keep experimenting to see if I can find anything.

pawelcomp, 2018-10-27:
I also have a GTX 1070 and the same thing happened after a fresh install.

7oxicshadow, 2018-10-27:
@pawelcomp Did this happen to you within the last couple of days or is it more of a general comment that it is something you have had happen in the past

pawelcomp, 2018-10-27:
I just reinstalled a few hours ago and this happened. Was working without problems earlier.

7oxicshadow, 2018-10-27:
@pawelcomp Thanks for confirming.

I guess we will have to wait if others start reporting problems. I wonder if there has been an ABI change or something in the latest kernel.

chrishiner, 2018-10-27:
Earlier today I updated to nvidia-driver-3:410.73-2.fc28.x86_64, and when I next rebooted, the mouse was really slow and it was performing really badly after I logged in and wasn't seeing the nvidia card. I went back to the login screen and reselected Gnome on X.org and it went back to normal. I see nvidia-driver-410.73-3.fc28.x86_64 just showed up in my update list, let's see what happens with the next reboot. This is a GTX 1070.

7oxicshadow, 2018-10-27:
@chrishiner When mine was running painfully slow, gnome-shell was at 100% on all 8 threads on an i7 6700k @ 4.5Ghz
Looks like GTX10XX series might be the ones having a problem.

leigh123linux, 2018-10-27:
I'm muting this thread as the crap written here is total drivel, not one of you has enough brain cells to include logging or the output for nvidia-bug-report.sh.

pkovacs, 2018-10-28:
I was able to roll back to 410.66 and this clears the problem for me on all machines.   I don't know what the issue is but I can report that the nvidia kernel modules were loaded (lsmod) and nouveau was still blacklisted.   The odd thing I saw with 410.73 was that `glxinfo | grep -i render`  reported a mix of NV and MESA items whereas normally I would only see NV.    To go back to 410.66, I cloned the nvidia-driver and nvidia-kmod repos and manually downloaded the 410.66 .run file from nvdia, then I checked out branches corresponding to the SHA entries for "Update to 410.66," e.g. 615c71b, then I ran the tarball generator, created the local rpm's with `fedpkg`, disabled the negative17 repo, added the rpm's to a local repo, and finally reinstalled nvidia-driver with `dnf` against my local repo and it works.   I don't know if the nvidia driver or the most recent changes to the .spec caused the problems, but a rollback to 410.66 does work.   I'm going to keep a few working versions always available for situations like this.

jp7677, 2018-10-28:
I think I have the same issue now since updating from 410.73-1 to version 410.73-3 this morning.
This commit https://github.com/negativo17/nvidia-driver/commit/4aeaa1c119524dc6043cc41b486f0d0b6523983b enables modesetting. Once I edit my boot parameters on startup (using `e` in grub) and use `nvidia-drm.modeset=0` instead of `nvidia-drm.modeset=1` (thus disabling modesetting) X starts fine again with the correct NVidia driver. Does disabling modesetting works as a temporary workaround for you too?

PS: I'm using a  GTX 1080.

jp7677, 2018-10-28:
It seems much simpler, well in my case. With modeset enabled I'm ending up in wayland session now, even with the NVidia drivers installed. The gdm session runs on wayland, but when I select "Gnome on Xorg" I'm ending up again with a fully working desktop again. Using "Gnome" wants to start a Wayland session. This seems to work now and doesn't fall back to X anymore like it did when modesetting was still disabled.

7oxicshadow, 2018-10-28:
@jp7677 Excellent work. I can confirm that setting nvidia-drm.modeset=0 allows everything to function normally again on the latest driver.

Many thanks for suggesting it!

jp7677, 2018-10-28:
@7oxicshadow Thanks, though as stated later, just making sure that you use X instead of Wayland is the better solution.

7oxicshadow, 2018-10-28:
Is there a way to set "Gnome on Xorg" by default? When I get to the log in manager my mouse is very unresponsive so selecting "Gnome on Xorg" is a slow painful process.

If anyone wants to modify grub permanently you can do the following on Fedora. **(Note: If this is done incorrectly it could stop your computer booting so only do this if you are confident in making system changes)**

`sudo -H gedit /etc/default/grub`

The update the GRUB_CMDLINE_LINUX to set **nvidia-drm.modeset**  to 0. Save your changes.
Then run:

`sudo grub2-mkconfig --output=/boot/grub2/grub.cfg`

To make it permanent. From now on it will be defaulted to 0. (This may be overridden by future driver updates)

jp7677, 2018-10-28:
> Is there a way to set "Gnome on Xorg" by default? When I get to the log in manager my mouse is very unresponsive so selecting "Gnome on Xorg" is a slow painful process.

You could convince GDM to start in Xorg, see https://wiki.archlinux.org/index.php/GDM#Use_Xorg_backend

decathorpe, 2018-10-28:
Yep, this just bit me too - please don't enable `nvidia-drm.modeset` on fedora 28. It renders the system barely usable.

dbeurle, 2018-10-28:
I can +1 this issue on Fedora 29.  Setting `nvidia-drm.modeset=0` allowed me to get back into gnome.

mati865, 2018-10-28:
@dbeurle either you meant Fedora 28 or something is wrong with your setup.

dbeurle, 2018-10-28:
I'm from a parallel universe where Fedora ships on-time ;-)  I meant Fedora 29 beta...
```
        :-------------------::        OS: Fedora 29 TwentyNine
      :-----------/shhOHbmp---:\      Kernel: x86_64 Linux 4.18.16-300.fc29.x86_64
    /-----------omMMMNNNMMD  ---:     Uptime: 1h 56m
   :-----------sMMMMNMNMP.    ---:    Packages: 3754
  :-----------:MMMdP-------    ---\   Shell: bash 4.4.23
 ,------------:MMMd--------    ---:   Resolution: 3840x2160
 :------------:MMMd-------    .---:   DE: GNOME 
 :----    oNMMMMMMMMMNho     .----:   WM: GNOME Shell
 :--     .+shhhMMMmhhy++   .------/   WM Theme: 
 :-    -------:MMMd--------------:    GTK Theme: Adwaita [GTK2/3]
 :-   --------/MMMd-------------;     Icon Theme: Adwaita
 :-    ------/hMMMy------------:      Font: Cantarell 11
 :-- :dMNdhhdNMMNo------------;       CPU: AMD Ryzen 7 2700 Eight-Core @ 16x 3.2GHz [50.6°C]
 :---:sdNMMMMNds:------------:        GPU: GeForce GTX 970/PCIe/SSE2
 :------:://:-------------::          RAM: 3457MiB / 16052MiB

```

scaronni, 2018-10-28:
> I have the following in my logs:
> ```
> "started fallback to nouveau if nvidia did not load"
> "File: /var/lib/dkms/nvidia/396.54/source/dkms.conf does not exist."
> ```
There is a known issue in DKMS, when updating the kernel-devel package and the drivers at the same time, there might be screw ups.
```
dkms remove nvidia/<version> --all
dkms install nvidia/<version>
reboot
```
Should fix it. Otherwise use `akmod-nvidia`.

scaronni, 2018-10-28:
> Fedora 27/28 `mutter` isn't built with EGLDevice option and as you can see it tries to use Wayland but fails miserably. Either editing `/etc/gdm/custom.conf` or setting `nvidia-drm.modeset=0` are fine but this package should only enable KMS for Fedora >= 29.

Funny because I enabled modesetting on all of my systems regardless of EGLDevice support and they work fine. I use the X.org session mostly. I guess that is fine for all Fedora releases, regardless of EGLDevice support.

> Fedora 29 doesn't suffer from it because this time `mutter` got EGLStream enabled but it [has another bug](https://bugzilla.redhat.com/show_bug.cgi?id=1643302). If you upgrade/install it and get black screen on log in you will have to remove `rghb` from the command line until it's fixed.

Wayland is disabled on Nvidia through some udev rules and a binary from GDM. You can currently use it only in the Gnome session itself.

scaronni, 2018-10-28:
Can someone with the problem on Fedora 28 just don't do anything (just leave whatever the package sets up for you in Grub, configs, etc.) and just select "Gnome on Xorg" from the login screen?

mati865, 2018-10-28:
@scaronni could you make sure you don't have https://copr.fedorainfracloud.org/coprs/mati865/mutter-eglstream enabled?
I'm almost sure I had poor GDM performance with Nvidia KMS enabled without patched mutter before my Fedora 28 died.

chrishiner, 2018-10-28:
@scaronni Selecting Gnome on Xorg fixes the issue after login, but the GDM screen was still really slow.
Setting WaylandEnable=false in the /etc/gdm/custom.conf fixes the GDM screen as well.

scaronni, 2018-10-29:
Funny, not slow for me on any system. Will look into it.

khanh1412, 2018-10-30:
I got the same problem on F28, nvidia driver versiom 410.73

scaronni, 2018-10-30:
Pushed an update where modesetting is disabled in Fedora 28.

closed 2018-10-30 by scaronni