negativo17/nvidia-driver #65: Fresh install F29 no display with nividia drivers

jeffclay, 2018-10-31:
As the title says, I have a fresh install of F29 (KDE spin). I installed the OS, ran dnf updates, rebooted, installed the rpmfusion and negativo-multimedia repo ran updates again (note that nvidia-driver-cuda-libs installed itself at this point even though 'dnf remove *\nvidia\*' did not return any installed nvidia packages), rebooted then ran 'dnf install nvidia-driver nvidia-settings kernel-devel akmod-nvidia'. I ran top and waited until the spawned processes (akmods) settled down and then rebooted again. I now don't get display on any monitor. Upon boot, I get the normal kernel selection screen then it flashes the fedora logo and then monitors go to sleep. trying ctrl+alt+f2 or f3, etc. do not seem to work now either.

OS: Fedora 29
Kernel: 4.18.16-300
Nvidia Version: 410.73-4
Card Model: 2x Quadro K620

jeffclay, 2018-10-31:
I noticed that 2 of my f28 installs started having this same no display issue a few days ago and then it was fixed yesterday after updating to the -4 version of the nvidia driver. Unfortunately, I had already formatted on those 2 systems and reinstalled in an attempt to diagnose the issue. Looking at the recent commits, I see 0c63203 which removed "nvidia-drm.modeset=1" from the boot string. I tried removing that myself from grub and was able to boot and login fine. I was also able to run nvidia-settings without any issue.

mati865, 2018-11-05:
Try removing `rhgb` from the kernel command line: https://bugzilla.redhat.com/show_bug.cgi?id=1643302

jeffclay, 2018-11-05:
Removing rhgb results in the same behavior of no display and my monitors going to sleep. I can only get to the GUI login by setting nvidia-drm.modeset=0 or by removing that option entirely. Reading the link provided by mati865, they are referencing an issue after upgrading from f28 to f29 (I have a fresh install of f29, not an upgrade) and they also show that they have [modeset set to 0](https://bugzilla.redhat.com/show_bug.cgi?id=1643302#c5)

mati865, 2018-11-05:
`rhgb` breaks both upgraded and fresh installs of Fedora 29 with GDM, apparently it's not the case with SDDM.
Have you checked if dmesg, journal and Xorg logs contain useful data about this issue?

jeffclay, 2018-11-05:
No, I haven't; I'm not sure what to look for.

jeffclay, 2018-11-28:
I ran system updates today on my F29 machine. Among the updates were kernel version 4.18.17-300 and akmod-nvidia 3:415.18-1. After reboot, my system now starts up fine without having to edit the boot flags in grub.

closed 2018-11-28 by jeffclay