negativo17/nvidia-driver #68: virtual terminal show white blocks

claudio4j, 2018-12-28:
X is working fine with SDDM. The problem is when I switch to a virtual terminal (ctrl alt f1 or f2) and the screen is full of white blocks, [see image](https://forums.fedoraforum.org/attachment.php?attachmentid=29963&d=1545916637)

Any hint on how to fix that ?

$ rpm -qa|grep nvidia|sort
akmod-nvidia-415.25-1.fc29.x86_64
kmod-nvidia-4.19.10-300.fc29.x86_64-415.25-1.fc29.x86_64
nvidia-driver-415.25-1.fc29.x86_64
nvidia-driver-cuda-libs-415.25-1.fc29.x86_64
nvidia-driver-libs-415.25-1.fc29.x86_64
nvidia-libXNVCtrl-415.25-1.fc29.x86_64
nvidia-settings-415.25-1.fc29.x86_64

OS: Fedora 29
Kernel: 4.19.10-300.fc29.x86_64
Nvidia Version: 415.25-1
Card Model: GM107GLM [Quadro M2000M]

claudio4j, 2018-12-28:
Dropping the rhgb from boot options fixes the problem, but is this a bug in fedora or nvidia driver ?

scaronni, 2019-02-06:
This was a known bug in Fedora's plymouth, it is fixed by now:

https://bugzilla.redhat.com/show_bug.cgi?id=1643302

closed 2019-02-06 by scaronni