negativo17/nvidia-driver #70: kmod-nvidia-410.93-3.el7.x86_64.rpm missing in https://negativo17.org/repos/nvidia/epel-7/x86_64/

brunckodusan, 2019-02-03:


brunckodusan, 2019-02-03:
Hello,
Yesterday were upgrade nvidia drivers in https://negativo17.org/repos/nvidia/epel-7/x86_64/. kmod-nvidia-410.93-3.el7.x86_64.rpm was today morning missing, now is o.k. But unfortunately  not only this package
was morning missing, for upgrade from 410.78 to 410.93 are need update of nvidia-settings-410.78-1.el7.x86_64, nvidia-config-xxx and nvidia-modprobe-xxx.
My large plea to you:
can you put these new updates also to the https://negativo17.org/repos/nvidia/epel-7/x86_64/?

Thank you very much in advance.

Best regards,
                        Dusan

scaronni, 2019-02-03:
Ouch. I had an issue with the system while regenerating the repository, it seems some files that were there before disappeared. I'm rebuilding the repository now and will re-upload soon.

brunckodusan, 2019-02-04:
Hello,
Thank you very much!
Now all is O.K.
Regards,
                    Dusan

closed 2019-02-04 by scaronni