negativo17/nvidia-driver #72: [Fedora 29] Latest Kernel 4.20.6-200.fc29.x86_64 break Boot display

AdsonCicilioti, 2019-02-06:
I was updating the system normally until after this version of the Kernel (4.20.6-200.fc29.x86_64) the video did not work anymore. Giving some green winks and not starting the GDM.
I do not know how to investigate what might be happening. If you could show me the cumin, I'd be grateful.

I enabled the Grub menu and chose version 4.20.5-200.fc29.x86_64, which is the last one, and that was working well, but for some reason also did not work.

I had to choose the version 4.20.3-200.fc29.x86_64 that worked again.

hevanaa, 2019-02-07:
I've had similar issues in the past and usually it is dkms that has failed to compile the nvidia module, either for a new kernel or a new nvidia driver version or both. Check with ls -l /var/lib/dkms/nvidia/ for missing or dead symlinks. Also check inside the driver version directory, e.g. 415.27. I use to remove any dead symlinks and then recompile dkms for current kernel with 'sudo dkms install nvidia/415.27' (set correct driver version here; from the directory name). To rebuild for all kernels add '--all'.

AdsonCicilioti, 2019-02-07:
But negativo17/nvidia-driver uses akmod for module kernel.
$ akmods (returnng:)
> /etc/rc.d/init.d/functions not found

AdsonCicilioti, 2019-02-07:
**sudo systemctl status akmods**
● akmods.service - Builds and install new kmods from akmod packages
   Loaded: loaded (/usr/lib/systemd/system/akmods.service; enabled; vendor preset: enabled)
   Active: **failed** (Result: exit-code) since Thu 2019-02-07 13:12:49 -02; 37min ago
  Process: 789 ExecStart=/usr/sbin/akmods --from-init (code=exited, status=1/FAILURE)
 Main PID: 789 (code=exited, status=1/FAILURE)

fev 07 13:12:49 fedora systemd[1]: Starting Builds and install new kmods from akmod packages...
fev 07 13:12:49 fedora akmods[789]: **/etc/rc.d/init.d/functions not found**
fev 07 13:12:49 fedora systemd[1]: akmods.service: Main process exited, code=exited, status=1/FAILURE
fev 07 13:12:49 fedora systemd[1]: akmods.service: Failed with result 'exit-code'.
fev 07 13:12:49 fedora systemd[1]: **Failed to start Builds and install new kmods from akmod packages.**

hevanaa, 2019-02-07:
Dkms is also supported by negativo17. See https://negativo17.org/nvidia-driver/.
It is a long time since I used akmods. That error message about init.d/functions looks quite old fashioned (now with systemd and all).  Do you have the package initscripts installed (that contains that file)?

AdsonCicilioti, 2019-02-08:
initscripts was not installed ..
I installed initscripts, restarted the akamods service and ran the command # akmods --force.
Everything working now.
Thank you.

But the question remains: how were you generating the kernel modules before? Because I upgraded the 2x kernel before this failure.
Weird. I can not say what happened.

hevanaa, 2019-02-08:
I have no idea how it worked before. But it looks like a bug in akmods. The service depends on initscripts and it didn't require the package. Are you sure you have the akmods package from negativo installed?

AdsonCicilioti, 2019-02-08:
dnf list | grep akmods
akmods.noarch        0.5.6-17.fc29         @updates

scaronni, 2019-02-08:
`akmods` is in main Fedora. It should require `initscripts` or better we should check if `initscripts` is still required on recent Fedora and if not drop the requirement from the code.

closed 2019-02-09 by AdsonCicilioti