negativo17/nvidia-driver #73: Nouveau appears not to be disabled

dbeurle, 2019-02-10:
I have a slightly strange configuration with Fedora 29 server.  It's basically a compute / development box with a monitor connected to the integrated Intel graphics and the nvidia gpu is left unconnected.  I installed the `nvidia-driver-cuda` and `cuda-devel` packages and rebooted.  The logs indicate that Nouveau wasn't disabled and the nvidia module couldn't find an available nvidia adapter.  Manually unloading the nouveau driver and loading the nvidia driver seems to work.

Trying `nvidia-modprobe` didn't seem to help.

Are there any suggestions to determine if this is a configuration error or a packaging problem?

Thanks again.

scaronni, 2019-02-10:
This is what manages the Nouveau/Nvidia modules:
```
$ rpm -ql nvidia-kmod-common 
/usr/lib/dracut/dracut.conf.d/99-nvidia.conf
/usr/lib/modprobe.d/nvidia.conf
/usr/lib/systemd/system-preset/95-nvidia-fallback.preset
/usr/lib/systemd/system/nvidia-fallback.service
/usr/lib/udev/rules.d/60-nvidia.rules
```
In particular:
```
$ cat /usr/lib/modprobe.d/nvidia.conf | grep black
blacklist nouveau
```
In my case that's enough to disable the module. The package should also have added the following:
```
$ cat /etc/default/grub | grep CMDLINE
GRUB_CMDLINE_LINUX="rd.driver.blacklist=nouveau rd.lvm.lv=vg0/root rd.lvm.lv=vg0/swap rhgb quiet "
```
To disable `nouveau` loading ALSO at `dracut` time. Can you confirm that you have that line and that the system has been booted as such?
```
$ cat /proc/cmdline 
BOOT_IMAGE=/boot/vmlinuz-4.20.6-200.fc29.x86_64 root=/dev/mapper/vg0-root ro rd.driver.blacklist=nouveau rd.lvm.lv=vg0/root rd.lvm.lv=vg0/swap rhgb quiet LANG=en_US.UTF-8
```

scaronni, 2019-02-10:
The entry in the Grub configuration file is added by the package script section, you can check with:
```
$ rpm -q --scripts nvidia-kmod-common
```

dbeurle, 2019-02-10:
Thanks for the reply.  I have that line in the grub configuration and it has been booted:

```
$ cat /proc/cmdline
BOOT_IMAGE=/vmlinuz-4.20.6-200.fc29.x86_64 root=UUID=8c7955f0-077d-403e-9ea7-c14b48f660d7 ro rootflags=subvol=root rd.driver.blacklist=nouveau
```
and all those files are there and appear to be working.  The systemctl log for that fall-back service:
```
sudo systemctl status nvidia-fallback.service 
● nvidia-fallback.service - Fallback to nouveau if nvidia did not load
   Loaded: loaded (/usr/lib/systemd/system/nvidia-fallback.service; enabled; vendor preset: enabled)
   Active: active (exited) since Sun 2019-02-10 10:57:02 CET; 8h ago
  Process: 829 ExecStart=/sbin/modprobe nouveau (code=exited, status=0/SUCCESS)
 Main PID: 829 (code=exited, status=0/SUCCESS)

Feb 10 10:57:01 maxwell systemd[1]: Starting Fallback to nouveau if nvidia did not load...
Feb 10 10:57:02 maxwell systemd[1]: Started Fallback to nouveau if nvidia did not load.
```
What's strange is that it was working a few days ago and the only thing that really changed was a custom kernel module for a wake on lan fix for the alx module.  Could that interfere?
```
$ cat /etc/systemd/system/fix-alx-wol.service 
[Unit]
Description=Reloads alx module to enable wake-on-LAN.
After=multi-user.target

[Service]
ExecStart=/etc/fix_alx_wol.sh

[Install]
WantedBy=default.target
```
```
$ cat /etc/fix_alx_wol.sh
#!/bin/sh

modprobe -r alx
modprobe alx
```

scaronni, 2019-02-11:
> The systemctl log for that fall-back service:

That is not ok, the `nvidia-fallback` service should be as follows:
```
● nvidia-fallback.service - Fallback to nouveau if nvidia did not load
   Loaded: loaded (/usr/lib/systemd/system/nvidia-fallback.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
Condition: start condition failed at Fri 2019-02-08 06:59:05 CET; 3 days ago
           └─ ConditionPathExists=!/sys/module/nvidia was not met
```
> Could that interfere?

No, don't think so.

mati865, 2019-02-11:
What is your `dkms status`?

dbeurle, 2019-02-11:
```
$ dkms status
alx, 4.20.6, 4.20.6-200.fc29.x86_64, x86_64: installed (original_module exists)
```

scaronni, 2019-02-11:
Correct. Also, considering you have alx in `dkms` format, you should switch from `akmod-nvidia` to `dkms-nvidia`.

dbeurle, 2019-02-12:
I've removed the akmod packages and installed nvidia-dkms:
```
$ dkms status
alx, 4.20.6, 4.20.6-200.fc29.x86_64, x86_64: installed (original_module exists)
nvidia, 415.27: added
```

It still doesn't load on boot.

leigh123linux, 2019-02-12:
@dbeurle Can you provide some basic info, what card?

Can you attach the file that this command generates

```
nvidia-bug-report.sh
```

dbeurle, 2019-02-12:
As booted configuration:
```
$ lspci -vv
01:00.0 VGA compatible controller: NVIDIA Corporation GM204 [GeForce GTX 970] (rev a1) (prog-if 00 [VGA controller])
	Subsystem: eVga.com. Corp. Device 1972
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 29
	Region 0: Memory at f2000000 (32-bit, non-prefetchable) [size=16M]
	Region 1: Memory at e0000000 (64-bit, prefetchable) [size=256M]
	Region 3: Memory at f0000000 (64-bit, prefetchable) [size=32M]
	Region 5: I/O ports at e000 [size=128]
	Expansion ROM at f3000000 [disabled] [size=512K]
	Capabilities: <access denied>
	Kernel driver in use: nouveau
	Kernel modules: nouveau, nvidia_drm, nvidia
```
```
$ sudo nvidia-bug-report.sh 

nvidia-bug-report.sh will now collect information about your
system and create the file 'nvidia-bug-report.log.gz' in the current
directory.  It may take several seconds to run.  In some
cases, it may hang trying to capture data generated dynamically
by the Linux kernel and/or the NVIDIA kernel module.  While
the bug report log file will be incomplete if this happens, it
may still contain enough data to diagnose your problem.

Please include the 'nvidia-bug-report.log.gz' log file when reporting
your bug via the NVIDIA Linux forum (see devtalk.nvidia.com)
or by sending email to 'linux-bugs@nvidia.com'.

Running nvidia-bug-report.sh...ls: cannot access '/proc/driver/nvidia/./gpus/': No such file or directory

If the bug report script hangs after this point consider running with
--safe-mode and --extra-system-data command line arguments.

 complete.
```

Manually unloading nouveau and loading the nvidia module
```
01:00.0 VGA compatible controller: NVIDIA Corporation GM204 [GeForce GTX 970] (rev a1) (prog-if 00 [VGA controller])
	Subsystem: eVga.com. Corp. Device 1972
	Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0, Cache Line Size: 64 bytes
	Interrupt: pin A routed to IRQ 16
	Region 0: Memory at f2000000 (32-bit, non-prefetchable) [size=16M]
	Region 1: Memory at e0000000 (64-bit, prefetchable) [size=256M]
	Region 3: Memory at f0000000 (64-bit, prefetchable) [size=32M]
	Region 5: I/O ports at e000 [size=128]
	Expansion ROM at f3000000 [disabled] [size=512K]
	Capabilities: <access denied>
	Kernel driver in use: nvidia
	Kernel modules: nouveau, nvidia_drm, nvidia
```
[nvidia-bug-report.log.failed.gz](https://github.com/negativo17/nvidia-driver/files/2854723/nvidia-bug-report.log.failed.gz)
[nvidia-bug-report.log.gz](https://github.com/negativo17/nvidia-driver/files/2854724/nvidia-bug-report.log.gz)

scaronni, 2019-02-12:
Ok, I think I got it, the load does not actually happen because when having `nvidia-kmod-common `as part of `nvidia-driver` the actual loading was triggered by the DDX driver and the stub configuration.

In the case here of no DDX, unless you load a CUDA program with `nvidia-modprobe` installed or explicitly load the module nothing happens, so when reaching the fallback service the module gets loaded. Not happening on one system I have because of an additional module configuration.

I will make some changes, probably to the `nvidia-driver-cuda` package to load it but making sure the configuration snippet is not included by `dracut` in the `initrd`.

leigh123linux, 2019-02-12:
Try adding **modprobe.blacklist=nouveau** to the cmdline as nouveau is loaded and prevent the nvidia module from loading.

```
[    0.000000] Command line: BOOT_IMAGE=/vmlinuz-4.20.6-200.fc29.x86_64 root=UUID=8c7955f0-077d-403e-9ea7-c14b48f660d7 ro rootflags=subvol=root rd.driver.blacklist=nouveau
```
```
[    5.435274] nouveau 0000:01:00.0: enabling device (0000 -> 0003)
[    5.436312] nouveau 0000:01:00.0: NVIDIA GM204 (124020a1)
[    5.603461] nouveau 0000:01:00.0: bios: version 84.04.1f.00.6f
[    5.688078] nouveau 0000:01:00.0: fb: 4096 MiB GDDR5
[    5.690892] nouveau 0000:01:00.0: bus: MMIO write of 8000012c FAULT at 10eb14 [ IBUS ]
[    5.761785] IPv6: ADDRCONF(NETDEV_UP): enp3s0: link is not ready
[    5.767225] IPv6: ADDRCONF(NETDEV_UP): enp3s0: link is not ready
[    5.870066] [TTM] Zone  kernel: Available graphics memory: 1852426 kiB
[    5.870530] [TTM] Initializing pool allocator
[    5.870979] [TTM] Initializing DMA pool allocator
[    5.871440] nouveau 0000:01:00.0: DRM: VRAM: 4096 MiB
[    5.871868] nouveau 0000:01:00.0: DRM: GART: 1048576 MiB
[    5.872289] nouveau 0000:01:00.0: DRM: TMDS table version 2.0
[    5.872702] nouveau 0000:01:00.0: DRM: DCB version 4.1
[    5.873109] nouveau 0000:01:00.0: DRM: DCB outp 00: 01000f02 00020030
[    5.873532] nouveau 0000:01:00.0: DRM: DCB outp 01: 02000f00 00000000
[    5.873947] nouveau 0000:01:00.0: DRM: DCB outp 02: 04011f82 00020030
[    5.874346] nouveau 0000:01:00.0: DRM: DCB outp 03: 02022f62 00020010
[    5.874745] nouveau 0000:01:00.0: DRM: DCB outp 05: 02833f76 04420020
[    5.875134] nouveau 0000:01:00.0: DRM: DCB outp 06: 02033f72 00020020
[    5.875526] nouveau 0000:01:00.0: DRM: DCB outp 15: 01df5ff8 00000000
[    5.875910] nouveau 0000:01:00.0: DRM: DCB conn 00: 00001030
[    5.876294] nouveau 0000:01:00.0: DRM: DCB conn 01: 01000131
[    5.876688] nouveau 0000:01:00.0: DRM: DCB conn 02: 00010261
[    5.877074] nouveau 0000:01:00.0: DRM: DCB conn 03: 00020346
[    5.877458] nouveau 0000:01:00.0: DRM: DCB conn 05: 00000570
[    5.930922] nouveau 0000:01:00.0: DRM: failed to create encoder 1/8/0: -19
[    5.931311] nouveau 0000:01:00.0: DRM: Virtual-1 has no encoders, removing
[    5.931712] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
[    5.932093] [drm] Driver supports precise vblank timestamp query.
[    6.001022] nouveau 0000:01:00.0: DRM: MM: using COPY for buffer copies
[    6.090415] [drm] Cannot find any crtc or sizes
[    6.105742] [drm] Initialized nouveau 1.3.1 20120801 for 0000:01:00.0 on minor 1
[    6.180492] [drm] Cannot find any crtc or sizes
[    6.229169] [drm] Cannot find any crtc or sizes
[    6.263528] alx 0000:03:00.0 eth0: Qualcomm Atheros AR816x/AR817x Ethernet [d0:50:99:51:02:3e]
[    6.276071] alx 0000:03:00.0 enp3s0: renamed from eth0
[    6.299496] IPv6: ADDRCONF(NETDEV_UP): enp3s0: link is not ready
[    6.300031] IPv6: ADDRCONF(NETDEV_UP): enp3s0: link is not ready
[    6.330763] nvidia: module license 'NVIDIA' taints kernel.
[    6.330790] Disabling lock debugging due to kernel taint
[    6.345101] nvidia-nvlink: Nvlink Core is being initialized, major device number 236
[    6.345646] NVRM: The NVIDIA probe routine was not called for 1 device(s).
[    6.345668] NVRM: This can occur when a driver such as: 
               NVRM: nouveau, rivafb, nvidiafb or rivatv 
               NVRM: was loaded and obtained ownership of the NVIDIA device(s).
[    6.345710] NVRM: Try unloading the conflicting kernel module (and/or
               NVRM: reconfigure your kernel without the conflicting
               NVRM: driver(s)), then try loading the NVIDIA kernel module
               NVRM: again.
[    6.345759] NVRM: No NVIDIA graphics adapter probed!
```

scaronni, 2019-02-12:
Thanks @leigh123linux. @dbeurle I will make some adjustment this evening (at work now).

remyabel, 2021-11-02:
Ran into this issue upon upgrading from fedora 34 to fedora 35 with the same issue mentioned in @leigh123linux's comment. Their workaround also worked. For right now, I'm going to be doing `grubby --update-kernel=ALL --args` until this gets fixed. Note that reinstalling the driver packages and switching from akmod to dkms didn't work and I'm not even sure what the issue is since I haven't run into this problem before.

scaronni, 2021-11-04:
> Ran into this issue upon upgrading from fedora 34 to fedora 35 with the same issue mentioned in @leigh123linux's comment. Their workaround also worked. For right now, I'm going to be doing `grubby [...]

Same happening here, but only on Fedora 35 systems, it seems the `modprobe.blacklist=nouveau` is now required along `rd.driver.blacklist=nouveau`. I've added it to the latest `nvidia-kmod-common` package.

jp7677, 2021-11-04:
I had/have the same issue . My work around for now was to add `blacklist nouveau` in `/etc/modprobe.d/nvidia.conf`. I do wonder though why this worked, because that blacklist statement already exists in `/lib/modeprobe.d/nvidia.conf`. It seems like the latter is ignored.

jp7677, 2021-11-04:
PS: I haven’t tested what happens if I rename one of the files. A wild theory could be that just one file  is picked up because both configuration files have the same name.

mwawra, 2021-11-04:
Same here on a Thinkpad P52 with a Quadro P3200 MaxQ and Fedora 34->35. Used the workaround with `/etc/modprobe.d/nvidia.conf` which indeed worked fine.

scaronni, 2021-11-04:
`blacklist nouveau` should always be present, regardless of what you're doing with the nvidia drivers, so i might as well leave it only as a boot parameter in `nvidia-kmod-common`.

@jp7677 and @mwawra can you confirm that nouveau is still disabled if you remove the entry in `/etc/modprobe.d/nvidia.conf` and you install the latest 495.44 packages?

mwawra, 2021-11-04:
I can confirm that!
Just for clarity, I removed `blacklist nouveau` from `/etc/modprobe.d/nvidia.conf` not from `/usr/lib/modprobe.d/nvidia.conf.

jp7677, 2021-11-04:
Hi @scaronni I haven't yet updated to 495.44, but I'm sure that your solution for blacklisting works.

That said, I deliberately played a bit to get to the root cause of this issue, since the `blacklist` statement is there. I figured that the issue is indeed that there are two files with the same name `nvidia.conf`. So I just renames `/etc/modprobe.d/nvidia.conf` to `/etc/modprobe.d/nvidia-modeset.conf` and that resolves the issue as well. There is `systemd-analyze cat-config modprobe.d` which shows the combines result of all `modprobe.d` directories, the output confirms that one of the two files are ignored when both have the same name.

So additionally to your fix, could you please also rename one of the two `nvidia.conf` files? Otherwise all the power  options in `/lib/modprobe.d/nvidia.conf` will be ignored.

jp7677, 2021-11-04:
@scaronni , I've updated now, I can confirm that your fix works.

My suggestion is to remove `blacklist nvidia` from `/lib/modeprobe.d/nvidia.conf` since this option is now set by grub arguments and to rename `/etc/modeprobe.d/nvidia.conf` to e.g. `/etc/modeprobe.d/nvidia-modeset.conf`to not discard all settings from `/lib/modeprobe.d/nvidia.conf`.

scaronni, 2021-11-06:
Done in https://github.com/negativo17/nvidia-kmod-common/commit/e7a8f52bd82707f24e02e494472f0e2e355a0c90.
Thanks!

closed 2021-11-06 by scaronni

jp7677, 2021-11-06:
Hi @scaronni thanks, looks better.
You are still missing the rename though. Everything in https://github.com/negativo17/nvidia-kmod-common/blob/master/nvidia.conf will still be ignored as long as the other `nvidia.conf` file is present in `/etc/modprobe.d/`.

jp7677, 2021-11-06:
PS: also the comment here is now outdated https://github.com/negativo17/nvidia-kmod-common/blob/master/nvidia-kmod-common.spec#L66

scaronni, 2021-11-07:
> Hi @scaronni thanks, looks better. You are still missing the rename though. Everything in https://github.com/negativo17/nvidia-kmod-common/blob/master/nvidia.conf will still be ignored as long as the other `nvidia.conf` file is present in `/etc/modprobe.d/`.

It's true, I just checked `/proc/driver/nvidia/params` and the settings from the file `/usr/lib/modeprobe.d/nvidia.conf` are ignored. Since when is this a thing? The files always had the same names, and I never had this issue.

Will fix in a moment.

> PS: also the comment here is now outdated https://github.com/negativo17/nvidia-kmod-common/blob/master/nvidia-kmod-common.spec#L66

Sure, thanks.

scaronni, 2021-11-07:
https://github.com/negativo17/nvidia-kmod-common/commit/0bf95dc2d61d6f2f7511631c2a02963ac27afdaf

jp7677, 2021-11-07:
Yeah, this looks good, thanks a lot. Will  report back once its pushed, but pretty sure that this fixes it.

> Since when is this a thing?

I think this has always been like this, but it only surfaced now with Fedora 35. The spec file says that older Fedora versions wont get the `nvidia.conf` in `/etc/modprobe.d/` and thus only one `nvidia.conf` was present which got read just fine. 
Looking at the output of `systemd-analyze cat-config modprobe.d` really help to see which configuration files are taken into account.

scaronni, 2021-11-08:
> I think this has always been like this, but it only surfaced now with Fedora 35. The spec file says that older Fedora versions wont get the `nvidia.conf` in `/etc/modprobe.d/` and thus only one `nvidia.conf` was present which got read just fine. I could be mistaken here though ;)

Good catch! Too many ifs and variations for me to remember...

> Looking at the output of `systemd-analyze cat-config modprobe.d` really help to see which configuration files are taken into account.

I was not aware of this command, pretty neat!

Thanks!