negativo17/nvidia-driver #74: Black Screen on Login Screen

maidis, 2019-02-12:
I'm using [Fedora 29 KDE spin](https://spins.fedoraproject.org/en/kde/) on my [GE62 2QL APACHE](https://www.msi.com/Laptop/support/GE62-2QL-Apache) which has a GeForce GTX 950M as graphics card.

I made a fresh installation yesterday. After installation and full system update, I installed NVIDIA drivers with [following commands](https://github.com/maidis/fedora-kde-spin-customizations#nvidia-drivers):

```bash
sudo dnf config-manager --add-repo=https://negativo17.org/repos/fedora-nvidia.repo
sudo dnf install nvidia-driver kernel-devel dkms-nvidia dkms acpi

sudo dnf install nvidia-settings nvidia-driver-libs.i686

sudo dnf -y install cuda nvidia-driver-cuda cuda-devel
sudo dnf update --refresh
```
Now I've fully working drivers but with a little problem. The login screen is completely black. Although I can't see anything, everything is where it should be. Because I can enter my password and log in. And then everything works normally.

I don't know if this issue is related to video card drivers or to the login manager. But my first suspect is NVIDIA drivers all the time. How can we tell if the problem is really about drivers?

According to my little research on the internet, there are two possible solutions (I haven't tried any of them yet):

* [Adding following line to the `10-nvidia.conf` file](https://bugs.launchpad.net/ubuntu/+source/apt/+bug/1765923/comments/10):

```config
Option "PrimaryGPU" "true"
```

* [Removing `~/.cache` directory](https://www.reddit.com/r/kde/comments/37ut0e/severe_black_screen_after_loginscreen/crrk4dv/):

```bash
mv ~/.cache ~/.cache_backup
```

scaronni, 2019-02-12:
I'm not a KDE spin user myself, does the spin enable KDM or uses GDM?

> sudo dnf install nvidia-driver kernel-devel dkms-nvidia dkms acpi
> sudo dnf install nvidia-settings nvidia-driver-libs.i686
> sudo dnf -y install cuda nvidia-driver-cuda cuda-devel

This can be simplified as follows for the basic drivers:
```
dnf install dkms-nvidia nvidia-driver nvidia-settings nvidia-driver-libs.i686
````
And like this for CUDA:
```
dnf install nvidia-driver-cuda cuda-devel 
```

maidis, 2019-02-12:
> I'm not a KDE spin user myself, does the spin enable KDM or uses GDM?
> 

It's [SDMM](https://github.com/sddm/sddm) I guess:

"Retire KDM as the default display manager of the KDE Fedora Spin in favor of SDDM."

https://fedoraproject.org/wiki/Changes/SDDMinsteadOfKDM

"yes , for example in fedora workstation the GDM , on fedora kde spin you have SDDM"

https://ask.fedoraproject.org/en/question/119968/fedorakde-vs-fedora-kde-spin/

> This can be simplified as follows for the basic drivers:
> 

Thanks, I updated it.

scaronni, 2019-02-12:
Is it an optimus laptop? judging from the link you sent it only has a simple Nvidia card, not an Optimus configuration. All the commands 

> onlyesterday16 [says](https://forums.fedoraforum.org/showthread.php?316354-SDDM-shows-black-screen-with-Nvidia):
> 
> > Hi. I'm using an optimus laptop. I installed Nvidia driver from negativo17's repository.
> > I followed tutorial from [Arch Wiki](https://wiki.archlinux.org/index.php/NVIDIA_Optimus#SDDM) to add "xrandr" to /etc/sddm/Xsetup and it works now.
> 
> And arijitlive [says](https://www.reddit.com/r/archlinux/comments/9knufw/sddm_displays_a_blank_screen_after_nvidia_driver/e71dy3z/):
> 
> > I see you've fixed your problem by downgrading the drivers. If you run into the problem in future, try this one and see if it solves your issue or not. It worked for my Fedora system but might work for you too. add the below lines in /etc/sddm/Xsetup
> 
> > `xrandr --setprovideroutputsource modesetting NVIDIA-0`
> > `xrandr --auto`
> 
> > Reboot the system.

Is it an Optimus laptop? Judging from the link you sent it only has a simple Nvidia card, not an Optimus configuration. All the commands should not be required.

maidis, 2019-02-12:
Yes, my system uses Optimus technology:

```bash
$  lspci | egrep 'VGA|3D'
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 5600 (rev 0a)
01:00.0 3D controller: NVIDIA Corporation GM107M [GeForce GTX 950M] (rev a2)
```

leigh123linux, 2019-02-13:
@maidis  You can use a global rule /etc/X11/xinit/xinitrc.d/ to run those xrandr commands on startup.

maidis, 2019-02-13:
> @maidis You can use a global rule /etc/X11/xinit/xinitrc.d/ to run those xrandr commands on startup.

I created a file named `99-nvidia.sh` that containing the followings:

```bash
xrandr --setprovideroutputsource modesetting NVIDIA-0
xrandr --auto
```
and copied it into `/etc/X11/xinit/xinitrc.d/`. But nothing changed.

After that, I tried copying the same commands to `/etc/sddm/Xsetup` and I could see the login screen.

leigh123linux, 2019-02-13:
> > @maidis You can use a global rule /etc/X11/xinit/xinitrc.d/ to run those xrandr commands on startup.
> 
> I created a file named `99-nvidia.sh` that containing the followings:
> 
> ```shell
> xrandr --setprovideroutputsource modesetting NVIDIA-0
> xrandr --auto
> ```
> 
> and copied it into `/etc/X11/xinit/xinitrc.d/`. But nothing changed.
> 

Did you remember to add a bash shebang to line 1?

> After that, I tried copying the same commands to `/etc/sddm/Xsetup` and I could see the login screen.

maidis, 2019-02-13:
@leigh123linux I forgot to add bash shebang. I've undoed the changes I made in the /etc/sddm/Xsetup file and created `99-nvidia.sh` with a bash shebang at first line in `/etc/X11/xinit/xinitrc.d/` but I have still black screen at login.

leigh123linux, 2019-02-13:
@maidis Does the file have execute?, to set it

```
sudo chmod a+x /etc/X11/xinit/xinitrc.d/99-nvidia.sh
```

maidis, 2019-02-13:
Yes It has. Does it make any difference if the file is run before or after the others?

```bash
ls /etc/X11/xinit/xinitrc.d/
00-start-message-bus.sh  10-qt5-check-opengl2.sh  50-systemd-user.sh  99-nvidia.sh  localuser.sh
```

Iolaum, 2019-06-09:
I am having the same issue after trying to install the negativo17 nvidia drivers on a fresh install of Fedora MATE 30 on an optimus laptop. I have written more details in [this askfedora post](https://ask.fedoraproject.org/t/installing-negativo17-nvidia-drivers-on-optimus-laptop/1967).

I haven't tried any of the `xrandr` related solutions/workarounds mentioned here because I don't understand them enough and how they would apply in my system.

closed 2024-12-09 by scaronni