negativo17/nvidia-driver #75: f29 :: Nouveau cannot be disabled - solved

adriansev, 2019-02-19:
Hi! I just upgraded an f27 to f29 and it seems that nvidia cannot be loaded because nouveau is always present an loaded!! (and no package with nouveau is installed)   
The only solution beside :   
`GRUB_CMDLINE_LINUX="rd.driver.blacklist=nouveau"`   
was :   
```
[root@paowall ~]# cat /etc/modprobe.d/nvidia.conf
blacklist nouveau
install nouveau /bin/true
```
N.B. without the install part nouveau will still be loaded

hevanaa, 2019-02-23:
Are you sure the nvidia driver module was actually compiled and available? Because on Fedora nowadays, nouveau and nvidia can co-exist. So if nvidia fails to load, nouveau loads instead. That happened to me today. For some reason, with the latest 418.43 driver, dkms didn't succeed to compile the module for the current kernel and of course it didn't load after a reboot. Instead nouveau was loaded. I deleted the previous driver directory in /var/lib/dkms/nvidia/ and run 'dkms autoinstall' to create the nvidia module and another reboot fixed the problem.

adriansev, 2019-02-24:
well, yes ... lspci -k shown me the driver options, the kmod package was built and present and just after i added the install /bin/true the nvidia driver was loaded everything worked

closed 2020-02-15 by scaronni