negativo17/nvidia-driver #77: Missing dependencies (EL7.6) on 418.74

ElCoyote27, 2019-05-11:
Hi,
Thank you for your work. Your packaging of the NVidia stuff is excellent and I also recommend it on EL7.
A new update came out but I'm unable to upgrade due to missing deps:

--> Processing Dependency: nvidia-kmod-common = 3:418.74 for package: 3:nvidia-driver-cuda-418.74-1.el7.x86_64
--> Finished Dependency Resolution
Error: Package: 3:nvidia-driver-418.74-1.el7.x86_64 (epel-nvidia)
           Requires: nvidia-kmod-common = 3:418.74
           Installed: 3:nvidia-kmod-common-418.56-2.el7.noarch (@epel-nvidia)
               nvidia-kmod-common = 3:418.56
               nvidia-kmod-common = 3:418.56-2.el7
           Available: 3:nvidia-kmod-common-418.56-1.el7.noarch (epel-nvidia)
               nvidia-kmod-common = 3:418.56
               nvidia-kmod-common = 3:418.56-1.el7

Any ideas? Is there another mirror?

ElCoyote27, 2019-05-11:
I just looked at https://negativo17.org/repos/nvidia/epel-7/x86_64/ and there does not appear to have an nvidia-kmod-common rpm for v418.74, only for v418.56 (the previous version).

scaronni, 2019-05-12:
Fixed it, sorry. Will be online shortly. Forgot to merge some branches while building.

Thanks for reporting.

closed 2019-05-12 by scaronni