negativo17/nvidia-driver #79: Fedora Silverblue/Atomic (ostree) support

p1u3o, 2019-06-04:
Apologies in advanced if this is the wrong place.

akmod works slightly differently on Fedora Atomic/Silverblue compared to a standard install, with the akmods being built inside an offline deployment, that if no errors occur, will be activated for the next boot. 

I haven't done much testing, but the only issue with this driver that prevents installing in Silverblue is that it attempts to modify /etc/default/grub at install, which doesn't exist in the offline deployment.

`
Jun 04 08:27:08 daniels-pc rpm-ostree(nvidia-kmod-common.post)[4211]: /usr/nvidia-kmod-common.post: line 2: /etc/default/grub: No such file or directory
Jun 04 08:27:08 daniels-pc rpm-ostree[4211]: Executed %post for akmod-wireguard in 11265ms
`

I'm going to guess the relevant code is https://github.com/negativo17/nvidia-kmod-common/blob/master/nvidia-kmod-common.spec#L81 and that checking for /run/ostree-booted before running this might be enough.

More info https://bugzilla.redhat.com/show_bug.cgi?id=1667014

scaronni, 2019-06-06:
> Apologies in advanced if this is the wrong place.

Well, here would be more appropriate: https://github.com/negativo17/nvidia-kmod-common

:)

> I'm going to guess the relevant code is https://github.com/negativo17/nvidia-kmod-common/blob/master/nvidia-kmod-common.spec#L81 and that checking for /run/ostree-booted before running this might be enough.
> 
> More info https://bugzilla.redhat.com/show_bug.cgi?id=1667014

Ok, will add it. Thanks.

scaronni, 2019-06-06:
Makes sense?
https://github.com/negativo17/nvidia-kmod-common/commit/3c4241da0a1f626cdecf42a292b42c058d8c0683

p1u3o, 2019-06-06:
Just built the kmod and looks like it's working.

Given grubby doesn't work, you just have to run `rpm-ostree kargs --append=nvidia-drm.modeset=1 append=rd.driver.blacklist=nouveau append=modprobe.blacklist=nouveau` manually, I don't think it's possible to run this is in the offline deployment though.

scaronni, 2019-06-07:
> Given grubby doesn't work, you just have to run `rpm-ostree kargs --append=nvidia-drm.modeset=1 --append=rd.driver.blacklist=nouveau --append=modprobe.blacklist=nouveau` manually, I don't think it's possible to run this is in the offline deployment though.

Any idea if we could somehow hook it in the package? PR on `nvidia-kmod-common` is welcome.
 
> Edit: nvidia-drm doesn't seem to start up Plymouth at full resolution anymore.

It never did for me... `drm/modeset` in nvidia does not support console modesetting AFAIK.

p1u3o, 2019-06-07:
>  Any idea if we could somehow hook it in the package? PR on nvidia-kmod-common is welcome.

I'll ask in the Silverblue channel.

> It never did for me... drm/modeset in nvidia does not support console modesetting AFAIK.

It does, at least on the RPMFusion driver. I did a comparison of their dmesgs and 

`[nvidia-drm] [GPU ID 0x00000100] Loading driver`

occurs sooner on the rpmfusion driver.

scaronni, 2019-06-07:
> It does, at least on the RPMFusion driver. I did a comparison of their dmesgs and
> 
> `[nvidia-drm] [GPU ID 0x00000100] Loading driver`
> 
> occurs sooner on the rpmfusion driver. You still get useless modesets, but plymouth is at full resolution.
> [dmesg-negativo.log](https://github.com/negativo17/nvidia-driver/files/3266885/dmesg-negativo.log)
> [dmesg-rpmfusion.log](https://github.com/negativo17/nvidia-driver/files/3266886/dmesg-rpmfusion.log)
> 
> SIlverblue uses a prebuilt initfamfs by default, I haven't tried with local generation turned on though.

So the wording that it does not support console has been removed from the documentation, will look into it. The dmesg look identical to me. Will look into it, thanks.

p1u3o, 2019-06-07:
> > It does, at least on the RPMFusion driver. I did a comparison of their dmesgs and
> > `[nvidia-drm] [GPU ID 0x00000100] Loading driver`
> > occurs sooner on the rpmfusion driver. You still get useless modesets, but plymouth is at full resolution.
> > [dmesg-negativo.log](https://github.com/negativo17/nvidia-driver/files/3266885/dmesg-negativo.log)
> > [dmesg-rpmfusion.log](https://github.com/negativo17/nvidia-driver/files/3266886/dmesg-rpmfusion.log)
> > SIlverblue uses a prebuilt initfamfs by default, I haven't tried with local generation turned on though.
> 
> So the wording that it does not support console has been removed from the documentation, I think the last time I checked was more than a year ago. The dmesg look identical to me. Will look into it, thanks.

I think grub might need to be hidden for it to work though, which Fedora 30 ships with by default.

p1u3o, 2019-06-14:
I was able to get plymouth working at full resolution again under 430 by running `rpm-ostree initramfs --enable`.

closed 2020-11-01 by scaronni