negativo17/nvidia-driver #81: Crazy Progress Indicator Spinner

maidis, 2019-07-13:
I'm using Fedora 30 KDE spin on my GE62 2QL APACHE which has a GeForce GTX 950M as graphics card. (it's also has Optimus technology, for other details please see #74)

I know this isn't directly related with these nvidia driver packages but I couldn't figure out how to apply the right workaround myself.

The progress indicator/spinner which appears in the panel when opening applications or copying files or connecting to the network spins as fast as it can and thus causes 100% CPU usage, but only on nvidia Optimus systems.

- [Progress Indicator Spinner in notifications causes 100% CPU usage](https://bugs.kde.org/show_bug.cgi?id=347237)
- [QtQuick Animation timing too fast on Linux nvidia optimus systems](https://bugreports.qt.io/browse/QTBUG-45959)
- [PRIME and PRIME Synchronization](https://devtalk.nvidia.com/default/topic/957814/linux/prime-and-prime-synchronization/)
- [Fix Screen Tearing in Ubuntu 18.04 Optimus Laptops](http://ubuntuhandbook.org/index.php/2018/07/fix-screen-tearing-ubuntu-18-04-optimus-laptops/)

scaronni, 2019-11-01:
Still happening with all the latest Optimus goodies in 435.x?

maidis, 2019-11-11:
I used a few workarounds to solve the problem for myself. I've removed all changes that I remember, but I'm not sure if there's anything left. For now it seems there isn't any craziness in my panel :)

I think it's best to close this issue. If I encounter a similar situation when I switch to Fedora 31, I can reopen this or create a new one. Thank you again for maintaining the driver.

closed 2019-11-12 by scaronni