negativo17/nvidia-driver #84: Nvidia audio device not loading with 435.21

Atticfire, 2019-09-11:
Using Fedora 5.2.11-200.fc30.x86_64
nvidia-driver.x86_64	3:430.40-1.fc30	@fedora-nvidia

If I update to nvidia-driver - 3:435.21-1.fc30
I no longer have an Nvidia audio device and pulseaudio no longer starts.

If I downgrade back to 3:430 then audio device returns
01:00.0 VGA compatible controller: NVIDIA Corporation GP107 [GeForce GTX 1050 Ti] (rev a1)
01:00.1 Audio device: NVIDIA Corporation GP107GL High Definition Audio Controller (rev a1)

No errors in dmesg or journalctl only lspci showing no Audio device
Video is still working fine.

I haven't found anyone else with this problem, any ideas where to start?

leigh123linux, 2019-09-12:
Works fine here

```
$ inxi -SGA
System:
  Host: leigh.computer-desktop Kernel: 5.2.11-200.fc30.x86_64 x86_64 
  bits: 64 Desktop: Cinnamon 4.2.3 Distro: Fedora release 30 (Thirty) 
Graphics:
  Device-1: NVIDIA GP107 [GeForce GTX 1050 Ti] driver: nvidia v: 435.21 
  Display: x11 server: Fedora Project X.org 1.20.5 driver: nvidia 
  unloaded: fbdev,modesetting,nouveau,vesa resolution: 3840x2160~60Hz 
  OpenGL: renderer: GeForce GTX 1050 Ti/PCIe/SSE2 v: 4.6.0 NVIDIA 435.21 
Audio:
  Device-1: AMD SBx00 Azalia driver: snd_hda_intel 
  Device-2: NVIDIA GP107GL High Definition Audio driver: snd_hda_intel 
  Sound Server: ALSA v: k5.2.11-200.fc30.x86_64 

```

```
$ lspci |grep NVIDIA
01:00.0 VGA compatible controller: NVIDIA Corporation GP107 [GeForce GTX 1050 Ti] (rev a1)
01:00.1 Audio device: NVIDIA Corporation GP107GL High Definition Audio Controller (rev a1)
```

What type of connection is used for the monitor? (DP or HDMI), and are any cable converters used?

jp7677, 2019-09-15:
Cross-posted here:  https://devtalk.nvidia.com/default/topic/1063181/linux/nvidia-audio-device-not-detected-with-435-21/

scaronni, 2019-09-17:
Should be fixed in kernel 5.3, which has just been released. I guess that in 435 they just removed the workaround that was available in 430 due to the imminente release:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=b516ea586d717472178e6ef1c152e85608b0ce32

Atticfire, 2019-09-22:
@scaronni interesting after reading that kernel update I tried setting the BIOS to enable onboard VGA
With either enabled or AUTO the machine would not start the Nvidia card at all.

Returning BIOS to only enable onboard VGA if the is no PEG just goes back to Video no Audio enabled
Just have to wait till kernel 5.3 is out then

jp7677, 2019-09-28:
The solutions turns out much easier in my case, thanks to a tip from the nvidia devtalk forum.
The issue is here:
https://github.com/negativo17/nvidia-kmod-common/blob/master/60-nvidia.rules#L15

Setting this rules (which explicitly states that it will remove the HDA devices ;)) in comments let the audio devices appear again, even with the current 5.2 kernel.

scaronni, 2019-09-29:
> I don't know what a good solution here would since having these dynamic power management features might be useful for laptop users with a turing GPU (these are the pre-conditions if I read it correctly). May be putting these rules into a separate package would makes sense?

Ah, that's interesting, thanks for the feedback. Basically the point is that before the GPU can go to "deep sleep" it needs to unbind all onboard devices. If that can't happen, the GPU can't reach the deep state.

I guess the best option is not enable the deep sleep (so it does not need to power off) with the registry key and keep the devices on it alive.

jp7677, 2019-09-29:
> I guess the best option is not enable the deep sleep (so it does not need to power off) with the registry key and keep the devices on it alive.

What do you mean with  "keep the devices on it alive"? Removing the udev rules as stated here http://download.nvidia.com/XFree86/Linux-x86_64/435.17/README/dynamicpowermanagement.html  (chapter "Automated Setup")? That would certainly work for me.

scaronni, 2019-09-29:
Yes, but in that case, to have the poweroff happen with "NVreg_DynamicPowerManagement=0x02", the devices onboard of the GPU board needs to be removed first. Ergo, 0x02 (whatever that is called) can not be reached while the HDA controller is alive.

Considering that the additional 0x02 power state is a complete poweroff compared to 0x01, there is no  way to achieve it with the drivers active on the additional onboard devices. I guess this is why it is not enabled by default. I will revert the registry setting + the UDEV rules, or at least leave it commented.

scaronni, 2019-10-01:
New build coming. I've left the 0x02 state enabled and just disabled the workaround. At that point, once the various fixes reach the kernel, the suspension of the onboard GPU devices should just work and power off as well.

closed 2019-10-01 by scaronni

jp7677, 2019-10-01:
Thanks a lot!

Atticfire, 2019-10-07:
@jp7677 @scaronni  Thank you, resolved problem for me.
Just like to add, help like this is why I love Linux and the Linux community. :)